platformio/platformio-core #3377: PLATFORMIO_... environment variable causes a crash

mcspr, 2020-02-13:
# Configuration

**Operating system**: Linux

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 4.2.0

### Description of problem

For example, `PLATFORMIO_SRC_BUILD_FLAGS=-DFOO` will crash instantly

https://github.com/platformio/platformio-core/commit/2763853d8 seems to be the cause

#### Steps to Reproduce

1. upgrade to pio 4.2.0
2. `pio run` to check that the project works
3. `env PLATFORMIO_SRC_BUILD_FLAGS='-DFOO' pio run` will instantly crash

### Actual Results

The specific issue is shown by this line:
```
 File "/home/max/git/platformio-core/platformio/project/config.py", line 276, in getraw
    value += ("" if value == MISSING else "\n") + envvar_value
TypeError: unsupported operand type(s) for +=: 'object' and 'str'
```

Full traceback:
```python
> env PLATFORMIO_SRC_BUILD_FLAGS='-DFOO' pio run
Can not remove temporary directory `/home/max/Documents/PlatformIO/Projects/sysenv-test/.pio/build`. Please remove it manually to avoid build issues
Error: Traceback (most recent call last):
  File "/home/max/git/platformio-core/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/max/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/max/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/max/git/platformio-core/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/max/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/max/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/max/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/max/.platformio/penv/lib64/python3.8/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/max/git/platformio-core/platformio/commands/run/command.py", line 128, in cli
    process_env(
  File "/home/max/git/platformio-core/platformio/commands/run/command.py", line 165, in process_env
    print_processing_header(name, config, verbose)
  File "/home/max/git/platformio-core/platformio/commands/run/command.py", line 191, in print_processing_header
    for k, v in config.items(env=env):
  File "/home/max/git/platformio-core/platformio/project/config.py", line 232, in items
    return [(option, self.get(section, option)) for option in self.options(section)]
  File "/home/max/git/platformio-core/platformio/project/config.py", line 232, in <listcomp>
    return [(option, self.get(section, option)) for option in self.options(section)]
  File "/home/max/git/platformio-core/platformio/project/config.py", line 305, in get
    value = self.getraw(section, option, default)
  File "/home/max/git/platformio-core/platformio/project/config.py", line 276, in getraw
    value += ("" if value == MISSING else "\n") + envvar_value
TypeError: unsupported operand type(s) for +=: 'object' and 'str'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================```

Note about the trace: platformio penv is using git installation
```
> cat ~/.platformio/penv/lib/python3.8/site-packages/platformio.egg-link
/home/max/git/platformio-core
```

### Expected Results

No crash

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:test]
platform = espressif32
board = esp32dev
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
void setup() {}
void loop() {}
```

### Additional info

First noticed this here:
https://travis-ci.org/xoseperez/espurna/jobs/649841340#L378

CI script uses PLATFORMIO_ variable to build

closed 2020-02-13 by ivankravets

ivankravets, 2020-02-13:
Thanks for the report. This my fault, I didn't cover this use case in a test when multiline system evironment is used :(

See https://github.com/xoseperez/espurna/pull/2146

maksim-ka, 2020-05-21:
A similar issue is present on version 4.3.4a4 when build_flags are set through extra_configs in platformio.ini. Though the resulting error is a RecurrsionError.

ivankravets, 2020-05-21:
Could you provide a simple project to reproduce this issue?

maksim-ka, 2020-05-21:
I just realized that i'm trying to add buildflags. By build_flags =  ${env.build_flags}...etc.
Example: 
platformio.ini: 
```
...
[env]
   build_flags = -DDEBUG
...
```
credentials.ini
```
[env]
build_flags = 
    ${env.build_flags}
    -DUSERNAME=user
    -DPASSWORD=pass
```

maksim-ka, 2020-05-21:
So the issue is actually not about the build_flag in particular, it's about the re-referencing. 
I'm sorry, about realizing it just now. Should I open a new issue for that?

ivankravets, 2020-05-21:
It’s not a bug. You are referencing to the same scope multiple time and are making infinite recursion. 

Change env in credentials to something [wifi] and in main platformio.ini use $wifi

commit 36a22282 (2020-02-13): Fixed "TypeError: unsupported operand type(s)" when system environment variable is used by project configuration parser // Resolve #3377