platformio/platformio-core #3378: PIO fails to generate compile_commands.json for an MBED project

ASMfreaK, 2020-02-13:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
```
$ uname -a
Linux  5.4.7-arch1-1 #1 SMP PREEMPT Tue, 31 Dec 2019 17:20:16 +0000 x86_64 GNU/Linux
```

**PlatformIO Version** (`platformio --version`):
```
$ pio --version
PlatformIO, version 4.2.1a1
```

### Description of problem
This is a continuation of #2990. Compilation DB fails to be generated if used on MBED project.

#### Steps to Reproduce

1. `mkdir someproject`
1. `pio project init`
1. Paste `platformio.ini` from below
1. Paste `mbed_app.json` from below
1. Run `pio -t compiledb`

### Actual Results

```
Processing BASE (board: bluepill_f103c8; framework: mbed; platform: ststm32)
----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 5.7.0 > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-mbed 5.51401.191023 (5.14.1) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Collecting mbed sources...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Building compilation database .pio/build/BASE/compile_commands.json
TypeError: changed_since_last_build_node() takes exactly 4 arguments (3 given):
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1346:
    _exec_main(parser, values)
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1309:
    _main(parser)
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1091:
    nodes = _build_targets(fs, options, targets, target_top)
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1283:
    jobs.run(postfunc = jobs_postfunc)
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Job.py", line 111:
    self.job.start()
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Job.py", line 410:
    task.executed()
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 226:
    SCons.Taskmaster.OutOfDateTask.executed(self)
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Taskmaster.py", line 310:
    t.release_target_info()
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Node/FS.py", line 2990:
    self.changed(allowcache=True)
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Node/FS.py", line 3241:
    has_changed = SCons.Node.Node.changed(self, node)
  File "[HOME]/.platformio/packages/tool-scons/script/../engine/SCons/Node/__init__.py", line 1460:
    if _decider_map[child.changed_since_last_build](child, self, prev_ni):
```

### Expected Results
Generated `compile_commands.json`

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
include_dir = include
lib_dir = lib
test_dir = test
build_cache_dir = .cache

[env]
lib_ignore = 
	mbed-rtos, mbed-LWIP, mbed-events, mbed-fs,
	mbed-net, mbed-netsocket, mbed-rpc, mbed-dsp,
	mbed-USBHost, mbed-USBDevice
upload_protocol = stlink
platform = ststm32
build_flags =
    -D MBED_BUILD_PROFILE_RELEASE
    -D MBED_FAULT_HANDLER_DISABLED
    -D ARDUINOJSON_USE_ARDUINO_STRING=0
    -D ARDUINOJSON_ENABLE_STD_STRING=1

[bluepill]
board = bluepill_f103c8
f_cpu = 72000000L
framework = mbed
lib_deps = modrob_mbed

[env:BASE]
board = ${bluepill.board}
board_build.f_cpu = ${bluepill.f_cpu}
framework = ${bluepill.framework}
build_flags = ${env.build_flags}
```

** mbed_app.json **
```
{
  "macros": [
    "NDEBUG=1"
  ],
  "target_overrides": {
    "*": {
      "target.console-uart": false,
      "platform.stdio-flush-at-exit": false,
      "platform.stdio-convert-newlines": false
    }
  },
  "requires": ["bare-metal", "minimal-printf"]
}
```

valeros, 2020-02-14:
Hi @ASMfreaK ! Thanks for the report. I followed every step you described above and wasn't able to reproduce the issue neither on Ubuntu 18.04 nor on Windows. Is there anything special about your environment?

herrfeuer, 2020-02-14:
same here 
PlatformIO, version 4.2.0
via Visual Studio Code Plugin on
Debian Buster  
Linux 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux

can reproduce it by throwing away everything
rm -rf  ~/.config/Code ~/.platformio ~/.vscode  && apt purge code 
apt install code
install platformio plugin, install nrf framework for micro:bit, create new mbed project for micro:bit, install libs 360 & 361, build

ASMfreaK, 2020-02-15:
My setup should be pretty standard.
I'm running Arch Linux.
I had my version of PIO installed in a virtual environment with Python 2 from an Atom plugin and kept it since.
```
$ . ~/.platformio/penv/bin/activate
$ python --version 
Python 2.7.17
```

Should I try recreating the venv with Python 3?

ivankravets, 2020-02-17:
Thanks for the report! The PIO Core 4.2.1 is planned for release today.

commit 6328206e (2020-02-17): Fixed an issue when generating of compilation database "compile_commands.json" does not work with Python 2.7 // Resolve #3378
closed 2020-02-17 by ivankravets