platformio/platformio-core #3379: path error with mbed on macOS

sstaub, 2020-02-13:
see https://community.platformio.org/t/path-problem-with-mbed-h/12001/12

valeros, 2020-02-14:
Many thanks! Resolved in fbfbf340c12caf56d8566e24df8c14441d5c9c68. Please retest with the latest platformio from dev branch, just run `pio upgrade --dev` in the terminal.

sstaub, 2020-02-14:
Solved, thank you.

closed 2020-02-16 by ivankravets

gracedotdotdot, 2020-03-06:
I also get a same problem on lubuntu, which is the Virtual Machine I add on my original windows PC.May I ask why?
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/home/ee2405/ee2405/mbed0/mbed-os-test-program/main.cpp).
[image]

ivankravets, 2020-03-09:
@gracedotdotdot do you use the latest PIO Core? Please open PlatformIO IDE Terminal and type `pio upgrade`.

sajadghawami, 2020-03-15:
I am on a Mac and i still got that same problem, even when upgrading with "pio upgrade --dev":

[image]

The error also says: "#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit "

I ve been searching for at least 2 days to solve this issue...

valeros, 2020-03-15:
Hi @sajadghawami ! What are the platform and board you're using? Could you please upload somewhere your `c_cpp_properties.json` file.

sajadghawami, 2020-03-15:
Thanks for the quick reply! :)

This is the plattformio.ini:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nano33ble]
platform = nordicnrf52
board = nano33ble
framework = arduino

```

And this is the `c_cpp_properties.json`:

```
{
    "configurations": [
        {
            "name": "!!! WARNING !!! AUTO-GENERATED FILE, PLEASE DO NOT MODIFY IT AND USE https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags"
        },
        {
            "name": "Mac",
            "macFrameworkPath": [],
            "includePath": [
                "/Users/CURRENT_USER/CURRENT_PROJECT/include",
                "/Users/CURRENT_USER/CURRENT_PROJECT/src",
                "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino",
                "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/api/deprecated",
                "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/variants/ARDUINO_NANO33BLE",
                "/Users/CURRENT_USER/.platformio/lib/ArduinoBLE_ID5878/src",
                "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/PDM/src",
                "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/SPI",
                "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBHID",
                "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBMSD",
                "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/Wire",
                "/Users/CURRENT_USER/.platformio/packages/tool-unity",
                ""
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": true,
                "path": [
                    "/Users/CURRENT_USER/CURRENT_PROJECT/include",
                    "/Users/CURRENT_USER/CURRENT_PROJECT/src",
                    "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino",
                    "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/api/deprecated",
                    "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/variants/ARDUINO_NANO33BLE",
                    "/Users/CURRENT_USER/.platformio/lib/ArduinoBLE_ID5878/src",
                    "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/PDM/src",
                    "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/SPI",
                    "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBHID",
                    "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBMSD",
                    "/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/Wire",
                    "/Users/CURRENT_USER/.platformio/packages/tool-unity",
                    ""
                ]
            },
            "defines": [
                "PLATFORMIO=40300",
                "ARDUINO_ARDUINO_NANO33BLE",
                "ARDUINO_ARCH_NRF52840",
[... 66 lines cut ...]
                "MBED_NO_GLOBAL_USING_DIRECTIVE=1",
                "USE_ARDUINO_PINOUT",
                "ARDUINO=10810",
                "ARDUINO_ARCH_MBED",
                ""
            ],
            "intelliSenseMode": "clang-x64",
            "cStandard": "c11",
            "cppStandard": "c++14",
            "compilerPath": "/Users/CURRENT_USER/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc",
            "compilerArgs": [
                "-mcpu=cortex-m4",
                "-mfloat-abi=softfp",
                "-mfpu=fpv4-sp-d16",
                "-mthumb",
                "-iprefix/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino",
                "@\"/Users/CURRENT_USER/.platformio/packages/framework-arduino-nrf52-mbedos/variants/ARDUINO_NANO33BLE/includes.txt\"",
                ""
            ]
        }
    ],
    "version": 4
}

```

valeros, 2020-03-15:
OK, that's strange, could you please try the following config:
```
[env:nano33ble]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = nano33ble
framework = arduino
```

sajadghawami, 2020-03-16:
after doing this, and rebuildig it seems to have worked! Thanks :)

Out of curiosity - how come?