platformio/platformio-core #3380: Freeze version of marshmallow to 2.20.5

dv01d, 2020-02-13:
What kind of issue is this?

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Customer

**PlatformIO Version** (`platformio --version`): 4.2.0

### Description of problem
Updating or installing platformio results in an incompatible version of marshmallow to be installed.

Traceback (most recent call last):
  File "/usr/bin/platformio", line 9, in <module>
    load_entry_point('platformio==4.2.0', 'console_scripts', 'platformio')()
  File "/usr/lib/python2.7/site-packages/setuptools-22.0.5-py2.7.egg/pkg_resources/__init__.py", line 542, in load_entry_point
  File "/usr/lib/python2.7/site-packages/setuptools-22.0.5-py2.7.egg/pkg_resources/__init__.py", line 2569, in load_entry_point
  File "/usr/lib/python2.7/site-packages/setuptools-22.0.5-py2.7.egg/pkg_resources/__init__.py", line 2229, in load
  File "/usr/lib/python2.7/site-packages/setuptools-22.0.5-py2.7.egg/pkg_resources/__init__.py", line 2235, in resolve
  File "/usr/lib/python2.7/site-packages/platformio/__main__.py", line 21, in <module>
    from platformio import __version__, exception, maintenance, util
  File "/usr/lib/python2.7/site-packages/platformio/maintenance.py", line 24, in <module>
    from platformio.commands.lib import CTX_META_STORAGE_DIRS_KEY
  File "/usr/lib/python2.7/site-packages/platformio/commands/lib.py", line 29, in <module>
    from platformio.package.manifest.schema import ManifestSchema
  File "/usr/lib/python2.7/site-packages/platformio/package/manifest/schema.py", line 17, in <module>
    import marshmallow
  File "/usr/lib/python2.7/site-packages/marshmallow/__init__.py", line 1, in <module>
    from marshmallow.schema import Schema, SchemaOpts
  File "/usr/lib/python2.7/site-packages/marshmallow/schema.py", line 131
    klass: type,
         ^
SyntaxError: invalid syntax

#### Steps to Reproduce

1. Install or update platformio
2. run platformio
3. fail

### Additional info
Just need to explicitly define version 2.20.5 for it to continue to work.

ivankravets, 2020-02-14:
What is your OS? Could you provide me an output of `pip --version`?

ivankravets, 2020-02-14:
1. This issue is caused by your outdated PIP version which does not handle Python compatibility when looking for a package
2. I did also a fix on our side. So, it should work with old PIP client too.

closed 2020-02-14 by ivankravets

jamesmyatt, 2020-02-18:
Might be better to implement this using built-in functionality in setuptools

https://setuptools.readthedocs.io/en/latest/setuptools.html#declaring-platform-specific-dependencies

ivankravets, 2020-02-18:
Do you have a problem with the latest PIO Core 4.2.1? Indeed, `pip` handles package compatibility with a Python version. The problem is linked with outdated `pip` and `setuptools` packages that do not understand new syntax. This is why a hard declaration should work for old and new packages.

commit a57ea79b (2020-02-14): Froze "marshmallow" dependency to 2.X for Python 2 // Resolve #3380