platformio/platformio-core #3381: [esp32] `sdkconfig.h` not found with  v4.2, was working with v4.1

giupo, 2020-02-15:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 (`Microsoft Windows [Version 10.0.18362.657]`)

**PlatformIO Version** (`platformio --version`): 4.2.0

### Description of problem
After upgrading to PlatformIO v4.2, `pio run` yields the following error:
```bash
fatal error: sdkconfig.h: No such file or directory
```
but if I downgrade PIO to v4.1, it yields the expected result (compilation successful)

#### Steps to Reproduce

I've setup a repo to reproduce the test:
1. `pip install platformio==4.2.0`
2. `git clone https://github.com/giupo/https://github.com/giupo/test.case.platformio` or create an empty project with `pio init` with the following specified `ini` and `cpp` in this issue.
3. `pio run`

### Actual Results

```bash
In file included from .pio\libdeps\esp12e\AsyncTCP_ID1826\src\AsyncTCP.cpp:24:0:
.pio\libdeps\esp12e\AsyncTCP_ID1826\src\AsyncTCP.h:26:23: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://platformio.org/lib/search?query=header:sdkconfig.h
*
*******************************************************************

 #include "sdkconfig.h"
                       ^
compilation terminated.
*** [.pio\build\esp12e\lib1ea\AsyncTCP_ID1826\AsyncTCP.cpp.o] Error 1
=============== [FAILED] Took 8.64 seconds ====================
```
### Expected Results

The expected result is obtained by downgrading to 4.1:

1. `pip install platformio==4.1.0`

Here's the expected output:

```bash
********************************************************************************
Obsolete PIO Core v4.1.0 is used (previous was 4.2.0)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
********************************************************************************
Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 2.3.3 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), framework-arduinoespressif8266 3.20603.200130 (2.6.3), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for ESP Async Webserver library in registry
Found: https://platformio.org/lib/show/306/ESP Async WebServer
LibraryManager: Installing id=306
ESP Async WebServer @ 1.2.3 is already installed
Installing dependencies
LibraryManager: Installing id=305
ESPAsyncTCP @ 1.2.2 is already installed
LibraryManager: Installing id=1826
AsyncTCP @ 1.1.1 is already installed
Looking for Hash library in registry
Warning! Library `{'platforms': ['espressif8266'], 'name': 'Hash'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'platforms': ['espressif8266'], 'name': 'Hash'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
Building in release mode
Retrieving maximum program size .pio\build\esp12e\firmware.elf
Checking size .pio\build\esp12e\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [===       ]  32.7% (used 26772 bytes from 81920 bytes)
PROGRAM: [==        ]  24.5% (used 255680 bytes from 1044464 bytes)
========================= [SUCCESS] Took 2.52 seconds =========================

```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env]
lib_deps =
    ESP Async Webserver
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
board_build.flash_mode = qio
upload_speed = 115200
monitor_speed = 115200

```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <Wire.h>
#include <ESPAsyncWebServer.h>

void setup() {

}

void loop () {

}
```

### Additional info
I guess the update was triggered by PlatformIO IDE for Visual Studio Code?

ivankravets, 2020-02-15:
Resolved in https://github.com/platformio/platformio-core/commit/5cc9a328ab4d64c42f3cda714f65e150eca31128

Please run `pio upgrade --dev`. Also, see other workarounds https://community.platformio.org/t/pio-4-2-0-complile-error-for-asynctcp-id1826/12025/5?u=ivankravets

closed 2020-02-15 by ivankravets

giupo, 2020-02-15:
Just for the info:
 - `lib_compat_mode = strict` worked, 
 - `pio upgrade --dev` worked, 
 - `lib_ignore=AsyncTCP` **didn't work**.

Many thanks @ivankravets  for the prompt solution.

mvadu, 2020-02-16:
This issues was affecting  ESPAsyncWebServer library as well,  which was compiling fine before `platformio upgrade`. I started getting bunch of errors (e.g. `'panic' was not declared in this scope`, `'ip_addr_t {aka struct ip_addr}' has no member named 'addr'` etc). 

`lib_compat_mode = strict` worked. However since the actual fix is still not released to mainstream, can we keep this issue open until the commit goes to a PR and gets merged?

ivankravets, 2020-02-16:
We plan to release PIO Core 4.2.1 tomorrow.

mvadu, 2020-02-16:
ah.. nice.. IMHO usually closing an issue hides it from default github search, and leads to many duplicates. Keeping it open until mainstream has the fix solves it.

ivankravets, 2020-02-16:
I agree, thanks for the report!