platformio/platformio-core #3387: UnicodeDecodeError when executing pio run

trentmillar, 2020-02-19:
What kind of issue is this?

- [x] **PlatformIO Core**.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Mac osx 10.15.3

**PlatformIO Version** (`platformio --version`): 4.2.1

### Description of problem

I have tried multiple boards, nodemcu, uno, espressif32 and I have uninstalled and reinstalled the last 3 version of PlatformIO Core but every time I execute `platformio run` I get this strange error:
`UnicodeDecodeError: 'utf-8' codec can't decode byte 0xca in position 0: invalid continuation byte:`

If it helps, here's my locale configuration:
```
$ locale
LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL="en_US.UTF-8"
```
#### Steps to Reproduce

1. `pip install -U platformio`
Requirement already up-to-date: platformio in /usr/local/lib/python3.7/site-packages (4.2.1)
Requirement already satisfied, skipping upgrade: colorama in /usr/local/lib/python3.7/site-packages (from platformio) (0.4.1)
Requirement already satisfied, skipping upgrade: marshmallow>=2 in /usr/local/lib/python3.7/site-packages (from platformio) (3.4.0)
Requirement already satisfied, skipping upgrade: bottle<0.13 in /usr/local/lib/python3.7/site-packages (from platformio) (0.12.18)
Requirement already satisfied, skipping upgrade: pyelftools<1,>=0.25 in /usr/local/lib/python3.7/site-packages (from platformio) (0.26)
Requirement already satisfied, skipping upgrade: semantic-version<3,>=2.8.1 in /usr/local/lib/python3.7/site-packages (from platformio) (2.8.4)
Requirement already satisfied, skipping upgrade: click<8,>=5 in /usr/local/lib/python3.7/site-packages (from platformio) (7.0)
Requirement already satisfied, skipping upgrade: tabulate<1,>=0.8.3 in /usr/local/lib/python3.7/site-packages (from platformio) (0.8.6)
Requirement already satisfied, skipping upgrade: pyserial!=3.3,<4,>=3 in /usr/local/lib/python3.7/site-packages (from platformio) (3.4)
Requirement already satisfied, skipping upgrade: requests<3,>=2.4.0 in /usr/local/lib/python3.7/site-packages (from platformio) (2.22.0)
Requirement already satisfied, skipping upgrade: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python3.7/site-packages (from requests<3,>=2.4.0->platformio) (3.0.4)
Requirement already satisfied, skipping upgrade: idna<2.9,>=2.5 in /usr/local/lib/python3.7/site-packages (from requests<3,>=2.4.0->platformio) (2.8)
Requirement already satisfied, skipping upgrade: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /usr/local/lib/python3.7/site-packages (from requests<3,>=2.4.0->platformio) (1.25.8)
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in /usr/local/lib/python3.7/site-packages (from requests<3,>=2.4.0->platformio) (2019.11.28)

2. `pio project init --board uno`
***********************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
***********************************************************************************************************************

The current working directory /Users/tmillar/dev/repo/aquabotics/pio will be used for the project.

The next files/directories have been created in /Users/tmillar/dev/repo/aquabotics/pio
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Project has been successfully initialized! Useful commands:
`pio run` - process/build project from the current directory
`pio run --target upload` or `pio run -t upload` - upload firmware to a target
`pio run --target clean` - clean project (remove compiled files)
`pio run --help` - additional information

3. `pio run`
Processing uno (platform: atmelavr; board: uno; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing atmelavr
atmelavr @ 2.0.0 has been successfully installed!
The platform 'atmelavr' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-atmelavr @ ~1.50400.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
toolchain-atmelavr @ 1.50400.190710 has been successfully installed!
PackageManager: Installing framework-arduino-avr @ ~5.0.0
Unpacking  [####################################]  100%          
framework-arduino-avr @ 5.0.0 has been successfully installed!
CorePackageManager: Installing tool-scons @ ~3.30102.0
Downloading  [####################################]  100%          
tool-scons @ 3.30102.0 has been successfully installed!
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xca in position 0: invalid continuation byte:
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/main.py", line 89:
    env = DefaultEnvironment(**DEFAULT_ENV_OPTIONS)
  File "/Users/tmillar/.platformio/packages/tool-scons/script/../engine/SCons/Defaults.py", line 88:
    _default_env = SCons.Environment.Environment(*args, **kw)
  File "/Users/tmillar/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 951:
    platform(self)
  File "/Users/tmillar/.platformio/packages/tool-scons/script/../engine/SCons/Platform/__init__.py", line 129:
    return self.generate(*args, **kw)
  File "/Users/tmillar/.platformio/packages/tool-scons/script/../engine/SCons/Platform/darwin.py", line 60:
    lines = f.readlines()
  File "/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/codecs.py", line 322:
    (result, consumed) = self._buffer_decode(data, self.errors, final)
============================================= [FAILED] Took 39.72 seconds =============================================

### Actual Results

UnicodeDecodeError

### Expected Results

No error

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino
```

**Source file to reproduce issue:**

There is no source
```cpp
Insert here...
```

### Additional info

ivankravets, 2020-02-19:
Could you help us to debug this issue?

1) Edit `/usr/local/lib/python3.7/site-packages/platformio/builder/main.py` and add the next line before 89 line:
```
print(DEFAULT_ENV_OPTIONS)
```

Do you see any non-ascii chars?

trentmillar, 2020-02-19:
Thanks for putting eyes on this...it's clear my env vars are causing the
issue and not pio. I resolved the issue on my side and I can now
successfully `run`.

closed 2020-02-19 by trentmillar

ivankravets, 2020-02-19:
How to reproduce this issue? Should we have a fix on our side?

trentmillar, 2020-02-19:
in `/etc/paths.d` I had an executable that I wanted on my `PATH`. However, unbeknownst to me executable was sending garbage that should instead have echo'ed out everything it wanted `path`'ed, `export`'ed, `alias`'ed to bash (at least this is my take on /etc/profile). This is all typical stuff except I had one bad executable.

This corner case aside, there may be a change to `main.py` specifically when loading env vars? The reason I say this is that this executable has been in my path for years. and I've never encountered the issue.

> How to reproduce this issue? Should we have a fix on our side?

Maybe? but probably not...
If you know what env vars you are looking for on a user's path then fetching just these vs. [environ](https://github.com/platformio/platformio-core/blob/1c87f8346323bbfa84c85da79c190232ef60b57b/platformio/builder/main.py#L70) which loads everything.

Basically anything that loads into env vars will come from this trace, in my case it is a lot,
`PS4='path to file: $BASH_SOURCE :' BASH_XTRACEFD=7 bash -xl 7>&2`