platformio/platformio-core #3390: Ble Adv - from Linux Zephyr to Windows platformio Zephyr

tcpipchip, 2020-02-20:
Hello Platformio Team!

I have using for a long time the Zephyr on Linux! All working very fine, i can include and build i2c devices, like the BME280, using Kconfig (make menuconfig)

After read that Platformio team started to port Zephyr to Platformio, i get interested to see if works, and yes, works! Compiled some Ble examples and runs well!

Now, i having problems to enable to use the I2c library and then the BME280, got error, not recognize the existence of them.

Here th GITHUB.

https://github.com/tcpipchip/zephyr_linux_to_platfomio

Can you compile and make the changes to compile ? Or give tips!

Thank you so much and again congratulation to the Platformio team and support!

valeros, 2020-02-26:
Hi @tcpipchip ! What version of `Zephyr` framework do you use on Linux?

tcpipchip, 2020-02-26:
Hi, 2.1
I am backing to PLATFORMIO, i can run 100% BLE ADV stack, but now success to access the BLE280, can you help me ?

valeros, 2020-02-27:
Not sure how you're able to compile this project on Linux, but it looks like the driver for `BME280` sensor relies on specific device tree configuration in order to select the correct communication bus. You need to specify which bus to use in your `board_name.overlay` file, e.g. for `nrf52832_mdk` board you need to create a new file `nrf52832_mdk.overlay` in `zephyr` folder with the following contents (I used `I2C0` since you selected it in `prj.conf`):
```
&i2c0 {
        bme280@76 {
                compatible = "bosch,bme280";
                label = "BME280";
                reg = <0x76>;
        };
};
```

tcpipchip, 2020-02-27:
hi @valeros 

Something like this ?
[image]

Looks that will program the GPIO 30 and GPIO 31 be the I2C.

Are those parameters enough ?

and 
**reg = <0x40003000 0x1000>;
clock-frequency = <100000>; 
compatible = "nordic,nrf-twi";
status = "okay";**

?

valeros, 2020-02-27:
@tcpipchip We don't provide here support for framework configuration process, but if this overlay works fine with Zephyr on Linux, it should work with PlatformIO as well.

tcpipchip, 2020-02-27:
ok, thanks @valeros

closed 2020-03-02 by ivankravets

GeorgeGkinis, 2020-03-20:
I have the same issue.
I am using **Windows10**, **clion** with **platformio** plugin.

My **platformio.ini** file :
``` 
[env:nrf52_dk]
platform = nordicnrf52
board = nrf52_dk
framework = zephyr

debug_tool = jlink
; SWD interface
upload_protocol = jlink
```

my **zephyt/nrf52_dk.overlay** :  
```
&i2c0 {
	bme280@76 {
		compatible = "bosch,bme280";
		reg = <0x76>;
		label = "BME280";
	};
};
```

**output** : 
```
Compiling .pio\build\nrf52_dk\bme280\bme280.c.o
In file included from C:\Users\dbkggk\.platformio\packages\framework-zephyr\drivers\sensor\bme280\bme280.c:24:
C:\Users\dbkggk\.platformio\packages\framework-zephyr\drivers\sensor\bme280\bme280.h:111:2: error: #error "BME280 device type not specified"
```

Any help?

valeros, 2020-03-20:
Hi @GeorgeGkinis ! You need to name your overlay file according to the official board name used in Zephyr framework, e.g. `nrf52_pca10040.overlay`

tcpipchip, 2020-03-20:
@GeorgeGkinis 

Let me know if you get success !

and where did you config the GPIOS used to BME280 ?

GeorgeGkinis, 2020-03-20:
Actually it works with **nrf52_pca10040.overlay**
thanks!

I used the default pins from 

https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf52_pca10040/nrf52_pca10040.dts

I have plugged the bme280 on i2c0 pins : 

```
arduino_i2c: &i2c0 {
	compatible = "nordic,nrf-twi";
	status = "okay";
	sda-pin = <26>;
	scl-pin = <27>;
};
```

It builds and runs, but cannot make a connection with the bme280.

tcpipchip, 2020-03-20:
I had the same problem too! On linux worked fine!

GeorgeGkinis, 2020-03-20:
Filthy cables. 

It works!

Thanks for the support @valeros

tcpipchip, 2020-03-21:
can you share your github platformio project ?

tcpipchip, 2020-03-22:
it´s working :)
Very very good!!!!!

tcpipchip, 2020-11-27:
Sir, i am using the BME280, where i define on Plaftormio the BME280_TEMP_OVER_1X to default value ?

valeros, 2020-11-30:
You can set it directly in your `prj.conf`.

tcpipchip, 2020-11-30:
i set on Kconfig and works!

@valeros 

Can you help me now how to disable the UART perifpheral or something else to get low power ?

My ZEPHYR RTOS BLE is getting 75uA comsuption...too much!

Thanks

valeros, 2020-11-30:
No, sorry. But you can read more about Zephyr power management here https://docs.zephyrproject.org/latest/reference/power_management/index.html