platformio/platformio-core #3391: Generating broken CMakeLists.txt file.

nickgzzjr, 2020-02-20:
What kind of issue is this?

PlatformIO CLI

This happens when I init a platformio project or when I update the index.

### Configuration

**Operating system**: Mac OSX

**PlatformIO Version** (`platformio --version`) : 4.2.2a1 (I upgraded to dev to see if this issue was fixed)

### Description of problem

Generating a broken CMakeLists.txt file. e.g. It is adding lines like `add_definitions(-D'')` and `add_definitions(-D'-DARM_MATH_CM4')` instead of `add_definitions(-D'ARM_MATH_CM4')`.

#### Steps to Reproduce

1. platformio init --ide clion --board nano33ble 
2. Open in CLion
3. CMake will Fail

### Actual Results

CMake Fails and CLion features break.

### Expected Results

CMake should succeed. 

### Additional info

I can workout around this by removing the extra "-D".

valeros, 2020-02-26:
Hi @nickgzzjr ! Thanks for the report. Both issues are fixed but not released yet. You can retest your project using the upstream version of `nordicnrf52` platform:
```
[env:upstream_develop]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = nano33ble
framework = arduino
```

ivankravets, 2020-02-29:
Fixed in https://github.com/platformio/platform-nordicnrf52.git

closed 2020-02-29 by ivankravets