platformio/platformio-core #3395: Could not load logical devices

GooseCapital, 2020-02-29:
Server error: {"args":[2,"The system cannot find the file specified",null,2,null],"message":"[WinError 2] The system cannot find the file specified","type":"FileNotFoundError"}

ivankravets, 2020-02-29:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

GooseCapital, 2020-03-03:
C:\Windows\system32>echo %PATH%
C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Common Files\Siemens\ACE\Bin;C:\Program Files (x86)\Siemens\Automation\SCADA-RT_V11\WinCC\Bin;C:\Program Files (x86)\Common Files\Siemens\Bin;C:\Program Files (x86)\Common Files\Siemens\CommonArchiving;C:\Program Files\PuTTY\;C:\Program Files\MATLAB\R2014b\runtime\win64;C:\Program Files\MATLAB\R2014b\bin;C:\Program Files (x86)\Common Files\Siemens\;C:\Program Files (x86)\Common Files\Siemens\Interfaces;C:\Program Files (x86)\Common Files\Siemens\ACE\Interfaces;C:\Program Files (x86)\Siemens\Automation\SCADA-RT_V11\WinCC\Interfaces;C:\Program Files\dotnet\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Users\GooseCapital\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Atmel\Flip 3.4.7\bin;C:\Users\GooseCapital\spicetify-cli;C:\Users\GooseCapital\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\GooseCapital\AppData\Local\Programs\Python\Python38-32;C:\Users\GooseCapital\AppData\Local\Programs\Python\Python38-32\Scripts;C:\Users\GooseCapital\AppData\Local\Programs\Fiddler;C:\Users\GooseCapital\.dotnet\tools;C:\Users\GooseCapital\AppData\Roaming\npm;C:\Users\GooseCapital\.platformio\penv\Scripts

C:\Windows\system32>python --version
Python 3.7.6

C:\Windows\system32>pip --version
pip 20.0.2 from C:\Users\GooseCapital\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\pip (python 3.7)

C:\Windows\system32>pip list
Package           Version
----------------- ----------
astroid           2.3.3
bottle            0.12.18
certifi           2019.11.28
chardet           3.0.4
Click             7.0
colorama          0.4.3
ecdsa             0.15
esptool           2.8
idna              2.9
isort             4.3.21
lazy-object-proxy 1.4.3
marshmallow       3.5.0
mccabe            0.6.1
pip               20.0.2
pyaes             1.6.1
pyelftools        0.26
pylint            2.4.4
pyserial          3.4
requests          2.23.0
semantic-version  2.8.4
six               1.14.0
tabulate          0.8.6
typed-ast         1.4.1
urllib3           1.25.8
wheel             0.34.2
wrapt             1.11.2

Thanks for helping me 🍡

ivankravets, 2020-03-03:
What do you see when run in PlatformIO IDE Terminal?
```
pio device list --json-output
```

closed 2020-03-09 by ivankravets