platformio/platformio-core #3397: PIO home security model

FedericoCeratto, 2020-03-01:
- [x] **PlatformIO Core**

### Configuration

**Operating system**: any

**PlatformIO Version**: any

### Description of problem

Initially described in: https://community.platformio.org/t/pio-home-security-issue/12196

When `platformio home` is run by a user, PIO has unlimited access to the user's home directory.

PIO home on `http://127.0.0.1:8008` exposes the home contents:
- to any other local user
- to any (potentially malicious) website that can connect to localhost (e.g. using javascript and websockets)
- to any (potentially malicious) PIO library

#### Suggested mitigations

On Linux the process can be sandboxed by either:
- Ship a firejail profile and start platformio under firejail
- Run platformio as a system process using systemd sandboxing capabilities
- Implement native sandboxing

The sandboxing is meant to limit which paths the process (and its children) can read and write from.

ivankravets, 2020-03-13:
What is a multiplatform solution for this issue? Could you provide a PR? Thanks in advance!

ivankravets, 2020-05-05:
Thanks for the report. We will generate a secure token on a server side and use it for clients.

closed 2021-01-25 by ivankravets

FedericoCeratto, 2021-01-26:
@ivankravets the token provides a basic form of authentication/authorization but cannot not provide any protection to the user's home directory from a platformio instance that is running untrusted code or has been otherwise compromised.
Can you please reopen the bug report? Thanks.

ivankravets, 2021-01-26:
> but cannot not provide any protection to the user's home directory from a platformio instance

Have you tried the latest development version? The upcoming PlatformIO IDE 2.1 will use a unique session ID for each session. So, you will not be able to connect to the existing session.

FedericoCeratto, 2021-01-26:
Yes, hence my answer. The unique session ID provides only authentication/authorization and is besides the point of sandboxing. PlatformIO downloads and executes a lot of code and tools and communicates with many devices, leading to  large attack surface.

ivankravets, 2021-01-26:
So, this is a different story. In this case, you can use Docker or similar virtualization platforms and connect your hardware to them. 

We agree, we had a security issue with PIO Home but it is now resolved.

commit 0ed99b76 (2021-01-25): Added a new ``--session-id`` option to `pio home` // Resolve #3397