platformio/platformio-core #3398: libArchive option in library.json not respected

maxgerhardt, 2020-03-02:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64 

**PlatformIO Version** (`platformio --version`): `PlatformIO, version 4.2.2a1`

### Description of problem

The FreeRTOS for STM32Duino library uses the `libArchive: false` option its [library.json](https://github.com/stm32duino/STM32FreeRTOS/blob/master/library.json#L12-L14) file to prevent being archived and thus not getting to hook the weak `SVCHandler` call which is needed by FreeRTOS to execute the task-switches in a privilged mode. The library is still being built as an archive anyways though and thus execution of the resulting firmware fails (broken `vTaskDelay`, semaphores etc)

The archive command is 

```
arm-none-eabi-ar rc ".pio\build\bluepill_f103c8_128k\lib7b0\libSTM32duino FreeRTOS_ID2093.a" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\FreeRTOS\Source\croutine.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\FreeRTOS\Source\event_groups.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\FreeRTOS\Source\list.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\FreeRTOS\Source\queue.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\FreeRTOS\Source\stream_buffer.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\FreeRTOS\Source\tasks.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\FreeRTOS\Source\timers.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\STM32FreeRTOS.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\cmsis_os.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\heap.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\mpu_wrappers.c.o" ".pio\build\bluepill_f103c8_128k\lib7b0\STM32duino FreeRTOS_ID2093\port.c.o"
arm-none-eabi-ranlib ".pio\build\bluepill_f103c8_128k\lib7b0\libSTM32duino FreeRTOS_ID2093.a"
```

However, by adding `lib_archive = no` the desired behaviour is restored and the final linker command contains all object files of the project 

```
".pio/build/bluepill_f103c8_128k/FrameworkArduinoVariant/PeripheralPins.c.o" ".pio/build/bluepill_f103c8_128k/FrameworkArduinoVariant/variant.cpp.o" 
..
".pio/build/bluepill_f103c8_128k/lib7b0/STM32duino FreeRTOS_ID2093/port.c.o" ".pio/build/bluepill_f103c8_128k/src/main.cpp.o"
```  

Shouldn't the behavior be the same as when the library.json says `libArchive: false`? This prevents these libraries from correctly building and executing, and needing special options in the user's `platformio.ini`. 

#### Steps to Reproduce

1. Create a new project `pio init -b bluepill_f103c8_128k` 
2. Use the `platformio.ini` from below
3. Build and upload firmware to bluepill board

### Actual Results

No blinkink LED (PC13).

### Expected Results

Blinking LED. Works when uncommenting `lib_archive = no`.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:bluepill_f103c8_128k]
board = bluepill_f103c8_128k
platform = ststm32
framework = arduino
upload_protocol = stlink
lib_deps = STM32duino FreeRTOS	
;lib_archive = no
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
/*
 * Example to demonstrate thread definition, semaphores, and thread sleep.
 */
#include <STM32FreeRTOS.h>

// Define the LED pin is attached
const uint8_t LED_PIN = LED_BUILTIN;

// Declare a semaphore handle.
SemaphoreHandle_t sem;
//------------------------------------------------------------------------------
/*
 * Thread 1, turn the LED off when signalled by thread 2.
 */
// Declare the thread function for thread 1.
static void Thread1(void* arg) {
  UNUSED(arg);
  while (1) {

    // Wait for signal from thread 2.
    xSemaphoreTake(sem, portMAX_DELAY);

    // Turn LED off.
    digitalWrite(LED_PIN, LOW);
  }
}
//------------------------------------------------------------------------------
/*
 * Thread 2, turn the LED on and signal thread 1 to turn the LED off.
 */
// Declare the thread function for thread 2.
static void Thread2(void* arg) {
  UNUSED(arg);
  pinMode(LED_PIN, OUTPUT);

  while (1) {
    // Turn LED on.
    digitalWrite(LED_PIN, HIGH);

    // Sleep for 200 milliseconds.
    vTaskDelay((200L * configTICK_RATE_HZ) / 1000L);

    // Signal thread 1 to turn LED off.
    xSemaphoreGive(sem);

    // Sleep for 200 milliseconds.
    vTaskDelay((200L * configTICK_RATE_HZ) / 1000L);
  }
}
//------------------------------------------------------------------------------
void setup() {
  portBASE_TYPE s1, s2;

  Serial.begin(9600);

  // initialize semaphore
  sem = xSemaphoreCreateCounting(1, 0);

  // create task at priority two
  s1 = xTaskCreate(Thread1, NULL, configMINIMAL_STACK_SIZE, NULL, 2, NULL);

  // create task at priority one
  s2 = xTaskCreate(Thread2, NULL, configMINIMAL_STACK_SIZE, NULL, 1, NULL);

  // check for creation errors
  if (sem== NULL || s1 != pdPASS || s2 != pdPASS ) {
    Serial.println(F("Creation problem"));
    while(1);
  }

  // start scheduler
  vTaskStartScheduler();
  Serial.println("Insufficient RAM");
  while(1);
}

//------------------------------------------------------------------------------
// WARNING idle loop has a very small stack (configMINIMAL_STACK_SIZE)
// loop must never block
void loop() {
  // Not used.
}
```

### Additional info

ivankravets, 2020-03-04:
This is a correct behavior. If you disable archiving per library, we will not archive it. But! The framework will be archived. However, if you disable archiving globally per project, we will disable archiving for frameworks and libraries.

Is this bad behaviour? :(

maxgerhardt, 2020-03-05:
Has this always been the behaviour? Because otherwise I think the library's author would have noticed that a basic blinky isn't working and would have written something to make people add `lib_archive = no` to their `platformio.ini`. I'll retest with a PIO 3.x version to check.

> Is this bad behaviour? :( 
Not per-se but then the library cannot specify that needed option for the whole project to make it work.

ivankravets, 2020-03-07:
> Has this always been the behaviour?

Yes, original behavior sine we introduced this configuration option.

kzyapkov, 2020-03-15:
> > Has this always been the behaviour?
> 
> Yes, original behavior sine we introduced this configuration option.

Doesn't this mean that https://github.com/platformio/platform-ststm32/issues/237 / https://github.com/stm32duino/STM32FreeRTOS/issues/17 were never really fixed?

valeros, 2020-03-15:
Hi @maxgerhardt @kzyapkov I'm not completely sure when this behavior was introduced, but I cannot reproduce the issue with the latest `platformio-core`. It looks like it was fixed in https://github.com/platformio/platformio-core/commit/59e1c8872680bd2c8af01471f3351707228f2c2c . Please upgrade via `pio upgrade --dev` and try again. Thanks!

closed 2020-03-15 by ivankravets

kzyapkov, 2020-03-16:
`pio upgrade --dev` broke it even with `lib_archive = no` in platformio.ini. I could not figure out how to rever to to the release, so after a complete reinstall of all platformio components it works again. This is my config:

```
[env:nucleo_l476rg]
platform = ststm32
board = nucleo_l476rg
framework = arduino
monitor_speed = 115200
lib_archive = no
build_flags =
  -Wl,-u,_printf_float,-u,_scanf_float

lib_deps =
  Adafruit BMP280 Library
  Adafruit TSL2561
  Adafruit Unified Sensor
  STM32duino FreeRTOS
  STM32duino HTS221
```

ivankravets, 2020-03-16:
1. What is your current PlatformIO Core version? `pio --version`
2. What is a version of `ststm32` dev-platform? `pio platform show ststm32`

kzyapkov, 2020-03-16:
PlatformIO, version 4.2.1

output slightly clipped:
```
$ pio platform show ststm32
ststm32 ~ ST STM32
==================
...
Version: 6.0.0
Home: http://platformio.org/platforms/ststm32
Repository: https://github.com/platformio/platform-ststm32.git
Vendor: http://www.st.com/web/en/catalog/mmc/FM141/SC1169?sc=stm32
License: Apache-2.0
Frameworks: arduino, cmsis, libopencm3, mbed, spl, stm32cube, zephyr
```

I did `rm -rf ~/.platformio`, removed the VSCode extention and reinstalled all components anew, so all versions should be "latest stable".

ivankravets, 2020-03-16:
It looks that you have outdated ST STM32 dev-platform. That was the problem.

pat1, 2020-03-19:
do not work for me too

```
pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 4.3.0
Release notes: https://docs.platformio.org/en/latest/history.html
```

```
pio --version
PlatformIO, version 4.3.0
```

```
pio platform show ststm32
ststm32 ~ ST STM32
==================
The STM32 family of 32-bit Flash MCUs based on the ARM Cortex-M processor is designed to offer new degrees of freedom to MCU users. It offers a 32-bit product range that combines very high performance, real-time capabilities, digital signal processing, and low-power, low-voltage operation, while maintaining full integration and ease of development.

Version: 6.0.0
Home: http://platformio.org/platforms/ststm32
Repository: https://github.com/platformio/platform-ststm32.git
Vendor: http://www.st.com/web/en/catalog/mmc/FM141/SC1169?sc=stm32
License: Apache-2.0
Frameworks: arduino, cmsis, libopencm3, mbed, spl, stm32cube, zephyr
```

```
pio platform update
Platform Atmel AVR
--------
Updating atmelavr                        @ 2.0.0          [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.190710 [Up-to-date]
Updating framework-arduino-avr           @ 5.0.0          [Up-to-date]
Updating tool-avrdude                    @ 1.60300.190424 [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.3.3          [Up-to-date]
Updating toolchain-xtensa                @ 2.40802.191122 [Up-to-date]
Updating framework-arduinoespressif8266  @ 3.20603.200130 [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20800.0      [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]

Platform ST STM32
--------
Updating ststm32                         @ 6.0.0          [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
Updating framework-arduinoststm32        @ 3.10700.191028 [Up-to-date]
Updating tool-stm32duino                 @ 1.0.1          [Up-to-date]
Updating tool-openocd                    @ 2.1000.190707  [Up-to-date]
Updating tool-dfuutil                    @ 1.9.200310     [Up-to-date]
```

pat1, 2020-03-20:
cat platformio.ini
```
[env]
framework = arduino
lib_extra_dirs = ../../../arduino/sketchbook/libraries/
monitor_speed = 115200

[env:nucleo_l476rg]
platform = ststm32
board = nucleo_l476rg
build_flags = -DARDUINO_ARCH_STM32  -std=c++11 -fexceptions
lib_ignore =
   ArduinoSTL
   FreeRTOS
lib_archive = no
```

ivankravets, 2020-03-20:
Sorry for the issue. We have fixed in https://github.com/platformio/platformio-core/commit/d32312e7386c1e59c7c2f841f1db1c7a60364268

Could you re-test with `pio upgrade --dev`?

pat1, 2020-03-20:
Now is OK
Thanks!

ivankravets, 2020-03-20:
Thanks! We will release PIO Core 4.3.1 soon. Thanks for the report!

ivankravets, 2020-03-20:
PlatformIO Core 4.3.1 is out! Please upgrade via `pio upgrade --dev`.