platformio/platformio-core #3401: Integrate project load and debug to Renode

carlosedp, 2020-03-04:
What kind of issue is this?

- [x] **Feature Request**.

What is Renode?

[Renode](https://renode.io/) is an open source software development framework with commercial support from Antmicro that lets you develop, debug and test multi-node device systems reliably, scalably and effectively. 

Renode is a fantastic tool to simulate applications on multiple boards. As an example, it supports boards from STMicro and some RISC-V boards like SiFive HiFive1, Kendryte K210 and many more. 

Full board list: https://renode.readthedocs.io/en/latest/introduction/supported-boards.html

This integration would be the best of both worlds where one can develop in PlatformIO and debug/simulate on Renode without requiring hardware. Renode is available for Windows, Mac and Linux just adjusting the execution path.

I've started testing the available options and identified Renode can be started as a Telnet server (over a choosen port) and issued commands to it.

As an example to run the HiFive 1 ZephyrOS Hello World project, I did (on MacOS):

1. Open Renode with Telnet option: `/Applications/Renode.app/Contents/MacOS/macos_run.command -P 1234`
2. Connected to the Renode Telnet server: `telnet 1234`
3. Loaded the HiFive1 board script: `include @scripts/single-node/sifive_fe310.resc`
4. Started Renode remote debugger port: `machine StartGdbServer 3333 True` 
5. Loaded my "Hello World" ELF binary built on PlatformIO: `sysbus LoadELF @/Users/cdepaula/Documents/PlatformIO/Projects/200303-171147-zephyr-hello-world/.pio/build/hifive1/firmware.elf`
6. Start simulation with: `start`

[image]

To start over, issue the command `Clear` on Renode console.

I successfully integrated the PlatformIO debugger to an existing instance of Renode by adding `debug_port = localhost:3333` to `platformio.ini` on my project. Added a test breakpoint and interacted with the debugging on the simulation:

1. Open Renode with Telnet option: `/Applications/Renode.app/Contents/MacOS/macos_run.command -P 1234`
2. Connected to the Renode Telnet server: `telnet 1234`
3. Loaded the HiFive1 board script: `include @scripts/single-node/sifive_fe310.resc`
4. Started Renode remote debugger port: `machine StartGdbServer 3333 True` 

From there, I clicked the debugger "Play" and PlatformIO built and loaded my application into Renode.

[image]

Continued the breakpoint:

[image]

I just was not able to "Reset" and start over via debugging or software without issuing the `Clear` command on Renode and loading everything from start (it's fast though).

I'm wiling to help writing this support if I can get some tips.

carlosedp, 2020-03-04:
Cc. @PiotrZierhoffer @pdp7 @pgielda @mgielda
Hope this drives interest :)

carlosedp, 2020-03-04:
Currently the `macos_run.command` has a small [bug](https://github.com/renode/renode/issues/79) that will be fixed soon.

carlosedp, 2020-03-10:
I've successfully made it work with no changes to PlatformIO Core, just by configuring the right bits on the project `platformio.ini`.

On Mac, I've installed Renode and made a link from `` to `/usr/local/bin/renode` to have it in the Path. This still requires the fix from the issue linked on previous comment.

My `platformio.ini` file is:

```ini
[env:hifive1]
platform = sifive
framework = zephyr
board = hifive1
monitor_speed = 115200
# Settings to integrate with Renode upload and debug
upload_command = renode -e "include @scripts/single-node/sifive_fe310.resc" -e "machine StartGdbServer 3333 True" -e "sysbus LoadELF @$SOURCE" -e "start"
debug_tool = custom
debug_port = localhost:3333
debug_server = renode
    -e
    include @scripts/single-node/sifive_fe310.resc
    -e
    machine StartGdbServer 3333 True
debug_extra_cmds =
    monitor start
```

And the integration for both running/uploading and also debugging worked perfectly as seen below:

[image]

ivankravets, 2020-03-10:
@carlosedp thanks for the great research! It can be easy integrated in PlatformIO. It will look for you as:
```
[env:hifive1]
platform = sifive
framework = zephyr
board = hifive1
monitor_speed = 115200
upload_tool = renode
debug_tool = renode
```

We will back soon to this issue!

carlosedp, 2020-03-10:
Nice thing is that Renode has support for more boards that are also supported by PlatformIO: https://renode.readthedocs.io/en/latest/introduction/supported-boards.html

carlosedp, 2020-03-10:
Hey @ivankravets, I'm still fine-tuning the parameters and let you know.
I have the serial console working too.

mgielda, 2020-03-11:
Hi @carlosedp glad to see this! I already had an integration with pio written up internally and was thinking how to make this public, so glad you went along and did this ;)

We were busy with the Renode 1.9 relase which just went out this week - https://github.com/renode/renode/releases/tag/v1.9.0 - but now that it's past, happy to support this effort - @PiotrZierhoffer bringing to your attention.

ivankravets, 2020-03-11:
Hi @mgielda ,

The only thing which we need for PlatformIO Core is a default GDB init config. See examples:
- https://github.com/platformio/platformio-core/blob/develop/platformio/commands/debug/initcfgs.py

The rest integration work should be done on the dev-platform side (https://github.com/platformio/platform-sifive), if we mean SiFive. We will help with this. So, what should be `GDB_RENODE_INIT_CONFIG`?

In the example above which works for @carlosedp if was:
```python
GDB_RENODE_INIT_CONFIG = """
define pio_reset_halt_target
    monitor reset halt
end

define pio_reset_run_target
    monitor reset
end

target extended-remote $DEBUG_PORT
monitor init
$LOAD_CMDS
pio_reset_halt_target
$INIT_BREAK
monitor start
"""
```

We plan to release PIO Core 4.3.0 next week. So, it would be great to have support for Renode.

carlosedp, 2020-03-11:
In my config, I've been able to both build/upload and debug. Also connect to the simulation over console (via telnet) with:

```ini
[env:hifive1]
platform = sifive
board = hifive1
framework = zephyr
## ----- Settings below are for Antmicro Renode integration ----- ##
# Monitor port for Renode integration
monitor_port = socket://localhost:1234
# Upload settings for Renode integration
upload_command = renode $UPLOAD_FLAGS
upload_flags =
    -e include @scripts/single-node/sifive_fe310.resc
    -e machine StartGdbServer 3333 True
    -e emulation CreateServerSocketTerminal 1234 \"externalUART\"
    -e connector Connect uart0 externalUART
    -e sysbus LoadELF @$SOURCE
    -e start
# Debug settings for Renode integration
debug_tool = custom
debug_port = localhost:3333
debug_server = renode
    -e include @scripts/single-node/sifive_fe310.resc
    -e machine StartGdbServer 3333 True
    -e emulation CreateServerSocketTerminal 1234 "externalUART"
    -e connector Connect uart0 externalUART
debug_extra_cmds =
    monitor start
```

Works perfectly. Do you folks think on any additional features that could be included?

carlosedp, 2020-03-11:
Guys, I'm seeing some problems debugging the console application on Renode.

On PlatformIO debug console I see the output below that I understand combined both PlatformIO and Renode outputs:

```
Processing hifive1 (platform: sifive; board: hifive1; framework: zephyr)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/sifive/hifive1.html
PLATFORM: SiFive 2.0.3 > HiFive1
HARDWARE: FE310 320MHz, 16KB RAM, 16MB Flash
DEBUG: Current (custom) On-board (ftdi, qemu)
PACKAGES:
 - framework-zephyr 2.20100.200220 (2.1.0)
 - framework-zephyr-civetweb 0.20100.190807 (2.1.0)
 - framework-zephyr-fatfs 0.20100.190522 (2.1.0)
 - framework-zephyr-libmetal 0.20100.190530 (2.1.0)
 - framework-zephyr-littlefs 0.20100.190811 (2.1.0)
 - framework-zephyr-lvgl 0.20100.190812 (2.1.0)
 - framework-zephyr-mbedtls 0.20100.191006 (2.1.0)
 - framework-zephyr-mcumgr 0.20100.190528 (2.1.0)
 - framework-zephyr-mipi-sys-t 0.20100.191024 (2.1.0)
 - framework-zephyr-nffs 0.20100.190523 (2.1.0)
 - framework-zephyr-open-amp 0.20100.190612 (2.1.0)
 - framework-zephyr-openthread 0.20100.191024 (2.1.0)
 - framework-zephyr-segger 0.20100.190421 (2.1.0)
 - framework-zephyr-tinycbor 0.20100.191016 (2.1.0)
 - tool-cmake 3.15.5
 - tool-dtc 1.4.7
 - tool-gperf 3.1.0
 - tool-ninja 1.9.0
 - toolchain-riscv 1.80300.190927 (8.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Checking size .pio/build/hifive1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  31.9% (used 5232 bytes from 16384 bytes)
Flash: [          ]   0.2% (used 25736 bytes from 16777216 bytes)
========================= [SUCCESS] Took 1.38 seconds =========================
Reading symbols from /Users/cdepaula/Documents/PlatformIO/Projects/Test/.pio/build/hifive1/firmware.elf...
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = custom
PlatformIO: Initializing remote target...
18:48:07.3470 [INFO] Loaded monitor commands from: /Applications/Renode.app/Contents/MacOS/./scripts/monitor.py
18:48:07.7633 [INFO] Including script: /Applications/Renode.app/Contents/MacOS/scripts/single-node/sifive_fe310.resc
18:48:07.7791 [INFO] System bus created.
18:48:08.2766 [DEBUG] Segment size automatically calculated to value 64KiB
18:48:08.2771 [DEBUG] Segment size automatically calculated to value 64KiB
18:48:08.2772 [DEBUG] Segment size automatically calculated to value 64KiB
18:48:08.3881 [DEBUG] Segment size automatically calculated to value 16MiB
18:48:08.6191 [INFO] sysbus: Loaded SVD: /var/folders/dv/d24tqkhn7zg55vj4ttrqq35m0000gp/T/renode-4287/178316e5-fb43-4b0c-83b4-935b1cf7a3e2.tmp. Name: FE310. Description: E31 CPU Coreplex, high-performance, 32-bit RV32IMAC core
      .
18:48:08.6460 [DEBUG] sysbus: Loading ELF /var/folders/dv/d24tqkhn7zg55vj4ttrqq35m0000gp/T/renode-4287/e70ddab6-e9ef-4eec-aea4-ebfc4fb40786.tmp.
18:48:08.6605 [INFO] sysbus: Loading segment of 20520 bytes length at 0x20400000.
18:48:08.6779 [DEBUG] sysbus: Segment loaded.
18:48:08.6780 [INFO] sysbus: Loading segment of 324 bytes length at 0x20405028.
18:48:08.6781 [DEBUG] sysbus: Segment loaded.
18:48:08.6782 [INFO] sysbus: Loading segment of 7944 bytes length at 0x80000148.
18:48:08.6788 [DEBUG] sysbus: Segment loaded.
18:48:08.7051 [INFO] cpu: Setting PC value to 0x20401BC4.
18:48:08.9003 [INFO] SiFive-FE310: GDB server with all CPUs started on port :3333
__start () at /Users/cdepaula/.platformio/packages/framework-zephyr/soc/riscv/riscv-privilege/common/vector.S:24
24		la t0, __irq_wrapper
No such command or device: init
Loading section vector, size 0x10 lma 0x20400000
Loading section exceptions, size 0x258 lma 0x20400010
[... 173 lines cut ...]
18:48:20.4882 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x64.
18:48:20.4882 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x65.
18:48:20.4882 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x66.
18:48:20.4883 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x67.
18:48:20.4883 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x68.
18:48:20.4883 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x69.
18:48:20.4883 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x6A.
18:48:20.4883 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x6B.
18:48:20.4883 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x6C.
18:48:20.4884 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x6D.
18:48:20.4884 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x6E.
18:48:20.4884 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x6F.
[... cut ...]
18:48:20.4884 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x72.
18:48:20.4885 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x73.
18:48:20.4885 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x74.
18:48:20.4885 [WARNING] uart0: Attempt to read Byte from peripheral that doesn't support Byte interface. Offset 0x75.
```

And it doesn't reach my breakpoint a couple lines below and stops here:

[image]

Any ideas?

carlosedp, 2020-03-11:
Found it! I had to add `--hide-log` to Renode debug command:

```
debug_server = renode
    --hide-log
    -e include @scripts/single-node/sifive_fe310.resc
    -e machine StartGdbServer 3333 True
    -e emulation CreateServerSocketTerminal 1234 "externalUART"
    -e connector Connect uart0 externalUART
```

Then it worked fine. I think @mgielda might need to point the debug commands since the logs show some commands as not existing like `init` and `reset`.

@ivankravets during the debug, if I hover on some variables or defines I get some errors like shown here:

[image]

pgielda, 2020-03-12:
As for "init" and "reset" those seem to be openocd commands, send via gdb using "monitor"command (its a transport to underlying server, not part of the GDB spec). Since its openocd-specific it does not do anything in Renode. You could potentially define functions like that in your renode script to get rid of the error and potentially do something when they're executed.

carlosedp, 2020-03-12:
I've successfully made the integration to use STM32F4 with Arduino:

[image]

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = arduino
## ----- Settings below are for Antmicro Renode integration ----- ##
# Monitor port for Renode integration
monitor_port = socket://localhost:1234
monitor_speed = 115200
# Upload settings for Renode integration
# Here we use uart2 that is the default UART attached in Arduino
# We have to open the UART2 window since the detfault is opening uart4
upload_command = renode $UPLOAD_FLAGS
upload_flags =
    -e include @scripts/single-node/stm32f4_discovery.resc
    -e machine StartGdbServer 3333 True
    -e emulation CreateServerSocketTerminal 1234 \"externalUART\"
    -e connector Connect uart2 externalUART
    -e showAnalyzer uart2
    -e sysbus LoadELF @$SOURCE
    -e start
# Debug settings for Renode integration
debug_tool = custom
debug_port = localhost:3333
debug_server = renode
    --hide-log
    -e include @scripts/single-node/stm32f4_discovery.resc
    -e machine StartGdbServer 3333 True
    -e emulation CreateServerSocketTerminal 1234 "externalUART"
    -e connector Connect uart2 externalUART
    -e showAnalyzer uart2
debug_extra_cmds =
    monitor start

```

@ivankravets, I understand we need a new GDB script without the `init` and `reset` commands as Peter said. 

On customizing the ini for Renode upload and debug, I think we need some parameters to be flexible depending on the board.

For example on the HiFive1, we use the script `@scripts/single-node/sifive_fe310.resc` and on STM32F4 `@scripts/single-node/stm32f4_discovery.resc`.

Also on HiFive1 the default UART is `uart0` and on STM32F4 it's `uart4`. Since the default STM32F4 script opens the uart4 console window by default, I've added the command `-e showAnalyzer uart2` to open uart2 in this case.

We could have customized the `stm32f4_discovery.resc` script, with correct uarts and etc. Since we are using the default scripts that come with Renode, I had to to this manually.

pgielda, 2020-03-12:
We are accidentaly working on a Monitor function called "alias" that will allow for custom commands to be easily defined "per script" or even "per machine". Stay tuned.

pgielda, 2020-03-16:
Hi @carlosedp -- just heads up that the alias feature is now in Renode master branch. It allows you to define anything as an alias, e.g.:
`alias reset "machine Reset"`
This will create a new command "reset" which will magically make gdb command "mon reset" work, etc.
Can also take arguments so e.g.
`alias print "echo"`
Will allow you to do:
`print "Hello World"`
in Renode monitor / scripts.
It supports scoping (can be global alias and per machine).

ivankravets, 2020-03-16:
Sorry for the delay from our side. We follow this thread, thanks for the ideas!

I tried yesterday Renode on my Mac and found a few issues:
1. The Renode's `macos_run.command` hardcoded to `/Library/Frameworks/Mono.framework`. I don't install 3rd party frameworks into the OS scope. It allows me to easily migrate between different macOS versions. I installed Mono framework with `brew install mono`.
2. Brew's `mono` does not work too because script uses `mono64`, there is no `mono64`. I tried to replace these lines in `macos_run.command` and it seems works now. As a minimum, `./macos_run.command -h`.

I've never used Mono/C#, I'm a noob here. Sorry.

------------

@pgielda please help with the next questions as for the [remode.zip](https://github.com/renode/renode/releases/download/v1.9.0/renode_1.9.0.zip) archive:

1) Are these binaries independent of an operating system?
2) Does it mean that the only 1 requirement for us will be Mono framework? So, if we have mono on Unix and .Net on Windows, we can use the same binaries from "bin"?
3) We have own CDN network with prebuilt packages. It helps developers to work on a project and to do not worry about packages. PlatformIO does this work automatically and maintains updates.. So, if there is a new Renode, we publish it into our Registry, and if it is compatible with SemVer requirement of dev-platform, developers receive this update automatically. No need to reinstall it again. The question: is it possible for us to have 1 `renode-x.y.z.tar.gz` which will contain universal mono's-bin folder and different launch scripts depending on OS type?

Thanks!

P.S: @carlosedp did great work! His work and article helped me a lot to realize that it should be a very easy integrated Renode into PlatformIO. Yes, it works now. The only what we want to do - to automate everything. The developer set in project `debug_tool = renode` and PlatformIO does the rest work automatically, starts Renode's GDB server, configure `renode` upload command, etc.

PiotrZierhoffer, 2020-03-17:
Hey @ivankravets 

About your issues: we never focused on Mono from brew as it used to be quite old and we needed a relatively fresh release. That's why our docs point to Mono website. Apparently, they do not have consistent packaging policies here.

We can definitely revisit that - using Mono from brew would be, in general, much easier. We will look at the startup scripts to make it work.

If you stumble upon similar problems in future, please feel free to let us know via [Renode issues](https://github.com/renode/renode/issues).

And regarding the questions you asked:

1. Renode is not platform independent yet - the "zip" from the release page is a Windows package (I will try to rename it to be more specific). There are also deb/rpm/pkg.tar.xz packages for Debian/Fedora/Arch and dmg for macOS. The reason for that is that some part of the code is in C or uses OS-specific features (like host-to-guest networking).

    In future we are planning on creating a cross-platform release as well, but it is not there yet.

2. On Windows you don't need Mono, you just need the .NET Framework 4.7, which is rather a common thing to have nowadays. Mono is required for macOS and Linux. There are some other Linux-specific requirements as well, listed here: https://github.com/renode/renode#installing-dependencies

3. As I said before, the cross-platform package is not there yet, but if there is interest we can investigate this further

ivankravets, 2020-03-21:
A few updates from PlatformIO side:

# 1. Renode package & CDN

We publish Renode package into our CDN (currently, we use @bintray). It is much faster than Github's Amazon S3. PlatformIO will automatically install Renode on the user machine into an isolated environment depending on the host machine.  We have own package manager and dev-platform declares own dependencies. For example, SiFive dev-platform can use Renode 1.9 where St STM32 Renode 1. PlatformIO resolves this automatically. 

We prepared packages and deployed them to https://bintray.com/beta/#/platformio/dl-packages/tool-renode?tab=files

# 2. GDB Configuration

We added [this GDB configuration](https://github.com/platformio/platformio-core/blob/develop/platformio/commands/debug/initcfgs.py#L127) for our PIO Unified Debugger. When user press "start debugging " or run from CLI `pio debug --interface gdb`, we automatically start Renode in the background and provide a bridge between IDEs and GDB clients.

Please check this configuration https://github.com/platformio/platformio-core/blob/develop/platformio/commands/debug/initcfgs.py#L127

If you will have any changes later, please inform us or do PR. A shortcut GDB command  named `pio_reset_run_target` is used by PlatformIO IDE or VSCode and CLion. It allows developers to do "hot-restart" of debugging session without restarting of GDB client and Debug server. It's very convenient for complex projects.

# 3. Renode's GDB server state

We had the problem with running Renode in the background. It's indeed our issue. The only we start GDB server, we try to connect to it. We have had problems with existing debugging server before which we support (J-Link, OpenOCD, pyOCD, etc.). Renode has different behavior. It starts the process and tries to download something from the Internet (my firewall reports me). So, and... the only when it downloaded dependencies, it starts GDB server. I'm not an expert here. Maybe it makes sense to start a socket server at the beginning and buffer income commands as openOCD and other tools do.

Nevertheless, it's not a blocker for us. We did a workaround and introduced support for [server.ready_pattern](https://github.com/platformio/platform-sifive/blob/develop/platform.py#L116). Currently, it is set to `GDB server with all CPUs started on port`. So, if you change this value in the next releases, please inform us. Thanks!

# 4. Renode logging

A simple quesiton - how to reduce logging? For example, to show everything below INFO+? Like, errors, warning, info, but not debug...

# 5. UART Console

I have not managed to get output in telnet from this example https://github.com/platformio/platform-sifive/tree/develop/examples/zephyr-synchronization

What is the correct configuration? I used this

```ini
[env:hifive1-revb]
platform = https://github.com/platformio/platform-sifive.git
framework = zephyr
board = hifive1-revb
build_type = debug
monitor_speed = 115200
debug_init_break = tbreak bg_thread_main
debug_tool = renode
```

```bash
pio debug --interface gdb -x .pioinit

# and 

nc localhost 1234
```
The configuration for Renode server is default by SiFive dev-platform => https://github.com/platformio/platform-sifive/blob/develop/platform.py#L105

------------

We have not released SiFive dev-platform yet. It would be good to close all the questions before. Then we can add support for Kendryte and STM32.

Thanks in advance!

carlosedp, 2020-03-23:
Hey Ivan, thanks for the release! The sifive-platform release required to support this right? Also the integration with Renode is only for Debug and not Upload?

To reduce the amount of logs, just pass `-e logLevel 3` for printing errors only. Check https://renode.readthedocs.io/en/latest/basic/logger.html.

To get the console, I've used:

```
monitor_port = socket://localhost:1234
monitor_speed = 115200
```

But faced the problem from https://github.com/platformio/platformio-core/issues/3421.

I've successfully added integration to the Kendryte K210 using FreeRTOS. One problem tho is that PIO tried to upload `firmware.bin` instead of `firmware.elf`. For this I needed to change the upload argument from `@$SOURCE` to `@$PROJECT_BUILD_DIR/$PIOENV/$PROGNAME".elf"`. I also had to add `upload_port = /dev/ttyUSB0` otherwise PIO didn't upload to Renode even by overriding by using `upload_command`. 

The complete config became:

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:sipeed-maix-go]
platform = kendryte210
board = sipeed-maix-go
framework = kendryte-freertos-sdk
## ----- Settings below are for Antmicro Renode integration ----- ##
# Monitor port for Renode integration
monitor_port = socket://localhost:1234
monitor_speed = 115200
# Upload settings for Renode integration
upload_port = /dev/ttyUSB0
upload_command = renode $UPLOAD_FLAGS
upload_flags =
    -e include @scripts/single-node/kendryte_k210.resc
    -e machine StartGdbServer 3333 True
    -e logLevel 3                       ; Loglevel = Errors
    -e emulation CreateServerSocketTerminal 1234 \"externalUART\"
    -e connector Connect uart externalUART
    ; -e sysbus LoadELF @$SOURCE        ; Does not work on K210-FreeRTOS
    -e sysbus LoadELF @$PROJECT_BUILD_DIR/$PIOENV/$PROGNAME".elf"
    -e start
# Debug settings for Renode integration
debug_tool = custom
debug_port = localhost:3333
debug_server = renode
    --hide-log
    -e include @scripts/single-node/kendryte_k210.resc
    -e machine StartGdbServer 3333 True
    -e logLevel 3           ;Loglevel = Errors
    -e emulation CreateServerSocketTerminal 1234 "externalUART"
    -e connector Connect uart externalUART
debug_extra_cmds =
    monitor start
```

carlosedp, 2020-04-24:
Hi @ivankravets, any news on this? Can I help with something?

ivankravets, 2020-04-27:
Sorry for the delay. We will back soon to this issue.

commit dbb9998f (2021-03-18): Refactor debugging configuration, add support for server_ready_pattern // Resolve #3401
closed 2021-03-18 by ivankravets