platformio/platformio-core #3403: Regression: "libArchive": false in library.json does not work.

JelleRoets, 2020-03-05:
What kind of issue is this?

- [ x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: osx 10.14.6

**PlatformIO Version** (`platformio --version`): 4.2.1

### Description of problem
According to documentation https://docs.platformio.org/en/latest/librarymanager/config.html#libarchive you should be able to disable archiving compiled source files of a library by setting the `build.libArchive` flag to false in the `library.json` file. This worked correctly in pio 4.1.0, however in 4.2.1 the library files are still archived although this flag is set to false.

#### Steps to Reproduce

1. Create a library with library.json and add "{"build": {"libArchive": false}}"
2. Compile your project

### Actual Results
The compiled library object files are archived into a .a file

### Expected Results
No archive step for this library

Example of library with libArchive flag set to false: https://github.com/JelleRoets/AtmelStart_PlatformIO/blob/master/library.json#L23

kzyapkov, 2020-03-20:
Broken for me as well, as reported in #3398 

`pip install -U "platformio<4.3"` installs `4.2.1` and works.

ivankravets, 2020-03-20:
Sorry, it is a little bit different issue. We have fixed in https://github.com/platformio/platformio-core/commit/d32312e7386c1e59c7c2f841f1db1c7a60364268

Could you re-test with `pio upgrade --dev`?

closed 2020-03-20 by ivankravets

ivankravets, 2020-03-20:
PlatformIO Core 4.3.1 is out! Please upgrade via `pio upgrade --dev`.

commit 59e1c887 (2020-03-05): Fixed an issue when ``"libArchive": false`` in "library.json" does not work // Resolve #3403