platformio/platformio-core #3406: Library Not Compiling - Not Finding Core Library

mriksman, 2020-03-06:
Hi,

- Created a new project. Framework `Arduino`, Platform `espressif8266`
- Added `lib_deps = HomeKit-ESP8266` to `platform.ini`
- Tried to compile. I get an error;
```
Compiling .pio\build\esp01_1m\libe5e\HomeKit-ESP8266_ID7045\base64.c.o
.pio\libdeps\esp01_1m\HomeKit-ESP8266_ID7045\src\arduino_homekit_server.cpp:2:25: fatal error: ESP8266WiFi.h: No such file or 
directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
```
But this file is part of the core framework/platform that is installed.

`library.json` for the library has;
```
  "frameworks": "arduino",
  "platforms": "espressif8266",
```
So it PlatformIO should be aware of what libraries to use...?

When I add `#include <arduino_homekit_server.h>` to `main.cpp`, it compiles fine....

ivankravets, 2020-03-06:
1. Did you test with the latest PIO Core 4.2.2-dev? `pio upgrade --dev`.
2. Could you provide a simple project to reproduce this issue?

mriksman, 2020-03-06:
I have just updated to 4.2.2a3. Issue still exists.

Simple project? It's a new project - no code.
```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Platformi.ini
```
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
build_flags =
  -Wformat
lib_deps = 
  HomeKit-ESP8266@^1.0.1
```

mriksman, 2020-03-06:
When I install the library via the Library Manager locally, and it (as of 4.2.2a3) appears in `{project}\.pio\libdeps\{env}\{library}`, when I compile, it doesn't try and compile the library, unless I add `#include "arduino_homekit_server.h"`, then it compiles OK.

But if I use lib_deps, and I **don't** use `#include "arduino_homekit_server.h"`, I still get this error;
```
Dependency Graph
|-- <HomeKit-ESP8266> 1.0.1
Building in release mode
Compiling .pio\build\esp01_1m\src\main.cpp.o
Generating LD script .pio\build\esp01_1m\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\esp01_1m\libe5e\HomeKit-ESP8266_ID7045\accessories.c.o
Compiling .pio\build\esp01_1m\libe5e\HomeKit-ESP8266_ID7045\arduino_homekit_server.cpp.o
Compiling .pio\build\esp01_1m\libe5e\HomeKit-ESP8266_ID7045\base64.c.o
.pio\libdeps\esp01_1m\HomeKit-ESP8266_ID7045\src\arduino_homekit_server.cpp:2:25: fatal error: ESP8266WiFi.h: No such file or 
directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
```

I'm curious.
1. Why can't this library not compile unless I use the `#include`?
2. Why when I locally install the project, it doesn't attempt to build the library, but if I use `lib_deps`, it does try and build the library? It doesn't seem consistent. The library resides in the same path using either method.

mriksman, 2020-03-06:
I think 4.2.2a3 has created a new problem. After ONE attempted build, I get the following;
```
> Executing task: C:\Users\Michael Riksman\.platformio\penv\Scripts\platformio.exe run <

Can not remove temporary directory `C:\Users\Michael Riksman\OneDrive\Documents\PlatformIO\Projects\ESP-HomeKit-Button\.pio\build`. Please remove it manually to avoid build issues
Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\Michael Riksman\\OneDrive\\Documents\\PlatformIO\\Projects\\ESP-HomeKit-Button\\.pio\\build\\esp01_1m':
  File "C:\Users\Michael Riksman\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 143:
    makedirs(env.subst("$BUILD_DIR"))
  File "C:\Users\Michael Riksman\AppData\Local\Programs\Python\Python38-32\lib\os.py", line 223:
    mkdir(name, mode)
```

I can't even get access to my own folder using all the usual Windows tricks.

[image]

I have to close VSCode, and the folder disappears by itself.

ivankravets, 2020-03-06:
Could you create a project outside OneDrive fodler? We just use native Python interfce to work with your file system. You OS, antivirus tools, permissions can make problem for PlatformIO.

ivankravets, 2020-03-06:
Please note that you don't need to have administrative /root permissions to work with PlatformIO. Everything should work from your current users with lowest permissions.

mriksman, 2020-03-06:
> Could you create a project outside OneDrive fodler? We just use native Python interfce to work with your file system. You OS, antivirus tools, permissions can make problem for PlatformIO.

Yeah this works much better. Could you please investigate why OneDrive and PlatformIO is not playing well together? I'd rather the project files stored their; like Arduino does. Like any other app does. PlatformIO must be upsetting OneDrive somehow. Arduino has never had these issues.

> When I install the library via the Library Manager locally, and it (as of 4.2.2a3) appears in `{project}\.pio\libdeps\{env}\{library}`, when I compile, it doesn't try and compile the library, unless [...]

Still hasn't solved this odd behaviour...

valeros, 2020-03-09:
Hi @mriksman ! I wasn't able to reproduce the issue with `OneDrive`, it works just fine in both CLI and IDE modes, but I suspect that a whitespace symbol in your user name might be the culprit. 

As for libraries,
>  When I install the library via the Library Manager locally, and it (as of 4.2.2a3) appears in {project}\.pio\libdeps\{env}\{library}, when I compile, it doesn't try and compile the library, unless I add #include "arduino_homekit_server.h", then it compiles OK.

I'd say that's expected behavior. PlatformIO has a special tool called `Library Dependency Finder` that automatically analyzes project source files and prepares library dependency tree according to used libraries (basically include commands). Since you haven't used any includes from `HomeKit-ESP8266` library, PlatformIO won't build it even if it's installed locally. But, if you explicitly specify this library in `lib_deps` field, you force PlatformIO build system to compile this library. In this case PlatformIO doesn't know what files to analyze for dependencies in `HomeKit-ESP8266` folder ([the default mode for LDF is `chain`](https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode) that only follows nested includes).

So I see here two solutions here: you either globally specify `lib_ldf_mode = deep` in your `platformio.ini` file or specify the same but in the  `HomeKit-ESP8266` library.json file.

mriksman, 2020-03-09:
Perhaps. I am having issues here as [well ](https://github.com/platformio/platform-espressif8266/issues/198) with .platformio being in a path with a white space. Any idea how to move .platformio to a new location?

Second part; makes sense. It's fine once I #include the .h file; I was more curious as to the different behaviour.

Before I close this - do you know of any tutorial on converting a CMakeLists to platformio.ini (or should it be library.json...?)

valeros, 2020-03-09:
@mriksman 
> Before I close this - do you know of any tutorial on converting a CMakeLists to platformio.ini (or should it be library.json...?)

I'm not aware of any tutorials on this topic. Which file to use it's pretty straightforward, if this `CMakeLists.txt` relates to a library then the `library.json` is the right choice, otherwise  you need to modify your `platformio.ini` (build flags, include dirs, etc).

mriksman, 2020-03-09:
Thanks. Yeah I guess it's considered a Library; [esp-homekit](https://github.com/maximkulkin/esp-homekit). It has a CmakeLists.txt  and a Makefile.projbuild/component.mk. I believe CMakeLists is CMake which generates the Makefile.projbuild/component.mk files? And CmakeLists is what the new ESP-IDF system now uses, whilst legacy used these other Make files (Makefile.projbuild/component.mk).

So I just need to convert CMakeLists into library.json, and ignore those other files...?

valeros, 2020-03-09:
What framework do you plan to use with that library? If I'm not mistaken, porting `CMakeLists.txt` should be enough, but I see that this library can be configured, so probably `Makefile.projbuild` file is also should be taken into account.

mriksman, 2020-03-09:
That's a fine question. I need to port this into something that works with ESP32 _and_ ESP8266. esp-homekit supports ESP_OPEN_RTOS and ESP-IDF. Unfortunately, Arduino (where I am coming from) is ESP8266 NONOS SDK - hence my move to PlatformIO. But now I find even ESP8266 RTOS SDK's is only v1.5 in PlatformIO, so I'm not sure how compatible it will be with the ESP-IDF (which I believe ESP8266 RTOS SDK 3.x and higher more closely resembles). I guess I won't know until I try.

The ESP32 version I'll also need Arduino library support.

So I guess it will need to support;
arduino, esp32
esp8266

Make sense?

valeros, 2020-03-09:
That's a complex task, there is [an issue](https://github.com/maximkulkin/esp-homekit/issues/32) about porting this library to Arduino, but looks like the maintainer decided that it doesn't worth it. Anyway, I'm closing this issue since it looks like we settled the issue and the conversation is getting a little off-topic.

closed 2020-03-09 by valeros