platformio/platformio-core #3409: Cannot change projects_dir with CLI

thedanbob, 2020-03-08:
What kind of issue is this?

- [✓] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Arch linux

**PlatformIO Version** (`platformio --version`): 4.2.1

### Description of problem
I cannot change the `projects_dir` setting via the CLI, only by directly editing `~/.platformio/appstate.json`.

#### Steps to Reproduce

1. `source ~/.platformio/penv/bin/activate`
2. `platformio settings set projects_dir <any path>`

### Actual Results
CLI returns `Error: Invalid value '<path>' for the setting 'projects_dir'`. Setting is not changed.

### Expected Results
`projects_dir` setting is changed.

### Additional info
Editing the setting in `appstate.json` and then running `platformio settings get` shows the correct value.

ivankravets, 2020-03-09:
it works for me. Does this folder exist?

thedanbob, 2020-03-09:
I just tested it again and can't reproduce the problem anymore. I can use the CLI to set `projects_dir` to any existing directory. Must have been a weird one-off.

closed 2020-03-09 by thedanbob