platformio/platformio-core #3411: UnicodeDecodeError: 'charmap' codec can't decode byte

Hofei90, 2020-03-12:
### Configuration

**Operating system**:
Windows_NT x64 10.0.17763

**PlatformIO Version** (`platformio --version`):
4.2.2b1
and
4.2.1

**Python Version**
3.8 32bit

### Description of problem
During compilation I get the Python Traceback UnicodeDecodeError: 'charmap' codec can't decode byte.

Full Message:
```
> Executing task in folder IR_Sender_WB: C:\Users\***\.platformio\penv\Scripts\platformio.exe run <

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.3.3 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
- framework-arduinoespressif8266 3.20603.200130 (2.6.3)
- tool-esptool 1.413.0 (4.13)
- tool-esptoolpy 1.20800.0 (2.8.0)
- toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <IRremoteESP8266> 2.7.4
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Users\***\AppData\Local\Programs\Python\Python38-32\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\***\AppData\Local\Programs\Python\Python38-32\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\***\.platformio\penv\lib\site-packages\platformio\proc.py", line 46, in run
    self.do_reading()
  File "c:\users\***\.platformio\penv\lib\site-packages\platformio\proc.py", line 69, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "c:\users\***\.platformio\penv\lib\site-packages\platformio\proc.py", line 69, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "C:\Users\***\AppData\Local\Programs\Python\Python38-32\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 167: character maps to <undefined>
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRac.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRrecv.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRsend.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRtext.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRtimer.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRutils.cpp.o
======================================================================================== [FAILED] Took 5.42 seconds ========================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
monitor_speed = 115200
```

### Additional info

On another PC with Python version 3.7 there are no problems compiling.
Here are the data of the PC where there is no problem:

Version: 1.43.0 (user setup)
Commit: 78a4c91400152c0f27ba4d363eb56d2835f9903a
Date: 2020-03-09T19:47:57.235Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.17763

PlatformIO: Core 4.2.1, Home 3.1.1

ivankravets, 2020-03-12:
1. Does both machines use the same user name? 
2. Do you have non-ASCII chars in the user name on a machine that fails? 
3. Do you use the same locale?

Hofei90, 2020-03-13:
>     1. Does both machines use the same user name?
no the machines have different user names
>     2. Do you have non-ASCII chars in the user name on a machine that fails?
no, the names contain only characters from [A-Z] [a-z] and char \ from the domain Name. But Domainname is same on both maschines. 
>     3. Do you use the same locale?
Do you mean the character encoding setting like UTF-8? In Visual Studio Code is it UTF-8 or you mean on the OS?

ivankravets, 2020-03-13:
Where did you get this Python 3.8? Could you try to uninstall it and install again the latest Python 3? Also, please remove .platformio/penv folder form user home folder. Restart VSCode.

Hofei90, 2020-03-14:
Hello,
all Python versions were downloaded from the official website. (https://www.python.org/downloads/)

What did I do:
1. uninstalled Python 3.8 and installed the last release with version 3.8.2 and deleted the penv folder and restarted VS Code
Result: Problem still exists
2. platformIO completely deleted from VS Code and reinstalled
Result Problem still exists
3. uninstall Python 3.8.2 and install Python 3.7.3, so that the versions are exactly the same as the working PC. delete the penv folder and restart VS Code.
Result: **Compilation works!**

Conclusion:
Python 3.8 causes the error posted above
In Python 3.7 these problems do not yet exist

Translated with www.DeepL.com/Translator (free version)

ivankravets, 2020-03-14:
@Hofei90 thanks for the info!

> Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld

@valeros what is an encoding of a common LD script? Could you reproduce this issue?

ivankravets, 2020-03-14:
@Hofei90 could you provide a simple project ot reproduce this issue?

Hofei90, 2020-03-14:
This is the project where the problems were.
Just removed the docs folder, .vscode, .git and .pio.
[IR_Sender_WB.zip](https://github.com/platformio/platformio-core/files/4332467/IR_Sender_WB.zip)