platformio/platformio-core #3412: LDF doesn't work as expected

kennycoder, 2020-03-12:
### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.2.1
### Description of problem

I experience problems with the LDF. It doesn't scan lib folder no matter what i try

The desired library `mylib.h` is shown in the dependency graph but using its elements leads to a compilation error (see below)

I am using this folder structure (pretty standard):

```
├── lib
│   ├── mylib
│   │   └── src
│   │   │   └── mylib.h
│   │   │   └── mylib.c
│   └── readme.txt
├── platformio.ini
└── src
    └── main.c
```

#### Steps to Reproduce

    1. Create files

    2. Paste content

    3. Build

### Actual Results

```
Dependency Graph
|-- <mylib>     
Building in release mode
Compiling .pio\build\esp32dev\esp-idf\src\main.c.o      
Generating LD script .pio\build\esp32dev\esp32_out.ld   
Generating partitions .pio\build\esp32dev\partitions.bin
src\main.c:1:10: fatal error: mylib.h: No such file or directory
```
### Expected Results

Successful build
### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
lib_deps = mylib
```

**Source file to reproduce issue:**
****`main.cpp`****

```c++
#include <mylib.h>

void app_main()
{
    return getValue();
}
```

****`mylib.h`****

```c++
#ifndef __MYLIB_H__
#define __MYLIB_H__

#define VALUE_ONE 1

int getValue();

#endif
```

****`mylib.c`****

```c++
#include "mylib.h"

int getValue() {
    return VALUE_ONE;
}
```
### Additional info

With previous versions of platformio this was not a problem.

valeros, 2020-03-12:
Hi @kennycoder ! I tried Win8, Win10 and everything works just fine. Could you please try to compile your project in verbose mode?

ivankravets, 2020-03-14:
Please reopen if you still need help

closed 2020-03-14 by ivankravets

saaymeen, 2020-03-31:
Can this possibly be reopened? I run into the exact issue with a clean install on MacOS and Linux. PIO version 4.3.1, `pio run -v` yields no useful additional information in my opinion.
Framework is ESP-IDF, and I tried various ESP32 boards.
The files are included if I include the full relative path from main.c, but there is no way the header gets included if you use `#include <mylib.h>`. Maybe @kennycoder found a solution meanwhile? Thanks

ivankravets, 2020-03-31:
@saaymeen  Please report to https://community.platformio.org/

saaymeen, 2020-04-27:
It was found [here](https://community.platformio.org/t/private-lib-not-found-since-latest-update/12687) that this indeed seems to be a bug. The current workaround seems to be registering the lib as IDF component, but I'm not sure whether this is the desired action required to get a library working with ESP_IDF.