platformio/platformio-core #3413: warnings from platformio Arduino Due ,when I #inlcude <Wire.h>

KiraVerSace, 2020-03-13:
Mac OS  The latest platformio, and in vscode .
Warning information:
Compiling .pio/build/due/lib586/Wire/Wire.cpp.o
In file included from /Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.cpp:25:0:
/Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.h: In constructor 'TwoWire::TwoWire(Twi*, void (*)(), void (*)())':
/Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.h:96:7: warning: 'TwoWire::twi' will be initialized after [-Wreorder]
  Twi *twi;
       ^~~
/Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.h:90:30: warning:   'void (* TwoWire::onBeginCallback)()' [-Wreorder]
  void (*onBeginCallback)(void);
                              ^
/Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.cpp:95:1: warning:   when initialized here [-Wreorder]
 TwoWire::TwoWire(Twi *_twi, void(*_beginCb)(void), void(*_endCb)(void)) :
 ^~~~~~~
In file included from /Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.cpp:25:0:
/Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.h:93:28: warning: 'TwoWire::onEndCallback' will be initialized after [-Wreorder]
  void (*onEndCallback)(void);
                            ^
/Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.h:72:10: warning:   'uint8_t TwoWire::rxBufferIndex' [-Wreorder]
  uint8_t rxBufferIndex;
          ^~~~~~~~~~~~~
/Users/kiraversace/.platformio/packages/framework-arduino-sam/libraries/Wire/src/Wire.cpp:95:1: warning:   when initialized here [-Wreorder]
 TwoWire::TwoWire(Twi *_twi, void(*_beginCb)(void), void(*_endCb)(void)) :

thankyou

KiraVerSace, 2020-03-13:
ok  i Fix the warning 
just change the code squence refrence:https://stackoverflow.com/questions/30364585/will-be-initialized-after-wreorder
like:
private:
	// TWI instance
	Twi *twi;

	// RX Buffer
	uint8_t rxBuffer[BUFFER_LENGTH];
	uint8_t rxBufferIndex;
	uint8_t rxBufferLength;

	// TX Buffer
	uint8_t txAddress;
	uint8_t txBuffer[BUFFER_LENGTH];
	uint8_t txBufferLength;

	// Service buffer
	uint8_t srvBuffer[BUFFER_LENGTH];
	uint8_t srvBufferIndex;
	uint8_t srvBufferLength;

	// TWI state
	enum TwoWireStatus {
		UNINITIALIZED,
		MASTER_IDLE,
		MASTER_SEND,
		MASTER_RECV,
		SLAVE_IDLE,
		SLAVE_RECV,
		SLAVE_SEND
	};
	TwoWireStatus status;

	// Callback user functions
	void (*onRequestCallback)(void);
	void (*onReceiveCallback)(int);

	// Called before initialization
	void (*onBeginCallback)(void);

	// Called after deinitialization
	void (*onEndCallback)(void);

	// TWI clock frequency
	static const uint32_t TWI_CLOCK = 100000;
	uint32_t twiClock;

	// Timeouts (
	static const uint32_t RECV_TIMEOUT = 100000;
	static const uint32_t XMIT_TIMEOUT = 100000;
};

[image]

ivankravets, 2020-03-13:
Great that you resolved this issue!

Happy coding with PlatformIO!

closed 2020-03-13 by ivankravets