platformio/platformio-core #3417: UnicodeDecodeError: 'charmap' codec can't decode byte

Hofei90, 2020-03-14:
### Configuration

**Operating system**:
Windows_NT x64 10.0.17763

**PlatformIO Version** (`platformio --version`):
4.2.2b1
and
4.2.1

**Python Version**
3.8 32bit

### Description of problem
During compilation I get the Python Traceback UnicodeDecodeError: 'charmap' codec can't decode byte.

Full Message:
```
> Executing task in folder IR_Sender_WB: C:\Users\***\.platformio\penv\Scripts\platformio.exe run <

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.3.3 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
- framework-arduinoespressif8266 3.20603.200130 (2.6.3)
- tool-esptool 1.413.0 (4.13)
- tool-esptoolpy 1.20800.0 (2.8.0)
- toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <IRremoteESP8266> 2.7.4
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Users\***\AppData\Local\Programs\Python\Python38-32\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\***\AppData\Local\Programs\Python\Python38-32\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\***\.platformio\penv\lib\site-packages\platformio\proc.py", line 46, in run
    self.do_reading()
  File "c:\users\***\.platformio\penv\lib\site-packages\platformio\proc.py", line 69, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "c:\users\***\.platformio\penv\lib\site-packages\platformio\proc.py", line 69, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "C:\Users\***\AppData\Local\Programs\Python\Python38-32\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 167: character maps to <undefined>
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRac.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRrecv.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRsend.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRtext.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRtimer.cpp.o
Compiling .pio\build\nodemcuv2\lib70a\IRremoteESP8266\IRutils.cpp.o
======================================================================================== [FAILED] Took 5.42 seconds ========================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
monitor_speed = 115200
```

### Additional info

On another PC with Python version 3.7 there are no problems compiling.
Here are the data of the PC where there is no problem:

Version: 1.43.0 (user setup)
Commit: 78a4c91400152c0f27ba4d363eb56d2835f9903a
Date: 2020-03-09T19:47:57.235Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.17763

PlatformIO: Core 4.2.1, Home 3.1.1

ivankravets, 2020-03-14:
> can you please make the path in your post unrecognizable

Sorry, done!

--------

So, I moved back this issue to PIO Core because it looks like a bug. What is a type of this drive? How can I create it?

Hofei90, 2020-03-15:
Hallo, 
es handelt sich hierbei um eine persönliche Netzwerkfreigabe innerhalb des lokalen Netzwerks. Weitere Details, welche Hardware und oder Software diese zu Verfügung stellt kann ich im Moment nicht sagen. Ich werde versuchen weitere Details herauszufinden

Hofei90, 2020-03-16:
Hello, the network drive is provided by a QNAP NAS. Unfortunately I cannot provide any further information about this

draeron, 2020-03-20:
i had this issue on windows (local disk). I solve this by changing the following line: 

https://github.com/platformio/platformio-core/blob/develop/platformio/proc.py#L33

from : 
```python
self._pipe_reader = os.fdopen(self._fd_read)
```

to :
```
self._pipe_reader = os.fdopen(self._fd_read, encoding='utf-8')
```

ivankravets, 2020-03-21:
@Hofei90 @draeron could you re-test with this PIO Core branch?

```
pip install -U https://github.com/platformio/platformio-core/archive/feature/issue-3417-unicodeerror-nas.zip
```

Does it work now?

draeron, 2020-03-21:
@ivankravets look like it solved the issue i had

ivankravets, 2020-03-21:
@draeron thanks!

@Hofei90  could you confirm?

ivankravets, 2020-03-21:
I merged into the development branch. Let see what people report.

ivankravets, 2020-04-28:
We reverted all changes, the fix for this issue broke PlatformIO for Windows when non-Latin chars are used in project path https://github.com/platformio/platformio-core/issues/3481

We will think about a better solution.

huexpub, 2020-05-08:
mark same error and changes files work...

ivankravets, 2020-05-08:
@huexpub do you also use network drive?

huexpub, 2020-05-17:
> @huexpub do you also use network drive?

local disk c:/administrators,  python3.8

ivankravets, 2020-05-17:
Could you try 1 solution? I saw in our forum https://community.platformio.org/ people downgrading Python to 3.7 and it helps them. Could you try this for the test? Uninstall Python 3.8 and install 3.7.  See https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

zekageri, 2020-06-26:
Reverted back to python 3.7 on windows 10, not solved my problem. Changing the proc.py file like drareon suggested does help.

ivankravets, 2020-06-26:
@zekageri do you also use network drive?

zekageri, 2020-06-26:
Nop. Local GitHub folder

ivankravets, 2020-06-26:
Please re-run `pio upgrade --dev` and provide info from 
```
pio system info
```

zekageri, 2020-06-29:
```
--------------------------  ----------------------------------------------------------
PlatformIO Core             4.4.0a4
Python                      3.7.0-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1250
PlatformIO Core Directory   C:\Users\Dr.Random\.platformio
PlatformIO Core Executable  C:\Users\Dr.Random\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\dr.random\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       2
Package Tools               19
--------------------------  ----------------------------------------------------------
```

stisa, 2021-01-12:
Hello, I'm having similar issues trying to compiler marlin, using vscode and platformio ide.

```python
Traceback (most recent call last):
  File "C:\Users\user.name\.platformio\python3\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\user.name\.platformio\python3\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\user.name\.platformio\penv\lib\site-packages\platformio\proc.py", line 47, in run
    self.do_reading()
  File "c:\users\user.name\.platformio\penv\lib\site-packages\platformio\proc.py", line 70, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "c:\users\user.name\.platformio\penv\lib\site-packages\platformio\proc.py", line 70, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "C:\Users\user.name\.platformio\python3\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 170: character maps to <undefined>
```
Output from `pio system info`:
```
--------------------------  --------------------------------------------------------------
PlatformIO Core             5.0.5a1
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\user.name\.platformio
PlatformIO Core Executable  C:\Users\user.name\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\user.name\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       3
Tools & Toolchains          9
--------------------------  --------------------------------------------------------------
```

Was this ever resolved?

ivankravets, 2021-01-12:
Do you have non-ascii chars in your account name?

stisa, 2021-01-12:
No, only a-z and a dot, like `name.surname`

ivankravets, 2021-01-12:
Does your FULL project path contain NON-ASCII chars?

stisa, 2021-01-12:
No, the full path should be: `L:\name.surname\Documents\Marlin-2.0.x`

ivankravets, 2021-01-12:
Could you try to move a project to C: disk? What is a type of L: drive?

stisa, 2021-01-12:
Moving the project to `C:` works, thanks!

I'm not sure what's the exact type of `L:`, it's a folder on a server, which then get mapped to act as my `home` folder, but the configuration is done by another department so I don't know more.

The only difference I can think of is that sometimes the path can become `\\{some ip address}\name.surname\Documents\Marlin-2.0.x` , but I doubt the `\\` at the beginning would cause this issue, right?

ivankravets, 2021-01-20:
Please re-test with the latest development version of PIO Core 

```
pio upgrade --dev
```

stisa, 2021-01-21:
did a quick test, seems to work, thanks!

YiffyToys, 2021-03-03:
I get the error message above in PlatformIO 5.1.0 but it should have been fixed in 5.0.5

kbfifi, 2021-03-22:
I'm at 5.1.1 and also getting this error. I AM using a network drive. Downgrade Python to 3.7 (How do I do that?) or other suggestion?

`PS X:\Arduino\libraries> pio system info
--------------------------  -----------------------------------------------------
PlatformIO Core             5.1.1
Python                      3.9.2-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\myname\.platformio
PlatformIO Core Executable  C:\Users\myname\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\myname\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          4`

ivankravets, 2021-03-22:
@kbfifi could you provide the error report?

claudobahn, 2021-03-22:
@ivankravets same here

* when I run `pio run` from `M:` (a mapped network drive)

> Building in release mode
> Compiling .pio\build\esp32dev\src\sketch.ino.cpp.o
> Generating partitions .pio\build\esp32dev\partitions.bin
> '\\SVR\sketch'
> CMD.EXE was started with the above path as the current directory.
> UNC paths are not supported.  Defaulting to Windows directory.
> '\\SVR\sketch'
> CMD.EXE was started with the above path as the current directory.
> UNC paths are not supported.  Defaulting to Windows directory.
> xtensa-esp32-elf-g++: error: src\sketch.ino.cpp: No such file or directory
> xtensa-esp32-elf-g++: fatal error: no input files
> compilation terminated.
> *** [.pio\build\esp32dev\src\sketch.ino.cpp.o] Error 1
> Traceback (most recent call last):
>   File "C:\Users\IEUser\.platformio\packages\framework-arduinoespressif32\tools\gen_esp32part.py", line 534, in <module>
>     main()
>   File "C:\Users\IEUser\.platformio\packages\framework-arduinoespressif32\tools\gen_esp32part.py", line 502, in main
>     os.makedirs(output_dir)
>   File "C:\Users\IEUser\AppData\Local\Programs\Python\Python39\lib\os.py", line 215, in makedirs
>     makedirs(head, exist_ok=exist_ok)
>   File "C:\Users\IEUser\AppData\Local\Programs\Python\Python39\lib\os.py", line 215, in makedirs
>     makedirs(head, exist_ok=exist_ok)
>   File "C:\Users\IEUser\AppData\Local\Programs\Python\Python39\lib\os.py", line 225, in makedirs
>     mkdir(name, mode)
> PermissionError: [WinError 5] Access is denied: 'C:\\Windows\\.pio'
> *** [.pio\build\esp32dev\partitions.bin] Error 1

ivankravets, 2021-03-23:
> PermissionError: [WinError 5] Access is denied: 'C:\Windows\.pio

This is not an issue with Python. You have a permission problem with your machine. Or, you use 3rd party antivirus tools that make problems for you.

claudobahn, 2021-03-23:
@ivankravets what seems to me is happening:
* I'm running this from `M:` (a mapped network drive
* pio then spits out a warning about UNC paths and switches its working directory to `c:\windows` which means it'll try to create `.pio` there:
> '\SVR\sketch'
> CMD.EXE was started with the above path as the current directory.
> UNC paths are not supported. Defaulting to Windows directory.
* Creating `c:\windows\.pio` fails because the `c:\windows` directory is not user writeable

Perhaps use the Windows temp directory instead for creating `.pio`?

kbfifi, 2021-03-23:
@ivankravets here's my error report:
`PIO Core Call Error: "Processing pro16MHzatmega328 (platform: atmelavr; board: pro16MHzatmega328; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nThere is a known issue with Python 3.8+ and mapped network drives on Windows.\r\nPlease downgrade Python to the latest 3.7. More details at:\r\nhttps://github.com/platformio/platformio-core/issues/3417\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/pro16MHzatmega328.html\r\nPLATFORM: Atmel AVR (3.2.0) > Arduino Pro or Pro Mini ATmega328 (5V, 16 MHz)\r\nHARDWARE: ATMEGA328P 16MHz, 2KB RAM, 30KB Flash\r\nDEBUG: Current (avr-stub) On-board (avr-stub, simavr)\r\nPACKAGES: \r\n - framework-arduino-avr 5.1.0 \r\n - toolchain-atmelavr 1.70300.191015 (7.3.0)\r\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\r\nLDF Modes: Finder ~ chain, Compatibility ~ soft\r\nFound 5 compatible libraries\r\nScanning dependencies...\r\nNo dependencies\r\nBuilding in debug mode\r\nCompiling .pio\\build\\pro16MHzatmega328\\src\\main.cpp.o\r\nArchiving .pio\\build\\pro16MHzatmega328\\libFrameworkArduinoVariant.a\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\CDC.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial0.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial1.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial2.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial3.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\IPAddress.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\PluggableUSB.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\Print.cpp.o\r\nCompiling .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\Stream.cpp.o\r\n\n\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\navr-g++: error: src\\main.cpp: No such file or directory\r\navr-g++: fatal error: no input files\r\ncompilation terminated.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\nc:\\users\\kurt\\.platformio\\packages\\toolchain-atmelavr\\bin\\../lib/gcc/avr/7.3.0/../../../../avr/bin/ar.exe: .pio\\build\\pro16MHzatmega328\\libFrameworkArduinoVariant.a: No such file or directory\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\n*** [.pio\\build\\pro16MHzatmega328\\src\\main.cpp.o] Error 1\r\n*** [.pio\\build\\pro16MHzatmega328\\libFrameworkArduinoVariant.a] Error 1\r\n'\\\\QNAPSTATION\\home\\Arduino\\libraries\\plaformio\\FakeIButton-1992'\r\nCMD.EXE was started with the above path as the current directory.\r\nUNC paths are not supported. Defaulting to Windows directory.\r\nAssembler messages:\r\nFatal error: can't create .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial0.cpp.o: No such file or directory\r\nAssembler messages:\r\nFatal error: can't create .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial.cpp.o: No such file or directory\r\nAssembler messages:\r\nFatal error: can't create .pio\\build\\pro16MHzatmega328\\FrameworkArduino\\PluggableUSB.cpp.o: No such file or directory\r\nAs [... cut ...] o\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial1.cpp.o] Error 1\r\n*** [.pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial3.cpp.o] Error 1\r\n*** [.pio\\build\\pro16MHzatmega328\\FrameworkArduino\\HardwareSerial.cpp.o] Error 1\r\n*** [.pio\\build\\pro16MHzatmega328\\FrameworkArduino\\IPAddress.cpp.o] Error 1\r\n*** [.pio\\build\\pro16MHzatmega328\\FrameworkArduino\\CDC.cpp.o] Error 1\r\n========================== [FAILED] Took 1.63 seconds =========================="`

ivankravets, 2021-03-24:
Could you try to force [PLATFORMIO_CORE_DIR](https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR) to `C:/.platformio`? Please set this environment variable. You might need to restart your machine. The only you restart VSCode, PlatformIO IDE/Core should use `C:/.platformio` and you can run a project from C: drive as well.

Does this workaround work?

kbfifi, 2021-03-24:
Not sure who you were asking. Tried it, makes no difference to me. I'm trying using a symbolic link for the  'Projects' sub folder (to the network mapped drive path) to get around the UNC path. But it looks like it's always translated to back to UNC again. Maybe I should downgrade portable Python to 3.7.x but I don't know how.

claudobahn, 2021-03-24:
@ivankravets I tried setting `PLATFORMIO_CORE_DIR` as you suggested, but the failure remained the same (though it did download all the tooling, again, successfully with the changed folder, so I know the setting was used).

That got me poking around a little bit more, so I changed `PLATFORMIO_WORKSPACE_DIR`. That took care of the build process not being able to create the `.pio` folder because of being unable to write to `c:\windows`, though of course the next problem is being unable to find `src` :) So just for fun I also tried to set `PLATFORMIO_LIB_DIR` and `PLATFORMIO_SRC_DIR`, although that would not be a practical solution with multiple projects, of course, but I wanted to see what happens. Unfortunately that wasn't enough, either.

Basically it comes down to what's described in this commit: https://github.com/platformio/platformio-core/commit/e31591a35e5bd630337ad8b8a9402f647d3d175f and unfortunately there's still no actual fix for that issue (other than downgrading Python or copying the project to the `C:` drive). I have been successfully working off a copy on the `C:` drive, but copying files from/to the mapped network drive is not a fun workflow.

From the commit comment:

> Starting with Python 3.8 paths to mapped network drives are resolved
> to their real path in the system, e.g.: "Z:\path" becomes "\\path" which
> causes weird errors in the default terminal with a message that UNC
> paths are not supported

Rotma, 2021-03-25:
Same problem here, started on a fresh installed win10 today! , download VScode , installed extension PlatformIO all releases from today.
Started project on a network drive and failed!
Same error as: @ivankravets 
See line 2/3 of his report
"There is a known issue with Python 3.8+ and mapped network drives on Windows.\r\n
Please downgrade Python to the latest 3.7. More details
https://github.com/platformio/platformio-core/issues/3417""

In the file "C:\Users\XXX\.platformio\python3\news.txt" you will find "Python 3.9.2 final"
and yes that version gives  problems with UNC paths (network drives).

Look it up on another computer in the network that was still working, that one has version "Python 3.7.7 final"
Made a copy of the complete C:\Users\XXX\.platformio\python3 directory from that computer and now its working.......

So the problem is defenitly the Python version!

The first post here says "Please uninstall all Pythons from the system and install the latest Python 3.7"
But the recent PIO install with has its own portable python in the .platformio dir and this version wil always be used!

Will try tomorrow if deleting the python version in the .platformio dir and installing genui python 3.7 will work...

kbfifi, 2021-03-25:
I'm new to PlatFormIO. Got it to work with using a local directory instead of a network drive. Really want to use the network drive. Unfortunately  cannot copy an old version. I tried to in stall 3.7.9 (https://sourceforge.net/projects/portable-python/files/Portable%20Python%203.7/) But PlatformIO didn't buy it

@ivankravets please let me know how to downgrade. I guess I need some additional steps using my first attempt.

Rotma, 2021-03-27:
After long fiddeling around with different python versions i give up... with older python verisons the PIO home screen will not work!! 
Looking further in the problem that cmd.exe is  not handling UNC names i found a very old KB

https://web.archive.org/web/20150518102450/https://support.microsoft.com/en-us/kb/156276

It was written for windows XP and NT but never the less its WORKING! under win10!
HKEY_CURRENT_USER\Software\Microsoft\Command Processor      <--- create this key "Command Processor"
add the value "DisableUNCCheck" REG_DWORD and set the value to 0 x 1 (Hex).
 
@kbfifi  , please let me know if its working for you as well

kbfifi, 2021-03-28:
Yesss! @Rotma Great find! Works a charm! Thanks a lot!

ivankravets, 2021-03-31:
@Rotma thanks that posted this solution! I updated the first comment and added screenshots.

claudobahn, 2021-03-31:
@Rotma thanks for finding that! Adding that registry key works great for me, as well. @ivankravets thanks for updating first post and the warning message in PIO.

jjbott, 2021-04-24:
For me it insists on using the (broken) version of python from %userprofile%\.platformio\penv\Scripts . Turning off "Platformio-ide: Use Builtin Python" seems to have to effect.

I ended up doing a "pip install platformio" under Python 3.7, and turning off the "Platformio-ide: Use Builtin PIOCore" setting.

ivankravets, 2021-04-24:
@jjbott have you tried the solution described above https://github.com/platformio/platformio-core/issues/3417#issue-581161485 ?

Rotma, 2021-04-24:
@ivankravets
#3417 is only a partial solution, compiling will work but intellisense is not working for network paths!

See mij comment in #3926

kbfifi, 2021-04-24:
@Rotma Same problem here. The problem can cause trouble with compiler errors as wel: 
- networkdrive referenced file (x:\path\file.c) with syntax problem is open in editor
-compile and click on the error
- file is opened again with full unc path \\hostname\path\file.c
- editing the latter one causes a warning about different versions being active. Which is correct

It workable by editing consistently from one of the open files but far from ideal

jjbott, 2021-04-24:
@ivankravets Yeah, I've done that which helped I think, but I still got compiler errors. And, I got the "There is a known issue with Python 3.8+" warning despite not having Python 3.8+ installed:

```
Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing melzi (platform: atmelavr; framework: arduino; board: sanguino_atmega1284p)
------------------------------------------------------------------------------------------------------------------------
There is a known issue with Python 3.8+ and mapped network drives on Windows.
Please downgrade Python to the latest 3.7. More details at:
https://github.com/platformio/platformio-core/issues/3417
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sanguino_atmega1284p.html
PLATFORM: Atmel AVR (3.3.0) > Sanguino ATmega1284p (16MHz)
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
FileNotFoundError: [WinError 161] The specified path is invalid: '\\\\networkdrive\\code\\Marlin\\https:\\github.com\\MarlinFirmware\\U8glib-HAL\\archive\\dev.zip':
  File "C:\Users\User\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
    ...
```

VB5000, 2021-05-06:
> 
> 
> @ivankravets Yeah, I've done that which helped I think, but I still got compiler errors. And, I got the "There is a known issue with Python 3.8+" warning despite not having Python 3.8+ installed. Here are the relevant logs:
> 
> ```
> Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
> Processing melzi (platform: atmelavr; framework: arduino; board: sanguino_atmega1284p)
> ------------------------------------------------------------------------------------------------------------------------
> There is a known issue with Python 3.8+ and mapped network drives on Windows.
> Please downgrade Python to the latest 3.7. More details at:
> https://github.com/platformio/platformio-core/issues/3417
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sanguino_atmega1284p.html
> PLATFORM: Atmel AVR (3.3.0) > Sanguino ATmega1284p (16MHz)
> HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
> DEBUG: Current (simavr) On-board (simavr)
> PACKAGES:
>  - framework-arduino-avr 5.1.0
>  - toolchain-atmelavr 1.70300.191015 (7.3.0)
> Converting Marlin.ino
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> FileNotFoundError: [WinError 161] The specified path is invalid: '\\\\networkdrive\\code\\Marlin\\https:\\github.com\\MarlinFirmware\\U8glib-HAL\\archive\\dev.zip':
>   File "C:\Users\User\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 177:
>     env.SConscript("$BUILD_SCRIPT")
>     ...
> ```

I ran into the same problems when compiling marlin from a network drive:

> [WinError 161] The specified path is invalid

The solution that worked for me was:
1. Install python 3.7.9
2. Delete the content of the \.platformio\python3 folder
3. Copy the content from AppData\Local\Programs\Python\Python37 into \.platformio\python3

Marlin compiled properly with no errors.  Hope this helps, its a very down and dirty solution but it works.

Bear in mind if you do anything to trigger a reinstall of the platformio extension, you will need to redo the steps above. It clearly proves the problem is with python.

JamesKR3D, 2021-06-07:
VB5000.  Thanks man.  Your solution worked for me whereas the other ones listed here didn't.

kbfifi, 2021-06-22:
@VB5000 Thanks for your solution it kinda works for me. However when using e.g. Platform->Quick access->Open for instance. It show a new tab "PIO Home" and Loading...
Eventually it crashes:

%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.57.1
  PIO IDE: v2.3.2
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 98, in cli
    home_url=home_url,
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\platformio\commands\home\run.py", line 98, in run_server
    log_level="warning",
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\uvicorn\main.py", line 386, in run
    server.run()
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\uvicorn\server.py", line 49, in run
    loop.run_until_complete(self.serve(sockets=sockets))
  File "C:\Users\kurt\.platformio\python3\lib\asyncio\base_events.py", line 587, in run_until_complete
    return future.result()
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\uvicorn\server.py", line 56, in serve
    config.load()
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\uvicorn\config.py", line 305, in load
    self.lifespan_class = import_from_string(LIFESPAN[self.lifespan])
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\uvicorn\importer.py", line 23, in import_from_string
    raise exc from None
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\uvicorn\importer.py", line 20, in import_from_string
    module = importlib.import_module(module_str)
  File "C:\Users\kurt\.platformio\python3\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\uvicorn\lifespan\on.py", line 6, in <module>
    from uvicorn._types import LifespanReceiveMessage, LifespanScope, LifespanSendMessage
  File "C:\Users\kurt\.platformio\penv\lib\site-packages\uvicorn\_types.py", line 5, in <module>
    from typing_extensions import Literal, TypedDict
ModuleNotFoundError: No module named 'typing_extensions'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\kurt\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\platformio-node-helpers\dist\index.js:1:5780
	at ChildProcess.a (c:\Users\kurt\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\platformio-node-helpers\dist\index.js:1:931270)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\kurt\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)

When using the original plaformio python3 folder it works again. @VB5000 Can you confirm the same problem? Or @ivankravets: Anything I can install to this 3.7.9 Phyton3 folder to fix this workaround?

VB5000, 2021-06-29:
> @VB5000 Thanks for your solution it kinda works for me. However when using e.g. Platform->Quick access->Open for instance. It show a new tab "PIO Home" and Loading...
> Eventually it crashes:
>
> %23 [...]

Hey @kbfifi , yes I should have mentioned this in my original post, the PIO Home tab doesn't work as it relies on the latest Python version. Personally, I don't use it so its a very acceptable trade off.

kbfifi, 2021-06-29:
@VB5000: Thanks for confirming. As I'm rather new to PlatformIO I like to see everything working. @ivankravets: Looks like the issue isn't solved on Python side very quickly. Would a temporary workaound implementation make sense?

Kilill, 2021-09-17:
Question: Why is even CMD.exe used ? powershell.exe should be able to work with UNC path's but setting COMSPEC to powershell still starts cmd.exe from somewhere....

closed 2021-11-12 by ivankravets

lmagyar, 2022-01-17:
Commit 4687665 caused another issue: #4155

richms, 2022-04-01:
Neither fix worked for me. The first one made it end up making the path within the mapped network drive, Seems it thinks that the space in the path that is mapped to the drive letter is a seperator because everything after that was created in the projects folder and had some things put into it as it was failing.

the second one seemed to have all my work undone by downloading python over what I replaced it with when I pressed build. 

Begrudgingly moved my project onto the local drive just so I could build it. This is a showstopper for the use of platformio for me as I rely on my files being on a nas so they are accessible on all computers

MarcusWolschon, 2022-04-01:
If it has problems with spaces in pathes, then clearly some parts use inproper escaping for pathes.

kbfifi, 2022-07-25:
I think this is a wonderful project with lots of potential, but as long there's no fix for this issue I won't use it.

ivankravets, 2023-04-20:
Hi there,

Sorry for the historical issue. It looks like we have finally fixed it. Please re-test with `pio upgrade --dev`.

Does it work now without any issues?

P.S.: Any Python 3.6 or above is OK now!

MarcusWolschon, 2023-04-20:
You have already closed it. So apparently tested is already complete.

[3 more replies]

commit f95230b8 (2020-03-21): Fixed UnicodeDecodeError on Windows when network drive (NAS) is used // Resolve #3417
commit 52b22b57 (2021-01-20): Fixed a "UnicodeDecodeError: 'utf-8' codec can't decode byte" // Resolve #3804 , Resolve #3417
commit 4687665f (2021-11-12): Improved support for projects located on a network share // Resolve #3417 , Resolve #3926 , Resolve #4102
commit 743fc8e6 (2023-04-20): Improved support for projects located on Windows network drives // Resolve #3417