platformio/platformio-core #3419: Ability to install libraries from library.json from the command line

jackjansen, 2020-03-16:
**Feature Request**

In the CI/CD script for my library with various dependencies listed in `library.json` I have to duplicate those dependencies as individual `platform lib install` commands.

It would be great if there was a command line option to have the library manager read a manifest file and install everything required from that manifest file. For example something like `platform lib install -f library.json`.

I presume the actual implementation may need a `--platform` and `--frameworks` option to limit what is installed.

ivankravets, 2020-03-16:
It should work, see docs https://docs.platformio.org/en/latest/core/userguide/lib/cmd_install.html

Have you tried `pio lib install file://<folder>`?

jackjansen, 2020-03-16:
Hmm. So that would be something like `pio lib install file://$PWD` in my Travis CI/CD script.... I guess it will also install the library I'm building itself (my original question was about only installing the _dependencies_), but that I can probably work around by removing the `--lib="."` from the subsequent `pio ci` commands. I'll give it a try.

closed 2020-03-17 by ivankravets