platformio/platformio-core #3421: Error on serial monitor console over Telnet

carlosedp, 2020-03-16:
What kind of issue is this?

- [x] **PlatformIO Core**.

### Configuration

PlatformIO on VSCode

**Operating system**:

MacOS Mojave

**PlatformIO Version** (`platformio --version`):

Home: 3.1.1
Core: 4.2.1 

### Description of problem

While trying to access the serial console on a simulated device via telnet, when I open the serial monitor, I see some garbled characters. After hitting enter some times, I see the console output.

My config:

```
[env:hifive1]
platform = sifive
board = hifive1
framework = zephyr
## ----- Settings below are for Antmicro Renode integration ----- ##
# Monitor port for Renode integration
monitor_port = socket://localhost:1234
monitor_speed = 115200
...
```

The project is: https://github.com/carlosedp/ZephyrConsoleEchoSIM

```
> Executing task: platformio device monitor <

Looking for advanced Serial Monitor with UI? Check http://bit.ly/pio-advanced-monitor
--- Miniterm on socket://localhost:1234  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��␀��␁��␃��"*** Booting Zephyr OS version 2.1.0  ***
Hello! I'm running Zephyr 2.1.0 on hifive1, a riscv32 board.

Enter a line finishing with Enter:
> 
```

#### Steps to Reproduce

1. Build and upload project linked below
2. Open the Serial Monitor
3. See the weird characters, hit enter multiple times to get the correct output.

### Actual Results

### Expected Results

Console to open correectly. I've successfully got the console by using the `telnet 127.0.0.1 1234` command.

ivankravets, 2020-03-16:
Could you try this?
```
platformio device monitor -p socket://localhost:1234 -f direct
```

carlosedp, 2020-03-16:
Similar problem, I get some weird chars and after a couple enters I get the output. Here you can see the monitor and a simple telnet connection:

```
❯ ./penv/bin/platformio device monitor -p socket://localhost:1234 -f direct
Looking for advanced Serial Monitor with UI? Check http://bit.ly/pio-advanced-monitor
--- Miniterm on socket://localhost:1234  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��������"*** Booting Zephyr OS version 2.1.0  ***
Hello! I'm running Zephyr 2.1.0 on hifive1, a riscv32 board.

Enter a line finishing with Enter:
>
--- exit ---
❯
❯ telnet localhost 1234
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.

Typed line:
Last char was: 0x0
> asd
Typed line: asd
Last char was: 0x64
```

ivankravets, 2020-03-16:
And 
```
device monitor -p socket://localhost:1234 -f debug
```

carlosedp, 2020-03-16:
Similar:

```
❯ ./penv/bin/platformio device monitor -p socket://localhost:1234 -f debug
Looking for advanced Serial Monitor with UI? Check http://bit.ly/pio-advanced-monitor
--- Miniterm on socket://localhost:1234  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
 [RX:'�'] � [RX:'�'] � [RX:'\x00']  [RX:'�'] � [RX:'�'] � [RX:'\x01']  [RX:'�'] � [RX:'�'] � [RX:'\x03']  [RX:'�'] � [RX:'�'] � [RX:'"'] " [TX:'\r\n']  [TX:'\r\n']  [TX:'\r\n']  [TX:'\r\n']  [TX:'\ [RX:'\n'] '\r\n']  [RX:'\r']
 [RX:'\n']  [RX:'y'] y [RX:'p'] p [RX:'e'] e [RX:'d'] d [RX:' ']   [RX:'l'] l [RX:'i'] i [RX:'n'] n [RX:'e'] e [RX:':'] : [RX:' ']   [RX:'\r']
 [RX:'L'] L [RX:'a'] a [RX:'s'] s [RX:'t'] t [RX:' ']   [RX:'c'] c [RX:'h'] h [RX:'a'] a [RX:'r'] r [RX:' ']   [RX:'w'] w [RX:'a'] a [RX:'s'] s [RX:':'] : [RX:' ']   [RX:'0'] 0 [RX:'x'] x [RX:'0']  [RX:'\n']
 [RX:'\n']  [RX:' ']   [TX:'a']  [RX:'a'] a [TX:'s']  [RX:'s'] s [TX:'d']  [RX:'d'] d [TX:'\r\n']  [RX:'\r']
 [RX:'\n']  [RX:'y'] y [RX:'p'] p [RX:'e'] e [RX:'d'] d [RX:' ']   [RX:'l'] l [RX:'i'] i [RX:'n'] n [RX:'e'] e [RX:':'] : [RX:' ']   [RX:'a'] a [RX:'s'] s [RX:'d'] d [RX:'\r']
 [RX:'L'] L [RX:'a'] a [RX:'s'] s [RX:'t'] t [RX:' ']   [RX:'c'] c [RX:'h'] h [RX:'a'] a [RX:'r'] r [RX:' ']   [RX:'w'] w [RX:'a'] a [RX:'s'] s [RX:':'] : [RX:' ']   [RX:'0'] 0 [RX:'x'] x [RX:'6']  [RX:'\n'] 4 [RX:'\r']
 [RX:'>'] > [RX:' ']
```

Maybe it's trying to emulate serial speed? Don't know anything about this Python module.

ivankravets, 2020-03-16:
It just prints what it receives.  I'll try to debug it later. We want to release PIO Core 4.3 this week, it would be great to have support for Renode! 

You can't imagine how I thankful for your examples and article! 😊

carlosedp, 2020-03-16:
The integration is pretty nice isn't! :)
Thanks!!

valeros, 2020-03-18:
Looks like at least basic telnet session negotiation is required, more info:
https://support.biamp.com/Tesira/Control/Telnet_session_negotiation_in_Tesira
http://mud-dev.wikidot.com/telnet:negotiation

ivankravets, 2020-07-29:
It's a little bit difficult to implement telnet client over the PySerial (re-invent the wheel). @valeros will back soon with updates where we can use Renode's device output as a native socket stream. He will describe solution in https://github.com/platformio/platformio-core/issues/3401

closed 2020-07-29 by ivankravets