platformio/platformio-core #3425: Is there a way to install OFFLINE?????

424778940z, 2020-03-18:
My friend have really slow and metered internet, the installation take forever and never complete...
I searched around, there are also people want make it portable have the same issue.
Even copy whole .vscode and .platformio folders won't work.

Cant understand why it is so hard to make it portable and there seems no progress at all on this issue.

I'm trying to get more people play with Arduino without using that shitty IDE, but the online slow download install only really killed everyone's patient...

424778940z, 2020-03-18:
If there is PATH or other local ENV required, then the script may should just patch that rather than download the whole thing from beginning, at least copy whole .platform folder will work in this way...

Jason2866, 2020-03-31:
I did a portable install VSC/Platformio for Window. You can take a look [Portable_VSC_PlatformIO](https://github.com/Jason2866/Portable_VSC_PlatformIO)

jiladahe1997, 2020-04-05:
If it's slow with downloading **_platformio packages_** _(like stm32HAL library, gcctools, which plateformio will download automatically after you create a new project_).

you could download it manually here:http://dl.bintray.com/platformio/ . And then just copy them to  \.platformio\packages\ .

424778940z, 2020-04-05:
no not the packages, I can install pio plugin but after that it will stuck at "Please do not close this window and do not open other folders until this process is completed"

jiladahe1997, 2020-04-11:
PlatformIO-IDE for vscode installation includes following steps:
1.check and download python3.7(if not installed) from _python official site_
2.pip install platfromIO core from _pip registry_
3.run CLI: platformIO home
4.install contrib-pysite from _http://dl.bintray.com/platformio/_
5.install contrib-piohome from _http://dl.bintray.com/platformio/_

You could open vscode developer tools by click "_Help->Toggle Developer Tools_" in the top bar, and some logs will be displayed in the console tab, which showing the currently running step.

By the way, I'm trying to dev a mirror site for PlatformIO in China(still under development), maybe will help you.

TCB13, 2023-10-07:
Related? https://github.com/platformio/platformio-core/issues/3333

qxdo, 2026-08-28:
Hi, is there any new update? It seems the link above are deprecated, because when I opened, "404 Not Found".