platformio/platformio-core #3430: how to change the ram size in platform?

KiraVerSace, 2020-03-20:
i know the platform support the  board  [nucleo_l476rg],
but in my project, i used  STM32L476RET6  that ram is less to 80KB , or i used STM32L476VG that pins are up to LQFP100.
which files i should to modified to adapt to the two chips?
thank you

ivankravets, 2020-03-20:
Please forward to https://community.platformio.org/c/development-platforms/17

closed 2020-03-20 by ivankravets

KiraVerSace, 2020-03-20:
I see nothing ?

| |
X.X.Yao
|
|
kiraversace@yeah.net
|
签名由网易邮箱大师定制

ivankravets, 2020-03-20:
https://community.platformio.org/

KiraVerSace, 2020-03-20:
> https://community.platformio.org/

thank you for your replay , but I can not find any topic about the question, and i just register in the community, so i cannot post a topic?

ivankravets, 2020-03-20:
Please try again to crate a post.