platformio/platformio-core #3431: cannot open serial monitor when using nodemcu

emrizkis, 2020-03-20:
Error: Traceback (most recent call last):
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/platformio/commands/device/command.py", line 197, in device_monitor
    platform, kwargs["project_dir"], kwargs["environment"]
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/platformio/commands/device/helpers.py", line 106, in register_platform_filters
    load_monitor_filter(path, project_dir, environment)
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/platformio/commands/device/helpers.py", line 90, in load_monitor_filter
    obj = cls(project_dir, environment)
  File "/home/emrizkis/.platformio/penv/lib/python2.7/site-packages/platformio/commands/device/filters/base.py", line 23, in __init__
    super(DeviceMonitorFilter, self).__init__()
TypeError: super(type, obj): obj must be an instance or subtype of type

closed 2020-03-20 by ivankravets

ru-nekit-android, 2020-03-20:
I setup Python 3.8 and use virtual-environment (https://docs.platformio.org/en/latest/core/installation.html#virtual-environment) for solve this problem.

ivankravets, 2020-03-20:
We have fixed this issue. Please re-test with `pio upgrade --dev`.

ivankravets, 2020-03-20:
PlatformIO Core 4.3.1 is out! Please upgrade via `pio upgrade --dev`.

commit 658b3df1 (2020-03-20): Fixed an TypeError "super(type, obj): obj must be an instance or subtype of type" when device monitor is used with a custom dev-platform filter // Resolve #3431