platformio/platformio-core #3434: executing menuconfig make target results in error

eliabieri, 2020-03-21:
### Configuration

**Operating system**:
MacOs

**PlatformIO Version** (`platformio --version`):
4.3.1

### Description of problem
When running `platformio run -t menuconfig`, the following error appears:
```
scons: *** Do not know how to make File target `menuconfig' (/private/tmp/test/menuconfig).
```

#### Steps to Reproduce

1. Run platformio run -t menuconfig
2. The error appears

### Actual Results
```
scons: *** Do not know how to make File target `menuconfig' (/private/tmp/test/menuconfig).
```

### Expected Results
The menuconfig UI is started

**The content of `platformio.ini`:**
```ini
[env:m5stick-c]
platform = espressif32
board = m5stick-c
framework = espidf
```

ivankravets, 2020-03-21:
ESP32 dev-platform with support for menuconfig has not been released yet. Pelase use upstream version https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream

closed 2020-03-21 by ivankravets

2dom, 2020-04-05:
Still present with upstream version:

```
[env:esp32thing]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32thing
monitor_baud = 115200
framework = arduino
lib_extra_dirs = C:\Users\dbuch\Dropbox\coding\Arduino\libraries
```

valeros, 2020-04-05:
Hi @2dom ! The `menuconfig` target is only available for `ESP-IDF` framework and Arduino uses prebuilt `ESP-IDF` so it's not configurable.

2dom, 2020-04-05:
I see ... thanks for the heads up

vindolin, 2021-01-22:
> 
> 
> Hi @2dom ! The `menuconfig` target is only available for `ESP-IDF` framework and Arduino uses prebuilt `ESP-IDF` so it's not configurable.

Are there any workarounds if I wanted to use arduino?
E.g. to enable CONFIG_PM_ENABLE on a ESP32s2 module.

pcontezini, 2021-04-03:
Did anyone made the menuconfig work with PlataformIO?

I'm using the develop framework as pointed out and it doesn't know the menuconfig target with pio run -t menuconfig.

Thanks

zekageri, 2021-10-05:
same problem here