platformio/platformio-core #3435: Fish shell completions

sdaitzman, 2020-03-22:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.0.0

### Description of problem
I would love to have Fish Shell completions!

ivankravets, 2020-03-24:
Have you tried this https://github.com/click-contrib/click-completion? if it works, we will add to our docs.

ivankravets, 2020-05-10:
Please re-test with `pio upgrade --dev`. See docs how to use it https://docs.platformio.org/en/latest/core/userguide/misc/completion/index.html

mcspr, 2020-05-11:
Thanks! re #3449, running `pio misc completion install` multiple times should overwrite instead of append to the completion file? `append` kwarg seems to do it.
```
$ cat ~/.config/fish/completions/pio.fish
complete --command pio --arguments "(env _PIO_COMPLETE=complete-fish COMMANDLINE=(commandline -cp) pio)" -f
complete --command pio --arguments "(env _PIO_COMPLETE=complete-fish COMMANDLINE=(commandline -cp) pio)" -f
complete --command pio --arguments "(env _PIO_COMPLETE=complete-fish COMMANDLINE=(commandline -cp) pio)" -f
```

New versions works for me in fish, as well as raw click-complete patch from the referenced PR. Sadly, dynamic things are still missing, like `--board` or `--environment` lists. I have not explored that part more in-depth.

---

There is also a weird bug when typing `pio foo .` and TAB
```python3
  File "/home/builder/runtime/platformio-core/platformio/commands/__init__.py", line 64, in get_command
    mod = __import__(mod_path, None, None, ["cli"])
ModuleNotFoundError: No module named 'platformio.commands.foo'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/builder/runtime/platformio-core/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/builder/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/builder/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 712, in main
    _bashcomplete(self, prog_name, complete_var)
  File "/home/builder/.platformio/penv/lib64/python3.8/site-packages/click_completion/patch.py", line 135, in _shellcomplete
    do_fish_complete(cli, prog_name)
  File "/home/builder/.platformio/penv/lib64/python3.8/site-packages/click_completion/core.py", line 198, in do_fish_complete
    for item, help in get_choices(cli, prog_name, args, incomplete):
  File "/home/builder/.platformio/penv/lib64/python3.8/site-packages/click_completion/core.py", line 86, in get_choices
    ctx = resolve_ctx(cli, prog_name, args)
  File "/home/builder/.platformio/penv/lib64/python3.8/site-packages/click_completion/lib.py", line 82, in resolve_ctx
    cmd = ctx.command.get_command(ctx, a[0])
  File "/home/builder/runtime/platformio-core/platformio/commands/__init__.py", line 70, in get_command
    raise click.UsageError('No such command "%s"' % cmd_name, ctx)
click.exceptions.UsageError: No such command "foo"
```

ivankravets, 2020-05-23:
Please re-test with `pio upgrade --dev`.

Also please note that command was changed from `pio misc` to `pio system`.

ivankravets, 2020-05-23:
Resoled in https://github.com/platformio/platformio-core/commit/7dce494ad62792927be171a302dd2d5ddd514be0

commit 01a1981c (2020-05-10): Added PlatformIO CLI Shell Completion for Fish, Zsh, Bash, and PowerShell // Resolve #3435
closed 2020-05-23 by ivankravets