platformio/platformio-core #3438: Package manager could not install cmake

adrianpanicek, 2020-03-23:
### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version**: 4.3.2a1

### Description of problem

When I try to build a project for espressif platform, PackageManager says

`Error: Could not find a version that satisfies the requirement '~3.16.0' for your system 'windows_x86'`

Platformio worked for me before. I tried removing `.platformio` from Users folder and even reinstalling it in VScode but it didn't help.

Full log:
```
> Executing task: C:\Users\Ado\.platformio\penv\Scripts\platformio.exe run <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

PackageManager: Installing tool-cmake @ ~3.16.0
Error: Could not find a version that satisfies the requirement '~3.16.0' for your system 'windows_x86'
```

#### Steps to Reproduce

1. Rebuild project
2. Observe error

### Actual Results

Platformio crashes out before building.

### Expected Results

Platformio builds a project

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
build_flags = -std=c++17
build_unflags = -std=gnu++11
monitor_speed = 115200
lib_ldf_mode = chain
monitor_flags=
        --raw
upload_protocol = esptool
```

valeros, 2020-03-23:
Hi @adrianpanicek ! Looks like you have 32-bit version of Python installed. We've added the missing package for this architecture, could you please try again?

adrianpanicek, 2020-03-24:
@valeros This worked, thank you very much.

closed 2020-03-24 by adrianpanicek