platformio/platformio-core #3441: CLION create project fails due to missing write_file_contents in fs.py

FiratBerkCakar, 2020-03-26:
### Configuration

**Operating system**: Ubuntu 19.04

**PlatformIO Version** (`platformio --version`): 4.3.1

### Description of problem
I have updated platfromio to its newest version today and wanted to try how the Clion IDE plugin works. When i tried to create a platfromio project for my Teensy 3.2 from CLion, it failed to create any folder with a readme.md file in it (other folders such as src are there). Python script was failing due to **write_file_contents** function not found in fs.py. 

 * pio init --ide clion --board teensy31 comand was working from CLI

I checked the aforementioned file and the function was really missing there. I solved the issue by copying the same fs.py file from an older version which surprisingly **did** have the write_file_contents functions.

I just wanted to bring this to your attention.

ivankravets, 2020-03-27:
There is no `write_file_contents` in our source code. Do you have multiple PIO Cores in a system? Please remove them https://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

We recommend using our new portable and isolated installer. See updated docs https://docs.platformio.org/en/latest/core/installation.html#installation-methods

FiratBerkCakar, 2020-03-28:
I already had a pio core installed when i decided to try the offical Clion Platformio plugin. I am not sure if installing the plug in also installs another pio-core. If thats the case, it's also possible that the plugin's own pio-core has this function or needs it but then its also weird that i copied the function from another pc that doesnt have the plugin installed.

ivankravets, 2020-03-31:
Sorry for the issue, let's move to this thread https://github.com/platformio/platformio-docs/issues/124

Please uninstall all PlatformIO Cores from a system and install it as described in CLion docs.

/cc @anastasiak2512 @elmot

closed 2020-03-31 by ivankravets