platformio/platformio-core #3442: Saving libraries in new project results in error "No option 'lib_deps' in section"

simonvanderveldt, 2020-03-26:
What kind of issue is this?

- Linux
- PlatformIO, version 4.3.1

### Description of problem
When creating a new project (https://docs.platformio.org/en/latest/core/quickstart.html) installing a library and trying to save it to `platformio.ini` the saving fails because the `lib_deps` section is missing (and it apparently doesn't automatically get created):
```
platformio lib install --save PubSubClient
Library Storage: /home/simon/src/simonvanderveldt/arduino-plc/.pio/libdeps/uno
Looking for PubSubClient library in registry
Found: https://platformio.org/lib/show/89/PubSubClient
LibraryManager: Installing id=89
Using cache: /home/simon/.platformio/.cache/60/bcf94560201e022015c1c79750fa0460
PubSubClient @ 2.7 has been successfully installed!
Error: Invalid '/home/simon/src/simonvanderveldt/testproject/platformio.ini' (project configuration file): 'No option 'lib_deps' in section: 'env:uno''
```

#### Steps to Reproduce
```
platformio project init --board uno
platformio lib install --save PubSubClient
```

### Actual Results
Error

### Expected Results
The library gets saved in the default environment

ivankravets, 2020-03-27:
Thanks for the report! Please re-test with `pio upgrade --dev`.

closed 2020-03-27 by ivankravets

simonvanderveldt, 2020-03-27:
@ivankravets It works :) Thanks for the quick fix!

ivankravets, 2020-03-27:
Happy coding with PlatformIO! 🙃

commit e92b498b (2020-03-27): Fixed an issue when saving libraries in new project results in error "No option 'lib_deps' in section" // Resolve #3442