platformio/platformio-core #3448: builtins.TypeError: the JSON object must be str, not 'bytes' with PIO Remote

Pyro-Fox, 2020-04-04:
### Configuration

**Operating system (Local)**:  Windows 10 (1903)
**Operating system (Remote)**:  Raspbian GNU/Linux 9 (stretch)

**PlatformIO Version** (`platformio --version`): Failing in both 4.3.1 and 4.3.2a1
This failed in the following combos (Local -> Remote):
- 4.3.1 -> 4.3.1
- 4.3.1 -> 4.3.2a1
- 4.3.2a1 -> 4.3.2a1

### Description of problem
When I use `remote upload` from PlatformIO IDE for VSCode to a remote agent running from PlatformIO Core, PIO core fails with the error "builtins.TypeError: the JSON object must be str, not 'bytes'"

#### Steps to Reproduce

1. Compile project (MarlinFw in this case)
2. Run the following on the RasPi/OctoPrint SSH: `. ~/platformio/venv/bin/activate ; pio remote agent start`
3. Click `remote upload` from PlatformIO IDE for VSCode

### Actual Results
**From PlatformIO IDE for VSCode:**
> Uploading firmware remotely
Agent has been terminated

**From PlatformIO Core:**
> PIO Remote v2.6.3
2020-04-04 03:08:11 [info] Name: octopi
2020-04-04 03:08:11 [info] Connecting to PIO Remote Cloud
2020-04-04 03:08:11 [info] Successfully connected
2020-04-04 03:08:11 [info] Authenticating
2020-04-04 03:08:11 [info] Successfully authorized
2020-04-04 03:08:20 [info] Remote command received: device.list
2020-04-04 03:08:53 [info] Remote command received: psync
Peer will receive following PB traceback:
Unhandled Error
Traceback (most recent call last):
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/spread/banana.py", line 173, in gotItem
    self.callExpressionReceived(item)
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/spread/banana.py", line 136, in callExpressionReceived
    self.expressionReceived(obj)
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 621, in expressionReceived
    method(*sexp[1:])
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 1025, in proto_message
    self._recvMessage(self.localObjectForID, requestID, objectID, message, answerRequired, netArgs, netKw)
--- <exception caught here> ---
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 1054, in _recvMessage
    netResult = object.remoteMessageReceived(self, message, netArgs, netKw)
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/spread/flavors.py", line 125, in remoteMessageReceived
    state = method(*args, **kw)
  File "<string>", line 46, in remote_acwrite
  File "<string>", line 282, in ac_write
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
builtins.TypeError: the JSON object must be str, not 'bytes'

### Expected Results
I don't have a copy of the successful run, but this was working before I updated to 4.3.1

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
#
# Marlin Firmware
# PlatformIO Configuration File
#
# For detailed documentation with EXAMPLES:
#
# http://docs.platformio.org/en/latest/projectconf.html
[... 818 lines cut ...]
  -DHAL_PCD_MODULE_ENABLED
  -DUSBD_USE_CDC
  -DDISABLE_GENERIC_SERIALUSB
  -DHAL_UART_MODULE_ENABLED
  -Os
lib_ignore    = Adafruit NeoPixel
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32> +<src/HAL/HAL_STM32_F4_F7> -<src/HAL/HAL_STM32_F4_F7/STM32F7>
upload_protocol = dfu

#
# Just print the dependency tree
#
[env:include_tree]
platform    = atmelavr
board       = megaatmega2560
build_flags = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
lib_deps    = ${common.lib_deps}
  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
src_filter  = +<src/Marlin.cpp>
```

### Additional info
It's been a couple months since I had to update FW via PIO so I don't 100% remember what the "Working" version was. I just see that #3396 was recently closed, but I seem to have an identical problem even on the dev build.

If it helps, I had the latest versions of PIO Core and for VSCode the last time I used PIO remote on 12/1/2019

valeros, 2020-04-06:
What is the Python version on your local machine? Both Pythons should have the same major and minor version.

valeros, 2020-04-06:
It's been a while since December, I'd highly recommend you to try Python 3.8 on both sides.

ivankravets, 2020-04-07:
@Pyro-Fox 

Do you use virtual env for both `platformio` commands? So,

```
bin/activate
pip install platformio
pio update
pio agent start
```

Pyro-Fox, 2020-04-08:
> @Pyro-Fox
> 
> Do you use virtual env for both `platformio` commands? So,
> 
> ```
> bin/activate
> pip install platformio
> pio update
> pio agent start
> ```

yes. I run all PIO commands from the virtual env.

Pyro-Fox, 2020-04-09:
> It's been a while since December, I'd highly recommend you to try Python 3.8 on both sides.

Well it was a total PITA (since I'm really a linux newb), but I got the virtual env updated to Python 3.8.2 and remote agent is working again. TY for the advice.

closed 2020-04-09 by Pyro-Fox