platformio/platformio-core #3450: Cannot get PVS Studio to work (Missing preprocessed file)

TD-er, 2020-04-06:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.3.2a1

### Description of problem
I try to use the (great) new feature of using PVS Studio to check my code.

Relevant part of my platformio.ini file:

```
[debug_pio]
build_type                = debug
;check_tool                = clangtidy
check_tool = pvs-studio
check_flags =
  pvs-studio: --analysis-mode=4 --errors-off=V532,V586 --lic-file=../pvs_studio.lic
build_flags               = ${compiler_warnings.build_flags}
```

The envs needing this include this via this PlatformIO extends option:
```
extends                   = debug_pio
```

### Actual Results

```
PS C:\GitHub\TD-er\ESPEasy> pio check --environment spec_debug_beta_custom_ESP8266_4M1M
Checking spec_debug_beta_custom_ESP8266_4M1M > pvs-studio (platform: https://github.com/platformio/platform-espressif8266.git; board: esp12e; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Error: Missing preprocessed file 'C:\Users\gijs\AppData\Local\Temp\piocheck20x7f88s\1luqj9ex.i'
Error: Missing preprocessed file 'C:\Users\gijs\AppData\Local\Temp\piocheck20x7f88s\1luqj9ex.i'
Error: Missing preprocessed file 'C:\Users\gijs\AppData\Local\Temp\piocheck20x7f88s\1luqj9ex.i'
Error: pvs-studio failed to perform check! Please examine tool output in verbose mode.
```

When I look at this directory while the check is still active, I do see the mentioned file is created (roughly 1.7 MB in my project)
It is created a few times.
But PVS studio is apparently not able to see the file.
Maybe an issue with `\` vs `/` ???

### Expected Results
Successful check by PVS studio.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[debug_pio]
build_type                = debug
;check_tool                = clangtidy
check_tool = pvs-studio
check_flags =
  pvs-studio: --analysis-mode=4 --errors-off=V532,V586 --lic-file=../pvs_studio.lic
build_flags               = ${compiler_warnings.build_flags}
```

I do have an active license file and the error is different when I rename the license file, so it appears to be loaded by PVS-studio.

valeros, 2020-04-06:
Hi @TD-er ! Thanks for the report. Could  you please also attach here the output in verbose mode?
`pio check --environment spec_debug_beta_custom_ESP8266_4M1M --verbose`

TD-er, 2020-04-06:
Sure.
[Verbose_PVSstudio_platformIO_3450.txt](https://github.com/platformio/platformio-core/files/4440543/Verbose_PVSstudio_platformIO_3450.txt)

As you can see, these errors are right at the start:
```
PS C:\GitHub\TD-er\ESPEasy> pio check --environment spec_debug_beta_custom_ESP8266_4M1M --verbose
Checking spec_debug_beta_custom_ESP8266_4M1M > pvs-studio (platform: https://github.com/platformio/platform-espressif8266.git; board: esp12e; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Missing preprocessed file 'C:\Users\gijs\AppData\Local\Temp\piocheckepvv1kzk\67yhazr9.i'

Error: Missing preprocessed file 'C:\Users\gijs\AppData\Local\Temp\piocheckepvv1kzk\67yhazr9.i'

Error: Missing preprocessed file 'C:\Users\gijs\AppData\Local\Temp\piocheckepvv1kzk\67yhazr9.i'

C:\Users\gijs\.platformio\packages\tool-pvs-studio\x64\pvs-studio --skip-cl-exe yes --language C++ --preprocessor gcc --cfg C:\Users\gijs\AppData\Local\Temp\piocheckepvv1kzk\0til68v0.cfg --source-file C:\GitHub\TD-er\ESPEasy\src\ESPEasy-Globals.cpp --i-file C:\Users\gijs\AppData\Local\Temp\piocheckepvv1kzk\67yhazr9.i --output-file C:\Users\gijs\AppData\Local\Temp\piocheckepvv1kzk\j7leno19.pvs --platform=arm --analysis-mode=4 --errors-off=V532,V586 --lic-file=../pvs_studio.lic
...
```

valeros, 2020-04-06:
It looks like that the toolchain (at least `xtensa`) fails to determine what language is used in `.ino` files.  I'll check how we can resolve this.

TD-er, 2020-04-07:
Glad it isn't something I could have solved by reading the docs even more than I already did :)

TD-er, 2020-04-07:
[Verbose_PVSstudio_platformIO_3450_20200407_1051.txt](https://github.com/platformio/platformio-core/files/4443037/Verbose_PVSstudio_platformIO_3450_20200407_1051.txt)

It is still telling me `Error: pvs-studio failed to perform check! Please examine tool output in verbose mode.`
So the verbose output is attached in a txt file.
It does seem to parse .ino files now, but I have no clue what's wrong here.

Also I wonder how it can parse .ino files and not the generated .cpp file by concatenating all .ino files and generated forward declarations of all found functions.

valeros, 2020-04-07:
The log seems fine to me. I don't have a license so I tried to scan your project in free mode and it worked just fine. I created a special branch with a few extra debug prints, you can install using it using pip:
```
pip install https://github.com/platformio/platformio-core/archive/feature/pvs-studio-3450.zip
```
I'll be great if you can scan your project again and attach here verbose output.
Thanks!

TD-er, 2020-04-08:
[Verbose_PVSstudio_platformIO_3450_20200408_0815.txt](https://github.com/platformio/platformio-core/files/4448540/Verbose_PVSstudio_platformIO_3450_20200408_0815.txt)

At first glance I don't see any difference.

TD-er, 2020-04-08:
Just to be sure, I also ran a test with the licence file in the same folder as the platformio.ini file.
It makes no difference.

valeros, 2020-04-27:
Hi @TD-er , sorry for the late reply. We've updated PVS-Studio to v7.07, so could you please retest your project with the latest PlatformIO from the development branch? If it still doesn't work, is there any chance that we could arrange a remote debug session (using AnyDesk or TeamViewer)?

Thanks is advance!

TD-er, 2020-04-28:
Hmm, tested it now on a different computer, just to be sure it was not related to incorrect install and also I'm not near that other computer right now.

I created the license file and get this:
```
License information is incorrect. Please check your registration data or contact Customer Support at support@viva64.com
```

So I guess I will also write an email to them.

TD-er, 2020-04-29:
OK, with the great (and swift) support of the people behind PVS-studio we managed to find out what's wrong here.

The license file should be in UTF-8 + BOM format.
Also it should only contain information like this:

```
name@domain.com
SXAB-CDEF-GHIJ-KLMN
```

The reported error in the previous version of PVS-studio was not helpful at all and this was a bit more indicative, but still it took a while before they also noticed the true issue here.

So this issue can be closed, but maybe you can add this information to the PlatformIO documentation?

valeros, 2020-04-29:
Thank for the clarification!

commit baa7aab1 (2020-04-07): Specify C++ as the language for .ino files when preprocessing them for PVS-Studio // Resolve #3450
closed 2020-04-29 by valeros