platformio/platformio-core #3453: On creating a node

Dpkpadhy, 2020-04-08:
On creating first node from ESPHome in hassio, getting below error. Trying from last 2 days, it is not accessible.  Seems service is down.

Any help here is really appreciable.

[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x044FF710>: Failed to establish a new connection: [Errno 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond',)) (incremented retry: max=5, total=5)
Error: [API] Could not connect to PlatformIO API Service. Please try later.

ivankravets, 2020-04-08:
Please check your firewall. Do you have access to https://api.platformio.org/lib/stats ?

closed 2020-04-08 by ivankravets

Dpkpadhy, 2020-04-08:
Thanks for your update, 
i tried to access https://api.platformio.org/lib/stats 
Here is the output, 
{"dlweek": [{"diff": 188, "total": 3867, "name": "AsyncTCP-esphome", "id": 6798}, {"diff": 187, "total": 4074, "name": "ESPAsyncTCP-esphome", "id": 6757}, {"diff": 170, "total": 3723, "name": "ESPAsyncWebServer-esphome", "id": 6758}, {"diff": 68, "total": 754, "name": "FastLED", "id": 126}, {"diff": 59, "total": 548, "name": "LiquidCrystal_I2C", "id": 576}, {"diff": 56, "total": 375, "name": "TMC2130Stepper", "id": 1493}, {"diff": 44, "total": 1057, "name": "Adafruit STMPE610", "id": 377}, {"diff": 37, "total": 1101, "name": "PubSubClient", "id": 89}, {"diff": 36, "total": 97, "name": "FreeRTOS", "id": 507}, {"diff": 32, "total": 56, "name": "Esp32WifiManager", "id": 6309}], "added": [{"date": "2020-04-06T09:10:10Z", "name": "arduino-tiny", "id": 7189}, {"date": "2020-04-06T09:10:06Z", "name": "TM1638-extended", "id": 7188}, {"date": "2020-04-05T17:30:07Z", "name": "tiny", "id": 7187}, {"date": "2020-04-05T17:30:05Z", "name": "LoopTicker", "id": 7186}, {"date": "2020-04-02T21:10:05Z", "name": "Birdhouse SDK", "id": 7185}], "dlday": [{"diff": 4, "total": 4, "name": "EspMQTT", "id": 181}, {"diff": 2, "total": 2, "name": "PID-AutoTune", "id": 3}, {"diff": 2, "total": 2, "name": "PCA9685", "id": 117}, {"diff": 2, "total": 2, "name": "EngduinoMagnetometer", "id": 574}, {"diff": 2, "total": 2, "name": "Simple-Wiegand-Library-for-Arduino", "id": 1814}, {"diff": 2, "total": 2, "name": "ESP32httpUpdate", "id": 2013}, {"diff": 2, "total": 2, "name": "DTH", "id": 3491}, {"diff": 2, "total": 2, "name": "ShiftRegister74HC595", "id": 4224}, {"diff": 2, "total": 2, "name": "Espalexa", "id": 5525}, {"diff": 2, "total": 2, "name": "ArduinoEigen", "id": 5648}], "lastkeywords": ["tiny", "state machine", "birdhouse", "indicators", "fdt"], "updated": [{"date": "2020-04-08T09:30:28Z", "name": "ThingsML", "id": 7169}, {"date": "2020-04-08T09:00:14Z", "name": "smart-input-filter", "id": 7007}, {"date": "2020-04-08T08:46:23Z", "name": "BlynkGSM_Manager", "id": 6988}, {"date": "2020-04-08T08:31:27Z", "name": "YACL", "id": 6917}, {"date": "2020-04-08T06:45:32Z", "name": "NRF_HAL", "id": 6350}], "topkeywords": ["communication", "sensors", "display", "control", "device", "other", "lcd", "i2c", "input", "output", "signal", "sensor", "data", "oled", "tft", "graphics", "arduino", "spi", "font", "glcd", "serial", "displaycore", "timing", "led", "esp8266", "iot", "processing", "temperature", "esp32", "storage", "motor", "wifi", "uncategorized", "http", "mqtt", "pwm", "driver", "rtc", "ethernet", "mbed", "time", "timer", "button", "humidity", "web", "eeprom", "gps", "servo", "nucleo", "rtos"], "dlmonth": [{"diff": 5449, "total": 34263, "name": "Adafruit NeoPixel", "id": 28}, {"diff": 2821, "total": 2882, "name": "Adafruit ADXL343", "id": 6117}, {"diff": 2656, "total": 2703, "name": "Adafruit STMPE610", "id": 377}, {"diff": 2590, "total": 2678, "name": "Adafruit TouchScreen", "id": 5430}, {"diff": 2182, "total": 2536, "name": "Adafruit ILI9341", "id": 571}, {"diff": 1995, "total": 26906, "name": "LiquidCrystal", "id": 136}, {"diff": 1646, "total": 1991, "name": "LoRa", "id": 1167}, {"diff": 1287, "total": 31030, "name": "TMCStepper", "id": 5513}, {"diff": 1284, "total": 8720, "name": "ArduinoJson", "id": 64}, {"diff": 1246, "total": 1268, "name": "arduino-timer", "id": 5820}]}

ivankravets, 2020-04-08:
> ESPHome in hassio

What is it? Could you try to restart this device?

Dpkpadhy, 2020-04-08:
Rebooted the device still the same issue. 

I am not able to access api.platformio.org.. Tried from multiple devices.

ivankravets, 2020-04-08:
You need to fix your internet connection/settings. We don't block any people. So, if you have access from your local machine, it should work from RPi too.

Dpkpadhy, 2020-04-08:
I am not only trying from RPi, accessing from multiple local device's. Also tried outside of my local network. 

When i try to accessmanually from web browser https://api.platformio.org/ , getting below error.

Error : 
===================================================
405 Method Not Allowed
Sorry, the requested URL 'https://api.platformio.org/' caused an error:

Method not allowed.
===================================================

ivankravets, 2020-04-08:
This is correct. There is no endpoint in the root of API. The valid is https://api.platformio.org/lib/stats

Could you run `ping api.platformio.org` on device where you don't have access? And also, `ping platformio.org`?

Dpkpadhy, 2020-04-08:
Just to update, i have added ESPHome plugin in Home assistant and trying to create a node. When i do a compile, connection problem is coming.

I tried couple of things but still failing.
Both the ping command are failing with request timeout. Though i am still able to browse these 2 links "platformio.org" and "nm1.platformio.org" 

Here is the Ping command output:
======================================
C:\Users\Admin>**ping platformio.org**

Pinging platformio.org [88.198.170.159] with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 88.198.170.159:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),

C:\Users\Admin>**ping api.platformio.org**

Pinging nm1.platformio.org [31.28.1.238] with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 31.28.1.238:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),
==============================================

ivankravets, 2020-04-08:
Sorry, we can't provide more help here. You don't have Internet access on your devices. Please ask people here https://community.platformio.org/

ivankravets, 2020-04-08:
Sorry, I forgot, ICMP packets are blocked on our servers. Could you try these commands on RPI?
```
curl -fsSL https://platformio.org
curl -fsSL https://api.platformio.org/stats
```

Dpkpadhy, 2020-04-09:
When i try the curl commands from RPI, getting "could not resolve host"

core-ssh:~# curl -fsSL https://platformio.org
curl: (6) Could not resolve host: platformio.org
core-ssh:~# curl -fsSL https://api.platformio.org/stats
curl: (6) Could not resolve host: api.platformio.org

Dpkpadhy, 2020-04-09:
To update more when i run the above curl commands in my local which is in same wifi network, it does work.

ivankravets, 2020-04-09:
Yes, please ask at https://www.raspberrypi.org/forums/ how to configure internet on RPI + DNS.