platformio/platformio-core #3456: RIOT-OS support

cselti, 2020-04-12:
Supporting [RIOT-OS](https://www.riot-os.org/) as a framework could be useful. It supports several platforms (ESP, ATMega, ARM, ...), several sensor modules (as [drivers](https://github.com/RIOT-OS/RIOT/tree/master/drivers)), etc... 
I don't know too much about the RIOT-OS, but I think it could be fit in the PlatformIO's frameworks (probably a similar way like to the Zephyr project?)
I have checked repo of the RIOT - I found a [former ticket](https://github.com/RIOT-OS/RIOT/issues/8591) about the topic, I think the RIOT developers don't plan to integration at the moment.

ivankravets, 2020-04-13:
Please upvote 👍 this feature request (smile in the right upper corner of your comment). We work on the most requested features by our Community. Otherwise, @gebart and his team can provide integration from their side.

/cc @Citrullin  @basilfx @douardda @andrea-nisti

Citrullin, 2020-04-15:
@ivankravets I thought about adding the support as well, because I worked with platformIO on something work related. :) So, feel free to ping me at any point, if you need help. Would like to understand platformIO even more :)

Dahkai, 2020-04-15:
Good luck 🤞

akshaim, 2020-05-05:
Hope this materialize soon !

aabadie, 2020-07-25:
Any progress @Citrullin ?

Citrullin, 2020-07-30:
> Any progress @Citrullin ?

I haven't worked on it. So, you probably should ask @ivankravets

dzarda, 2020-08-20:
@ivankravets Perhaps could be viewed as *Makefile project support*? I.e. ChibiOS has a thoroughly Makefile-based structure and would also benefit from it.

leonty, 2021-01-16:
@ivankravets Hello, thanks again for a great tool! Do you guys still consider adding support of the RIOT OS? It's really a serious OS with support of many SoCs (>40) and boards (>200). Still it has very simple build system based on makefiles. There are 114 upvotes already)

ivankravets, 2021-01-16:
Hi there,
The RIOT-OS team works on integration. Stay tuned! 🥳

mderouss, 2021-01-16:
Great that this is happening ! I've looked at RIOT for a few projects in the last 18 months, but have gotten used to PlatformIO :)  Will certainly encourage me to use RIOT.

mansk789, 2021-08-17:
any news for this topic? It would be great if there platformio would support RIOT-OS

leonty, 2021-08-17:
There was an open issue in the RIOT OS repo, but now it's closed, no one seemed to be interested in this integration...
https://github.com/RIOT-OS/RIOT/issues/8591

Citrullin, 2021-08-17:
> 
> 
> There was an open issue in the RIOT OS repo, but now it's closed, no one seemed to be interested in this integration...
> [RIOT-OS/RIOT#8591](https://github.com/RIOT-OS/RIOT/issues/8591)

Yes, I closed it, since it isn't so much of an issue in RIOT itself. I still would like to see it :)

leonty, 2021-08-18:
@ivankravets Please, give us some details if any on the work progress.

smurfix, 2022-04-11:
Is anybody interested in doing the work? I seem to need it and would contribute some $$$ to the effort.

Citrullin, 2022-04-11:
> Is anybody interested in doing the work? I seem to need it and would contribute some $$$ to the effort.

I added it to the [RIOT OS forum](https://forum.riot-os.org/t/platformio-integration/3590). Maybe you can add some details there. It's the best place to reach out to the RIOT OS users.

bitsformbyte, 2022-09-12:
Is there anyone working on this? @ivankravets

redradist, 2024-09-10:
Some updates on it ??

Citrullin, 2024-09-20:
> Some updates on it ??

Sadly not. I would love to work on it or fund it eventually, but it's not a priority right now. 
Also haven't heard of anyone in the RIOT OS community working on it.