platformio/platformio-core #3458: platformio_lib_dirs gives error TypeError: expected str, bytes or os.PathLike object, not list:

ASL07, 2020-04-16:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Os X

**PlatformIO Version** (`platformio --version`): 4.3.1

### Description of problem
When addding
`lib_extra_dirs = /Users/developer/Code/common
` 
to platformio.ini I get the following error. I have been reading the docs and researching info for several hours and don't understand what is wrong.
I am following the documentation 100%

`TypeError: expected str, bytes or os.PathLike object, not list:
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/developer/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/developer/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/developer/.platformio/platforms/espressif8266/builder/main.py", line 203:
    target_elf = env.BuildProgram()
  File "/Users/developer/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/Users/developer/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/Users/developer/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 1041:
    project.search_deps_recursive()
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 406:
    self.process_dependencies()
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 890:
    self.depend_recursive(lb)
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 401:
    lb.search_deps_recursive(search_files)
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 406:
    self.process_dependencies()
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 285:
    self.depend_recursive(lb)
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 401:
    lb.search_deps_recursive(search_files)
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 416:
    for inc in self._get_found_includes(search_files):
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 322:
    [self.env.Dir(d) for d in lb.get_include_dirs()]
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 762:
    include_dirs = LibBuilderBase.get_include_dirs(self)
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 182:
    include_dir = self.include_dir
  File "/usr/local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 681:
    return realpath(self._manifest.get("build").get("includeDir"))
  File "/usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/posixpath.py", line 394:
    filename = os.fspath(filename)`

valeros, 2020-04-16:
Hi @ASL07 ! Thanks for the report. Could you please provide more information about your project? What libraries are there in `/Users/developer/Code/common` folder? Looks like one of them declared a list of directories in `includeDir` option while only [one folder is allowed](https://docs.platformio.org/en/latest/librarymanager/config.html#includedir). 

Thanks!

ASL07, 2020-04-17:
@valeros  
Thanks, that was the problem. I have a library with include files segregated amongst several directories.
Something like this (simplified):
```
library
├── CMakeLists.txt
├── library.json
├── src
│   ├── module1
│   │   ├── include
│   │   │   ├── module1.h
│   │   ├── src
│   │   │   ├── module1.cpp
│   ├── module2
│   │   ├── include
│   │   │   ├── module2.h
│   │   ├── src
│   │   │   ├── module2.cpp
```

What is the best way to configure library.json for this? I don't want to modify the file tree structure because it will mess with the cmake build 

I have resorted to  adding -Isrc/module1/include and -Isrc/module2/include to the build_flags , but I wonder if this is the best way

Regards

valeros, 2020-04-17:
The `build_flags` field in `library.json` is totatally fine, if you need more control over the build environment you can use [an extra_script](https://docs.platformio.org/en/latest/librarymanager/config.html#extrascript).

I'm closing this issue as there is no problems with `platformio-core`.

closed 2020-04-17 by valeros

ASL07, 2020-04-17:
Just one more question, one of the modules depends on ESP8266 libraries such as ESP8266mDNS.h
When I try to compile the library inside a project with 
```
platform = espressif8266
board = esp8285
framework = arduino
```
I get the error
`ESP8266mDNS.h: No such file or directory`

Are the espressif and arduino files not available to the library?

Please see my relevant library.json and platformio.ini sections:
```
{
    "name": "myLibraryName",
    "version": "1.0.0",
    "authors":
    {
        "name": "authorName",
        "email": "author@email.com"
    },
    "repository":
    {
        "type": "git",
        "url": ""
    },
    "export": {
        "include":
        [
        ],
        "exclude":
        [          
        ]
    },
    "build": {
        "flags":
        [
            "-Isrc/module1/include",
        ], 
        "srcFilter":
        [
            "+<*>",
            "-<tests/*>"
        ]
    },
    "frameworks": "*",
    "platforms": "*"
}

```

```
[env:esp01_1m]
platform = espressif8266
board = esp8285
lib_extra_dirs = /Users/developer/Code/common
lib_deps = myLibraryName
```

ASL07, 2020-04-17:
> Just one more question, one of the modules depends on ESP8266 libraries such as ESP8266mDNS.h
> When I try to compile the library inside a project with
>
> ```
> platform = espressif8266
> board = esp8285 [...]

Solved it by adding
`"dependencies": [
{ "name": "ESP8266mDNS" }
]
`
Is this the proper way to do it?

valeros, 2020-04-17:
I don't how your code is structured, but LDF in default mode might not be able to find internal dependencies if they're deeply hidden in source files or included under `#ifdef` directive. Probably you need to configure dependency finder mode for your project more precisely. See docs for [LDF](https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode).

ASL07, 2020-04-17:
It also compiles if I add both
`lib_ldf_mode = deep+` to platformio.ini and `"libLDFMode": "deep+"` to library.json
I think this is a better option as it doesn't force me to explicitely add dependencies which are internal to the Arduino or framework