platformio/platformio-core #3464: Including sources from extra script with global_env.Append(SRC_FILTER=  doesn't work

ASL07, 2020-04-20:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Os X

**PlatformIO Version** (`platformio --version`): 4.3.1

### Description of problem
I have a multiplatform library and I want to add files to src_filter dynamically based on certain CPPDEFINES.
For this, I am using an extra script

For example, when `__FRAMEWORK_ESP8266__` is defined I want all the source files inside 
`module1/src/esp8266/ ` to be built.
For this, I have the following code in my extra script:

```
# pass flags to a global build environment (for all libraries, etc)
global_env = DefaultEnvironment()
script_dir = dirname(realpath('__file__'))

for item in global_env.get("CPPDEFINES", []):
    if item == "__FRAMEWORK_ESP8266__":
        # Arduino internal sources
        global_env.Append(SRC_FILTER=[
            "+<module1/src/esp8266/*>"
        ])
        env.Append(CPPPATH=[
            join(script_dir, "src/module1/src/esp8266/include")
        ])
        global_env.Append(CPPPATH=[
            join(script_dir, "src/module1/src/esp8266/include")
        ])

```

The headers are added correctly to the build the source files inside module1/src/esp8266/* are not added

Could you help or provide an example of adding sources dynamically from a script?

ivankravets, 2020-04-20:
1. Could you replace `global_env = DefaultEnvironment()` with `Import('env')` and use `env` instead of `global_env`?
2. You can replace `for item in global_env.get("CPPDEFINES", []):    if item == "__FRAMEWORK_ESP8266__":` with `if "__FRAMEWORK_ESP8266__" in env.get("CPPDEFINES", []):`

ASL07, 2020-04-20:
Thanks for the help,

Actually, this is a library, so I am looking for defines in the global environment.
For example, the application that uses the library should set the compile definitions, and then the library will search for specific definitions based on hardware and only include the related sources.

Does `Import('env')` get the global environment or the library environment ?
If it does get the global environment, what is the equivalent for getting the local(library) environment?

Also, is there a way to get the current platformio and framework from an extra_script inside a library? This way, I could use that and skip usage of additional CPPDEFINES
 
Something unrelated, but that will be very helpful, what is the easiest way to debug platformio extra_scripts and build_scripts in VSCode?

valeros, 2020-04-24:
Hi @ASL07 !
Looks like you have `srcFilter` specified in `library.json`. Please delete it and then dynamically replace `SRC_FILTER` variable in the `env` environment, e.g.:
```python
for item in env.get("CPPDEFINES", []):
    if "__FRAMEWORK_ESP8266__" in item:
        env.Replace(SRC_FILTER=["-<*>", "+<module1/src/esp8266/*>"])
```
Feel free to reopen the issue if you still need help.

closed 2020-04-24 by valeros

ASL07, 2020-04-24:
I add generic sources in library.json, and then specific ones are added dynamically in the script.
Do you mean that if I use a script, then I need to add all the sources dynamically in the script, even the generic ones?

ivankravets, 2020-04-24:
> Do you mean that if I use a script, then I need to add all the sources dynamically in the script, even the generic ones?

You are right.

ASL07, 2020-04-29:
I have mover everything to the extra_script and it is working now. Maybe the documentation should clarify that you cannot use src_filter in library.json and also add sources dynamically in extra_script, as it is not obvious