platformio/platformio-core #3465: Recursive glob expressions do not work

ASL07, 2020-04-20:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Os X

**PlatformIO Version** (`platformio --version`): 4.3.1

### Description of problem

I want to exclude recursively all files under any subdirectory called "test" "tests" "example" "examples"

I added the following glob expressions to the src_filter
```
src_filter =
    +<*>
    -<**/example/**>
    -<**/examples/**>
    -<**/test/**>
    -<**/tests/**>
```

But still, the files under these directories are built anyway

This happens both for platformio.ini and for library.json

How should I achieve what I want?

ivankravets, 2020-04-20:
Do you use Python 3? Python 2 does not support `**` and we will drop Python 2.7 soon. So, please use PlatformIO with Python 3.

ASL07, 2020-04-20:
I have python3 set as default on my system, I'm not sure about which one does platformio use
```
Mac-mini-de-Developer:$ python --version
Python 3.7.6
```
How do I check/modify which python interpreter is used by platformio?

ivankravets, 2020-04-21:
How did you install PlatformIO Core? Do you use PlatformIO IDE?

ASL07, 2020-04-21:
I had installed it with brew
Today, just to test, I uninstalled it and installed it using pip

valeros, 2020-04-24:
Hi @ASL07 ! I tried your `src_filter` and cannot reproduce the issue. It works just fine with `library.json` as well as with project sources. Please provide a minimal project to reproduce the issue. Also, you can try the latest PlatformIO from the [development branch](https://docs.platformio.org/en/latest/core/installation.html#development-version).

Feel free to reopen the issue if the problem persists.

closed 2020-04-24 by valeros