platformio/platformio-core #3476: Platformio 4.3+ generates unusable CMake files in CLion

thorrak, 2020-04-25:
Under Platformio 4.2, when running `platformio init --ide=clion` I get  a CMakelists.txt that starts with the following line and has numerous build targets, including ones for uploading a filesystem, rebuilding the project index, etc:

`cmake_minimum_required(VERSION 3.2)`

Under Platformio 4.3+, the cmake_minimum_version becomes <3.2 and the build targets are replaced with something unusable.

ivankravets, 2020-04-25:
Did you install PlatformIO for CLion plugin? See https://docs.platformio.org/en/latest/integration/ide/clion.html

thorrak, 2020-04-25:
I did, but my (existing) project didn't get picked up, and the CLion plugin doesn't seem to support the same feature set that was previously supported.

ivankravets, 2020-04-25:
Could you remove all CLion files and folders from project and regenerate? Also, please remove CMake files.

thorrak, 2020-04-25:
I've removed all files as requested, but the CMake files that were generated are still a far cry from what they were prior to Platformio 4.3.  

My CMakeLists.txt files are as follows:
CMakeLists under 4.2.1: [CMakeLists.txt](https://github.com/platformio/platformio-core/files/4533630/CMakeLists.txt)

CMakeLists under 4.3.1: [CMakeLists 4.3.txt](https://github.com/platformio/platformio-core/files/4533629/CMakeLists.4.3.txt)

I still am missing functionality like the ability to upload a SPIFFS filesystem for ESP32 devices, compile for different environments, or run the custom cmake commands I added. Additionally, functionality I don't use/want ("build" vs. "build production"?) is what seems to be added. 

Is there any way under Platformio 4.3 to _NOT_ use the plugin and revert to using the command line interface with the functionality that was available under 4.2?

valeros, 2020-04-25:
Hi @thorrak ! Could you please what it means "unusable"? If it's broken what is the error message? What platform/framework you're using?
Thanks!

valeros, 2020-04-25:
Hi @thorrak ! Could you please explain what you mean by "unusable"? Are there any errors? 

> Is there any way under Platformio 4.3 to NOT use the plugin and revert to using the command line interface with the functionality that was available under 4.2?

Probably the best way is to use PlatformIO v4.2 until the plugin has the required features.

thorrak, 2020-04-25:
> Hi @thorrak ! Could you please explain what you mean by "unusable"? Are there any errors?
>
>
> Probably the best way is to use PlatformIO v4.2 until the plugin has the required features.

What I mean by unusable is that after reinitializing the project in Platformio 4.3.1, the build options disappear completely:

[image]

Is the plan going forward that the plugin is now required for CLion use?

valeros, 2020-04-25:
If I'm not mistaken you need to manually load build environments as decribed [here](https://docs.platformio.org/en/latest/integration/ide/clion.html#clion-does-not-load-project-build-environments-from-platformio-ini) at least for the first time. 

> Is the plan going forward that the plugin is now required for CLion use?

At the moment this plugin is the only way to use PlatformIO projects with CLion. As a hacky workaround, you can create a custom [extra script](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html) and implement there a simplified version of our project generator with the previous CMake template.

blotfi, 2020-05-01:
Hi There
Thanks for your work. I installed CLion, VSCode and PIO, 2 days ago.
I am a happy user of IntelliJ IDEs (PyCharm, Android Studio, PHPStorm) and I find that Arduino IDE is so bad... that I was writing my programs on EditPlus or Sublime then compile/upload with Arduino IDE...
So the couple CLion-PIO is something great!

I wrote a tutorial in french today: https://baghli.blogspot.com/2020/05/clion-et-pateformio-au-lieu-darduino-ide.html
to encourage Arduino/ESP community to use CLion-PIO

The main issues I had:

I Couldn't install the PIO utilities from CLion alone. I had to first install them with VSCode
Same for the Platforms and tools (Espressif, AVR...), but I found that if I add to the path the PIO scripts, then CLion can found the PIO utility and install them like VSCode
Debug is not working (tried on ESP32 for Arduino)
Syntax highlighting of Arduino, Lib function and names (String, uint8_t, SoftwareSerial, U8g2,...) doesn't work, though the programs compile and upload well. It seems that CLion doesn't found them in its search path. I tried to add them to the CMakeLists, so far without success. For me, it is very important, since I want to clic on a function to go directly to its definition in the c:\Users\xxx.platformio\lib\yyyy
So if you and CLion team can fix this ASAP, it would be great.
Thanks

blotfi, 2020-05-02:
more info here : https://community.platformio.org/t/syntax-highlighting-not-working-for-external-lib-of-plateformeio-on-clion-ide/13431