platformio/platformio-core #3479: WinError 5 Access is denied error thrown when installing a dependency - not solved!

knopserl, 2020-04-26:
What kind of issue is this?

PlatformIO Core.
If you’ve found a bug, please provide an information below.
Configuration

Operating system: Windows 10 19.09 

VS Code
Version: 1.44.2 (user setup)
Commit: ff915844119ce9485abfe8aa9076ec76b5300ddd
Date: 2020-04-16T16:36:23.138Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.18363

PlatformIO Version (platformio --version): PlatformIO, version 4.3.1, Home 3.1.1
Description of problem

Whenever I install a dependency from GitHub repo (or full git url) the installation fails with:
Steps to Reproduce

    Create a simple PIO project (I used ESP32 devkit, see platformio.ini).
    Execute pio lib install mlewand/Transformation in the directory where platformio.ini file is located.

Actual Results

Installation fails with PermissionError: [WinError 5] Access is denied: 'X:\\Users\\userName\\Documents\\PlatformIO\\Projects\\pio-test-lib2\\.pio\\libdeps\\esp32doit-devkit-v1\\_tmp_installing-syxlieim-package\\.git\\objects\\06\\3bd79ae35475237188c6fcfed3b97aa076600f', however the lib gets installed and renamed.

The have been at least two different issues created and closed, but problem ist still there and reproducable with 5+ Windows 10 + VS Code + PlatformIO istallation on different PC's.

Issues have been also created at VS Code team which pointed to VS Code Add-In team but none of them helped.

ivankravets, 2020-05-10:
Duplicate of https://github.com/platformio/platformio-core/issues/2844

@knopserl could you provide a remote session do debug this issue?

closed 2020-05-10 by ivankravets

Bmooij, 2020-09-09:
Fix for : change the following line:
https://github.com/platformio/platformio-core/blob/99d4e0c390fed7cc6d4e4fd552803c3dea9574e5/platformio/managers/package.py#L569

```
shutil.move(pkg_dir, join(self.package_dir, target_dirname))
````
to
```
shutil.copytree(pkg_dir, join(self.package_dir, target_dirname))
```

ivankravets, 2020-09-09:
PlatformIO Core 5.0 is released. Please run `pio upgrade`. Do you have any issues?