platformio/platformio-core #3481: Could not initialize project

sesancho246, 2020-04-28:
PIO Core Call Error: "The current working directory C:\\Users\\sesan\\OneDrive\\Документы\\PlatformIO\\Projects\\telegram_bot will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\sesan\\OneDrive\\Документы\\PlatformIO\\Projects\\telegram_bot\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html\r\nPLATFORM: Espressif 8266 2.4.0 > NodeMCU 1.0 (ESP-12E Module)\r\nHARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash\r\nPACKAGES: \r\n - framework-arduinoespressif8266 3.20603.200130 (2.6.3) \r\n - tool-esptool 1.413.0 (4.13) \r\n - tool-esptoolpy 1.20800.0 (2.8.0) \r\n - toolchain-xtensa 2.40802.191122 (4.8.2)\r\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\r\nLDF Modes: Finder ~ chain, Compatibility ~ soft\r\nFound 27 compatible libraries\r\nScanning dependencies...\r\nNo dependencies\r\nBuilding in release mode\r\nException in thread Thread-7:\r\nTraceback (most recent call last):\r\n  File \"C:\\Users\\sesan\\.platformio\\python37\\lib\\threading.py\", line 926, in _bootstrap_inner\r\n    self.run()\r\n  File \"C:\\Users\\sesan\\.platformio\\python37\\lib\\threading.py\", line 870, in run\r\n    self._target(*self._args, **self._kwargs)\r\n  File \"C:\\Users\\sesan\\.platformio\\penv\\lib\\site-packages\\platformio\\proc.py\", line 50, in run\r\n    self.do_reading()\r\n  File \"C:\\Users\\sesan\\.platformio\\penv\\lib\\site-packages\\platformio\\proc.py\", line 103, in do_reading\r\n    for line in iter(self._pipe_reader.readline, \"\"):\r\n  File \"C:\\Users\\sesan\\.platformio\\python37\\lib\\codecs.py\", line 322, in decode\r\n    (result, consumed) = self._buffer_decode(data, self.errors, final)\r\nUnicodeDecodeError: 'utf-8' codec can't decode byte 0xc4 in position 1193: invalid continuation byte\r\n\r\n========================= [SUCCESS] Took 8.24 seconds ========================="

ivankravets, 2020-05-20:
Do you use Python 3.8?

ivankravets, 2020-06-03:
Please try PlatformIO IDE for VSCode 2.0 which uses own built-in Python 3 https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097

commit 5e4b4bba (2020-04-28): Fix "UnicodeDecodeError: 'utf-8' codec can't decode byte" when non-Latin chars are used in project path // Resolve #3481
closed 2020-06-03 by ivankravets