platformio/platformio-core #3482: Remote agent doesn't  start because of native library problems with cryptography library on raspberry pi1

kgyrtkirk, 2020-04-28:
### Configuration

**Operating system**:

Raspberry pi1b running: Raspbian GNU/Linux 10 (buster)

**PlatformIO Version**:

4.3.2 ; 4.3.3

### Description of problem

Remote agent startup bumps into an error.

#### Steps to Reproduce

run `pio remote agent start`

### Actual Results

```
x2@raspberrypi:~ $ pio remote agent start
********************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
********************************************************************************************************

CorePackageManager: Installing contrib-pysite @ ~2.37.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
contrib-pysite @ 2.37.191017 has been successfully installed!
Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/remote/command.py", line 65, in remote_agent_start
    from platformio.commands.remote.client.agent_service import RemoteAgentService
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/remote/client/agent_service.py", line 25, in <module>
    from platformio.commands.remote.client.base import RemoteClientBase
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/remote/client/base.py", line 30, in <module>
    from platformio.commands.remote.factory.ssl import SSLContextFactory
  File "/usr/local/lib/python3.7/dist-packages/platformio/commands/remote/factory/ssl.py", line 16, in <module>
    from OpenSSL import SSL  # pylint: disable=import-error
  File "/home/x2/.platformio/packages/contrib-pysite/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/home/x2/.platformio/packages/contrib-pysite/OpenSSL/crypto.py", line 12, in <module>
    from cryptography import x509
  File "/home/x2/.platformio/packages/contrib-pysite/cryptography/x509/__init__.py", line 8, in <module>
    from cryptography.x509.base import (
  File "/home/x2/.platformio/packages/contrib-pysite/cryptography/x509/base.py", line 18, in <module>
    from cryptography.x509.extensions import Extension, ExtensionType
  File "/home/x2/.platformio/packages/contrib-pysite/cryptography/x509/extensions.py", line 20, in <module>
    from cryptography.hazmat.primitives import constant_time, serialization
  File "/home/x2/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: /home/x2/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_constant_time.abi3.so: cannot open shared object file: No such file or directory

```

ivankravets, 2020-04-29:
Thanks! It fixed now. You can try to remove `contrib-pysite` package, PlatformIO Core will install default which does not work. However, it should automatically build a new package on your machine.

Danya0x07, 2020-05-02:
Still getting the same error on RaspberryPI 3B+ with Raspbian 9.11.
`ImportError: /home/pi/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_constant_time.abi3.so: cannot open shared object file: No such file or directory`

I tried to upgrade via `pio upgrade --dev`. There were a lot of packages being installed by pip and some warnings aka `WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/cryptography already exists. Specify --upgrade to force replacement.`

After that on `pio remote agent start` there was another error: 
*ImportError: /usr/lib/arm-linux-gnueabihf/libssl.so.1.1: version `OPENSSL_1_1_1' not found (required by /home/pi/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_openssl.abi3.so)*

ivankravets, 2020-05-02:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install libssl-dev

commit 5f79ab34 (2020-04-29): Automatically build ``contrib-pysite`` package on a target machine when pre-built package is not compatible // Resolve #3482
closed 2020-05-02 by ivankravets