platformio/platformio-core #3488: error on start platformio remote agent

pigghino, 2020-05-04:
hi,
i use the verion of platformio 4.3.3,when i try to run the command "platformio remote agent start"
it return this:

pi@raspberrypi:~ $ platformio remote agent start
Error: Traceback (most recent call last):
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/commands/remote/command.py", line 65, in remote_agent_start
    from platformio.commands.remote.client.agent_service import RemoteAgentService
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/commands/remote/client/agent_service.py", line 25, in <module>
    from platformio.commands.remote.client.base import RemoteClientBase
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/commands/remote/client/base.py", line 30, in <module>
    from platformio.commands.remote.factory.ssl import SSLContextFactory
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/commands/remote/factory/ssl.py", line 16, in <module>
    from OpenSSL import SSL  # pylint: disable=import-error
  File "/home/pi/.platformio/packages/contrib-pysite/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/home/pi/.platformio/packages/contrib-pysite/OpenSSL/crypto.py", line 12, in <module>
    from cryptography import x509
  File "/home/pi/.platformio/packages/contrib-pysite/cryptography/x509/__init__.py", line 8, in <module>
    from cryptography.x509.base import (
  File "/home/pi/.platformio/packages/contrib-pysite/cryptography/x509/base.py", line 18, in <module>
    from cryptography.x509.extensions import Extension, ExtensionType
  File "/home/pi/.platformio/packages/contrib-pysite/cryptography/x509/extensions.py", line 20, in <module>
    from cryptography.hazmat.primitives import constant_time, serialization
  File "/home/pi/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: /home/pi/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_constant_time.abi3.so: cannot open shared object file: No such file or directory

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

what can i do?

ivankravets, 2020-05-04:
It was fixed in the latest version. Could you try to run `pio upgrade --dev` and try again?

closed 2020-05-04 by ivankravets

pigghino, 2020-05-05:
i try the command but now (with version 4.3.4a1) i have this error:

Processing ./.cache/pip/wheels/de/75/c2/01f1e7a1ad37cb86d120f96cb1517d44627f2051de343a6080/Twisted-20.3.0-cp37-cp37m-linux_armv7l.whl
Collecting constantly>=15.1
  Using cached constantly-15.1.0-py2.py3-none-any.whl (7.9 kB)
Processing ./.cache/pip/wheels/84/93/fd/6bcb85b4bbf697c15b066350384cf3651feb47dcd5029a0b8d/zope.interface-5.1.0-cp37-cp37m-linux_armv7l.whl
Collecting Automat>=0.3.0
  Using cached Automat-20.2.0-py2.py3-none-any.whl (31 kB)
Collecting hyperlink>=17.1.1
  Using cached hyperlink-19.0.0-py2.py3-none-any.whl (38 kB)
Collecting attrs>=19.2.0
  Using cached attrs-19.3.0-py2.py3-none-any.whl (39 kB)
Collecting PyHamcrest!=1.10.0,>=1.9.0
  Using cached PyHamcrest-2.0.2-py3-none-any.whl (52 kB)
Collecting incremental>=16.10.1
  Using cached incremental-17.5.0-py2.py3-none-any.whl (16 kB)
Collecting setuptools
  Using cached setuptools-46.1.3-py3-none-any.whl (582 kB)
Collecting six
  Using cached six-1.14.0-py2.py3-none-any.whl (10 kB)
Collecting idna>=2.5
  Using cached idna-2.9-py2.py3-none-any.whl (58 kB)
Installing collected packages: constantly, setuptools, zope.interface, six, attrs, Automat, idna, hyperlink, PyHamcrest, incremental, twisted
Successfully installed Automat-20.2.0 PyHamcrest-2.0.2 attrs-19.3.0 constantly-15.1.0 hyperlink-19.0.0 idna-2.9 incremental-17.5.0 setuptools-46.1.3 six-1.14.0 twisted-20.3.0 zope.interface-5.1.0
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/constantly-15.1.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/zope already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/six-1.14.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/pkg_resources already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/incremental-17.5.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/attr already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/Automat-20.2.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/exampleproj already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/incremental already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/setuptools already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/automat already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/idna already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/constantly already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/zope.interface-5.1.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/easy_install.py already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/hamcrest already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/attrs-19.3.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/hyperlink-19.0.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/Twisted-20.3.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/PyHamcrest-2.0.2.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/hyperlink already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/zope.interface-5.1.0-py3.7-nspkg.pth already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/twisted already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/six.py already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/idna-2.9.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/setuptools-46.1.3.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/pi/.platformio/packages/contrib-pysite/bin already exists. Specify --upgrade to force replacement.
Collecting autobahn==20.4.3
  Using cached autobahn-20.4.3-py2.py3-none-any.whl (1.1 MB)
Collecting txaio>=20.3.1
[... cut ...]
ModuleNotFoundError: No module named 'twisted'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2020-05-05:
```
sudo apt-get install libssl-dev
```

Try again.

pigghino, 2020-05-05:
sorry i have read now another topic with the same problem.
now it work.
just one question if i need to reinstall all in the future i need to reinstall lib ssl?

tanks again
you are simply the best 
congratulations for the job

ivankravets, 2020-05-05:
New PIO Remote uses TLS secured channel. We need the SSL library to keep the whole communication secure.