platformio/platformio-core #3490: remote upload dosen't work

pigghino, 2020-05-05:
i have this problem, when i try to upload the code on my arduino mega it return this error:

============================================================ [SUCCESS] Took 1.72 seconds ============================================================
Uploading firmware remotely
Upon execvpe platformio [b'platformio', b'--force', b'run', b'-d', b'/home/pi/.platformio/remote/projects/mqtt_bus-efcc1677c3298e53bdc01fd9d32f1c68230b1370', b'-t', b'upload', b'-t', b'nobuild', b'--disable-auto-clean'] in environment id 3021732288
:Traceback (most recent call last):
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 406, in _fork
    environment)
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 484, in _execChild
    os.execvpe(executable, args, environment)
  File "/usr/lib/python3.7/os.py", line 580, in execvpe
    _execvpe(file, args, env)
  File "/usr/lib/python3.7/os.py", line 613, in _execvpe
    raise last_exc
  File "/usr/lib/python3.7/os.py", line 604, in _execvpe
    exec_func(fullname, *argrest)
FileNotFoundError: [Errno 2] No such file or directory: b'/bin/platformio'
The terminal process terminated with exit code: 1

what can i do?
Thank

ivankravets, 2020-05-05:
1. How did you install PlatformIO? Please use this installer script https://docs.platformio.org/en/latest/core/installation.html#installer-script
2. Do you use the latest version of PIO Core?

pigghino, 2020-05-06:
i have used the first line of the page:
python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)".

now my configuration is: 
1 develop machine 
 virtual machine (wmvare) debian 10
 visualstudiocode 1.44.2
 pio core              4.3.4a1
 pio home            3.2.1

2 remote machine:
 raspberry pi 4 raspbian 10
 PlatformIO, version 4.3.4a1

if i try to upload the firmware via host computer with arduino direct connected all is work
if i launch the command "pio run --target upload" on raspberry all is work
but if i try to launch the command "platformio remote run --target upload" not work

ivankravets, 2020-05-06:
How do you run `pio remote agent start` on RPi side?

pigghino, 2020-05-06:
What do you mean?

ivankravets, 2020-05-06:
Please read docs https://docs.platformio.org/en/latest/plus/pio-remote.html

you need to start PIo Remote agent on RPi side.

pigghino, 2020-05-06:
@ivankravets i start the agent by hand via ssh, and it work becose if i try to launch remote device i can see the arduino connected on raspberry

ivankravets, 2020-05-09:
Please send a private message in Community Forum https://community.platformio.org/u/ivankravets/summary

ivankravets, 2020-05-19:
Resolved via installing dependent packages:
```
sudo apt install python3-dev libffi-dev
pio remote agent start
```

closed 2020-05-19 by ivankravets