platformio/platformio-core #3491: Travis CI integration is not good. file.h: No such file or directory

sblantipodi, 2020-05-05:
Using platformio 4.3.3 here
this code:
https://github.com/sblantipodi/pc_ambilight

run well on my desktop

but not in travis ci.

if I run
`platformio run -e ambilight `

on my desktop it compiles without any problems.

on travis ci it says:
`"include/Ambilight.h:33:30: fatal error: BootstrapManager.h: No such file or directory"`

it's pretty bad to have different behaviour on travis than on desktop.

any help please?

sblantipodi, 2020-05-05:
it seems that travis does not read the 
`lib_extra_dirs = arduino_bootstrapper`
that I have in my `platformio.ini`

sblantipodi, 2020-05-05:
I have similar error with GitHub Actions.

Why CI ignore the lib_extra_dirs  I have in my platformio.ini?

sblantipodi, 2020-05-05:
it seems that lib_extra_dirs wants full path instead of relative path in linux.
this is a bug. please fix :)

sblantipodi, 2020-05-05:
it seems that platformio want the full path under linux for lib_extra_dirs
this is a bug, please fix :)

sblantipodi, 2020-05-09:
thanks for the great platform guys, PlatformIO is super awesome!

valeros, 2020-05-15:
Hi @sblantipodi ! I'm not sure how it works now, but there is obviously an issue with `lib_extra_dirs` usage in your project. In the [docs](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-extra-dirs) clearly stated that:
```
`lib_extra_dirs` is a not direct path to a library with source code. It should be a path to storage that contains libraries grouped by folders.
```
Besides, `arduino_bootstrapper` doesn't look like a library in terms of project structure, so you'll need to create a custom [library.json](https://docs.platformio.org/en/latest/librarymanager/config.html) with proper paths to header files, so PlatformIO will be able to pick   `arduino_bootstrapper` as a library.

closed 2020-05-15 by valeros