platformio/platformio-core #3504: problem in compiling  on platformio with ThinyWireM

JKWork3, 2020-05-10:
What kind of issue is this?

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

i have the same problem: Il processo del terminale è stato terminato. Codice di uscita: 1

Terminale verrà riutilizzato dalle attività, premere un tasto qualsiasi per chiuderlo.

> Executing task: platformio run <

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 2.0.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES: 
 - framework-arduino-avr 5.0.0 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <LiquidCrystal_I2C>
|   |-- <Wire> 1.0
|-- <RTClib> 1.6.0
|   |-- <TinyWireM> 1.0.1
|   |-- <Wire> 1.0
|-- <EEPROM> 2.0
|-- <LiquidCrystal> 1.5.0
|   |-- <TinyWireM> 1.0.1
|   |-- <Wire> 1.0
|-- <Adafruit Unified Sensor> 1.1.2
|   |-- <Adafruit ADXL343> 1.3.0
|   |   |-- <Adafruit BusIO> 1.2.3
|   |   |   |-- <Wire> 1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|-- <Adafruit BME280 Library> 2.0.2
|   |-- <Adafruit Unified Sensor> 1.1.2
|   |   |-- <Adafruit ADXL343> 1.3.0
|   |   |   |-- <Adafruit BusIO> 1.2.3
|   |   |   |   |-- <Wire> 1.0
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio/build/megaatmega2560/src/main.cpp.o
Compiling .pio/build/megaatmega2560/lib525/Wire/Wire.cpp.o
Compiling .pio/build/megaatmega2560/lib525/Wire/utility/twi.c.o
Compiling .pio/build/megaatmega2560/lib9ab/LiquidCrystal_I2C_ID576/LiquidCrystal_I2C.cpp.o
src/main.cpp:19:32: warning: invalid conversion from 'int' to 't_backlightPol' [-fpermissive]
 LiquidCrystal_I2C lcd(0x27,16,2); // 0x27 is the I2C bus address for an unmodified backpack 
                                ^
In file included from src/main.cpp:7:0:
/Users/giacomofidanza/.platformio/lib/LiquidCrystal_ID136/LiquidCrystal_I2C.h:65:4: note:   initializing argument 3 of 'LiquidCrystal_I2C::LiquidCrystal_I2C(uint8_t, uint8_t, t_backlightPol)'
    LiquidCrystal_I2C (uint8_t lcd_Addr, uint8_t backlighPin, t_backlightPol pol);
    ^
src/main.cpp: In function 'int temperatureHumidityControl()':
src/main.cpp:1245:47: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
        if (((currentMillis - previousMillisT) >= timeOff) && !digitalRead(tout)) {
                                               ^
src/main.cpp:1252:48: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (((currentMillis - previousMillisT) >= timeOn) && digitalRead(tout)) {
[... 186 lines cut ...]
         ^
lib/TinyWireM-master/USI_TWI_Master.cpp:302:21: error: 'USISIF' was not declared in this scope
   if( !(USISR & (1<<USISIF)) )
                     ^
lib/TinyWireM-master/USI_TWI_Master.cpp: In function 'unsigned char USI_TWI_Master_Stop()':
lib/TinyWireM-master/USI_TWI_Master.cpp:316:3: error: 'PORT_USI' was not declared in this scope
   PORT_USI &= ~(1<<PIN_USI_SDA);           // Pull SDA low.
   ^
lib/TinyWireM-master/USI_TWI_Master.cpp:316:20: error: 'PIN_USI_SDA' was not declared in this scope
   PORT_USI &= ~(1<<PIN_USI_SDA);           // Pull SDA low.
                    ^
lib/TinyWireM-master/USI_TWI_Master.cpp:317:19: error: 'PIN_USI_SCL' was not declared in this scope
   PORT_USI |= (1<<PIN_USI_SCL);            // Release SCL.
                   ^
lib/TinyWireM-master/USI_TWI_Master.cpp:318:12: error: 'PIN_USI' was not declared in this scope
   while( !(PIN_USI & (1<<PIN_USI_SCL)) );  // Wait for SCL to go high.  
            ^
lib/TinyWireM-master/USI_TWI_Master.cpp:324:9: error: 'USISR' was not declared in this scope
   if( !(USISR & (1<<USIPF)) )
         ^
lib/TinyWireM-master/USI_TWI_Master.cpp:324:21: error: 'USIPF' was not declared in this scope
   if( !(USISR & (1<<USIPF)) )
                     ^
*** [.pio/build/megaatmega2560/lib124/TinyWireM-master/USI_TWI_Master.cpp.o] Error 1
================================================================ [FAILED] Took 3.19 seconds =

JKWork3, 2020-05-10:
i work on mac whit VSC, the same program is compiled fine from Arduino IDE, i try the guide in this link but dont work: https://github.com/platformio/platformio-core/issues/1783

ivankravets, 2020-05-15:
Please forward to https://community.platformio.org/ and provide a simple project to reproduce this issue.

closed 2020-05-15 by ivankravets

JKWork3, 2020-05-17:
i try other way but dont have the same problem, i create this post:
https://community.platformio.org/t/compiling-problem-on-platformio-on-arduino-compiling-done-adafruit-tinywirem/13763?u=jackekrawattwork