platformio/platformio-core #3506: Pio tries to compile files specifically excluded in src_filter

ASL07, 2020-05-11:
### Configuration

**Operating system**: OS X

**PlatformIO Version** (`platformio --version`): 4.3.3

### Description of problem
Platformio is trying to compile a library which is not included anywhere
Please see relevant parts of my folder structure:
I am compiling this library:
[image]

This is the library.json content:
```
"export": {
        "exclude":
        [
            "src/iplace-common/extern/cpplogger/tests/*",
            "src/iplace-common/extern/cpplogger/build/*",
            "src/iplace-common/extern/etl/examples/*",
            "src/iplace-common/extern/etl/test/*",
            "src/iplace-common/extern/conectino/tests/**"
        ]
    },
    "build": {
        "flags":
        [
            "-Isrc/iplace-admin/include",
            "-Isrc/iplace-client/include",
            "-Isrc/iplace-common/include",
            "-Isrc/iplace-common/include/iplace-common/etl",
            "-Isrc/iplace-common/extern/etl/include",
            "-Isrc/iplace-common/extern/cpplogger/include",
            "-Isrc/iplace-device/include",
            "-Isrc/iplace-net/include",
            "-D __IPLACE_CONTROLLER_VERSION_MAJOR__=2",
            "-D __IPLACE_CONTROLLER_VERSION_MINOR__=2",
            "-D __IPLACE_CONTROLLER_VERSION_REVISION__=0"
        ], 
        "extraScript": "extra_script.py"
    },
    "frameworks": "*",
    "platforms": "*",
    "dependencies": [
        {
            "name": "FlashStorage"
        }
    ],
    "libLDFMode": "deep+"
```

This is extra_script content. Please see that iplace-common/conectino sources are excluded `"-<iplace-common/extern/conectino/src/**>"and` only included if "__BOARD_PRODINO_GPRS__" is defined:
```
Import('env')
from os.path import join, realpath, dirname

# pass flags to a global build environment (for all libraries, etc)
global_env = DefaultEnvironment()
script_dir = dirname(realpath('__file__'))

# private library flags
env.Replace(SRC_FILTER=[
    "+<*>",
    "-<**/example/**>",
    "-<**/examples/**>",
    "-<**/test/**>",
    "-<**/tests/**>",

    # Exclude HW dependent sources which will be included depending on environment
    "-<iplace-net/src/esp8266/**>",
[... 167 lines cut ...]
Archiving .pio/build/olimex_esp8266_evb/lib07c/libDallasTemperature_ID54.a
Indexing .pio/build/olimex_esp8266_evb/lib07c/libDallasTemperature_ID54.a
Compiling .pio/build/olimex_esp8266_evb/lib714/conectinolibrary/common/cn_tcp.cpp.o
Compiling .pio/build/olimex_esp8266_evb/lib714/conectinolibrary/common/cn_udp.cpp.o
Compiling .pio/build/olimex_esp8266_evb/lib714/conectinolibrary/modem/cn_modem.cpp.o
Compiling .pio/build/olimex_esp8266_evb/lib714/conectinolibrary/utils/cn_logger.cpp.o
/Users/developer/Code/common/conectinolibrary/src/modem/cn_modem.cpp: In member function 'virtual void Conectuino::Modem::poll()':
/Users/developer/Code/common/conectinolibrary/src/modem/cn_modem.cpp:292:31: warning: unused variable 'urc_end' [-Wunused-variable]
                         char* urc_end = strstr(_buffer, get_rsp_terminator()) + strlen(get_rsp_terminator());        
                               ^
/Users/developer/Code/common/conectinolibrary/src/common/cn_udp.cpp: In member function 'bool Conectuino::Udp::is_duplicate(uint8_t*, size_t)':
/Users/developer/Code/common/conectinolibrary/src/common/cn_udp.cpp:275:64: error: 'mem_in_mem' was not declared in this scope
     if (mem_in_mem(rx_buf, rx_bytes, new_packet, new_packet_len))
                                                                ^
/Users/developer/Code/common/conectinolibrary/src/common/cn_tcp.cpp: In member function 'bool Conectuino::Tcp::is_duplicate(uint8_t*, size_t)':
/Users/developer/Code/common/conectinolibrary/src/common/cn_tcp.cpp:340:64: error: 'mem_in_mem' was not declared in this scope
     if (mem_in_mem(rx_buf, rx_bytes, new_packet, new_packet_len))
                                                                ^
*** [.pio/build/olimex_esp8266_evb/lib714/conectinolibrary/common/cn_udp.cpp.o] Error 1
*** [.pio/build/olimex_esp8266_evb/lib714/conectinolibrary/common/cn_tcp.cpp.o] Error 1
```

It is really strange that conectino is listed as a dependency of -- <ESP8266WiFi> 1.0

Do you have any suggestions? Is there any way to get more information about where specific files/libraries are added to the build

valeros, 2020-05-18:
Hi @ASL07 ! Is there a repository for your project or maybe you can upload it somewhere?

ASL07, 2020-05-25:
Hi @valeros, I have added you as collaborator to one of the projects where I am having the same issue.
Please see details on the repository readme
Thank you

valeros, 2020-05-26:
Resolved in https://github.com/platformio/platformio-core/commit/0eb8895959c90cda07f014294d854cf76913c0d7

Looks like the recursive `**` pattern wasn't fully supported. Please retest with the latest platformio-core from the development branch. Keep in mind that only Python 3 supports this feature.

ASL07, 2020-05-27:
I have updated to the last dev branch and I still face the issue.
The recursive glob expressions work for me in the main project's platformmio.ini, but they don't work for libraries (when using glob expressions inside librarty.json or in the library extra_script)

valeros, 2020-05-27:
That's strange. I tested it on your project `modbus` and it works as expected. What OS do you use?

ASL07, 2020-05-30:
I am using Os X and Python 3.7.7. 
Have retested again using the repo modbus and I am experiencing the same issue

valeros, 2020-06-01:
Looks like we're testing different things. Could you please try to compile the following basic project with your library specified as a dependency: [test-recursive.zip](https://github.com/platformio/platformio-core/files/4711509/test-recursive.zip)

valeros, 2020-06-16:
Hi @ASL07 ! Any updates on the issue?

closed 2020-06-16 by ivankravets