platformio/platformio-core #3523: PIO Unit Testing fails to build multiple environments at once

mje-nz, 2020-05-26:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS 10.13.6, also Ubuntu 16.04 in a VM

**PlatformIO Version** (`platformio --version`): 4.3.4

### Description of problem
If you run `pio test` in a project with multiple environments, many environments fail to build. Running the tests in each environment one-by-one works fine.

#### Steps to Reproduce

1. Clone [platformio/platformio-examples](https://github.com/platformio/platformio-examples)
2. Run `pio test` in `unit-testing/calculator`

### Actual Results
```
Verbose mode can be enabled via `-v, --verbose` option
Collected 3 items

Processing test_common in uno environment
----------------------------------------------
Building...
Uploading...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
===== [FAILED] Took 1.90 seconds =====

Processing test_common in nodemcu environment
----------------------------------------------
Building...
Uploading...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
===== [FAILED] Took 3.82 seconds =====

Processing test_common in native environment
----------------------------------------------
Building...

*** Multiple ways to build the same target were specified for: platformio-examples/unit-testing/calculator/.pio/build/native/test/output_export.o  (from ['platformio-examples/unit-testing/calculator/test/output_export.c'] and from ['platformio-examples/unit-testing/calculator/test/output_export.cpp'])
File "~/.pyenv/versions/3.7.3/lib/python3.7/site-packages/platformio/builder/tools/platformio.py", line 347, in <listcomp>
===== [FAILED] Took 0.66 seconds =====

Processing test_desktop in native environment
----------------------------------------------
Building...

*** Multiple ways to build the same target were specified for: platformio-examples/unit-testing/calculator/.pio/build/native/test/output_export.o  (from ['platformio-examples/unit-testing/calculator/test/output_export.c'] and from ['platformio-examples/unit-testing/calculator/test/output_export.cpp'])
File "~/.pyenv/versions/3.7.3/lib/python3.7/site-packages/platformio/builder/tools/platformio.py", line 347, in <listcomp>
===== [FAILED] Took 0.64 seconds =====

Processing test_embedded in uno environment
----------------------------------------------
Building...
.pio/build/uno/test/output_export.cpp.o (symbol from plugin): In function `output_start':
(.text+0x0): multiple definition of `output_start'
.pio/build/uno/test/output_export.c.o (symbol from plugin):(.text+0x0): first defined here
.pio/build/uno/test/output_export.cpp.o (symbol from plugin): In function `output_start':
(.text+0x0): multiple definition of `output_char'
.pio/build/uno/test/output_export.c.o (symbol from plugin):(.text+0x0): first defined here
.pio/build/uno/test/output_export.cpp.o (symbol from plugin): In function `output_start':
(.text+0x0): multiple definition of `output_complete'
.pio/build/uno/test/output_export.c.o (symbol from plugin):(.text+0x0): first defined here
.pio/build/uno/test/output_export.cpp.o (symbol from plugin): In function `output_start':
(.text+0x0): multiple definition of `output_flush'
.pio/build/uno/test/output_export.c.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/uno/firmware.elf] Error 1
===== [FAILED] Took 0.94 seconds =====

Processing test_embedded in nodemcu environment
----------------------------------------------
Building...
~/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/nodemcu/test/output_export.cpp.o: in function `output_start':
output_export.cpp:(.text.output_start+0xc): multiple definition of `output_start'; .pio/build/nodemcu/test/output_export.c.o:output_export.c:(.text.output_start+0x0): first defined here
~/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/nodemcu/test/output_export.cpp.o: in function `output_char':
output_export.cpp:(.text.output_char+0x4): multiple definition of `output_char'; .pio/build/nodemcu/test/output_export.c.o:output_export.c:(.text.output_char+0x8): first defined here
~/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/nodemcu/test/output_export.cpp.o: in function `output_flush':
output_export.cpp:(.text.output_flush+0x8): multiple definition of `output_flush'; .pio/build/nodemcu/test/output_export.c.o:output_export.c:(.text.output_flush+0x8): first defined here
~/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/nodemcu/test/output_export.cpp.o: in function `output_complete':
output_export.cpp:(.text.output_complete+0x8): multiple definition of `output_complete'; .pio/build/nodemcu/test/output_export.c.o:output_export.c:(.text.output_complete+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/nodemcu/firmware.elf] Error 1
===== [FAILED] Took 2.08 seconds =====

Test           Environment    Status    Duration
-------------  -------------  --------  ------------
test_common    uno            FAILED    00:00:01.898
test_common    nodemcu        FAILED    00:00:03.821
test_common    native         FAILED    00:00:00.663
test_desktop   uno            IGNORED
test_desktop   nodemcu        IGNORED
test_desktop   native         FAILED    00:00:00.637
test_embedded  uno            FAILED    00:00:00.936
test_embedded  nodemcu        FAILED    00:00:02.080
test_embedded  native         IGNORED
====== 6 failed, 0 succeeded in 00:00:10.035 ======
```

### Expected Results
Uno and NodeMCU tests should at least build. Native tests should build and pass.

### Additional info
I'm not sure if running tests on multiple embedded targets at once is supposed to work, but the example in the docs is `pio test -e uno -e native` and it fails in the same way.

I came across this problem on a personal project with an ESP32 environment and a native environment -- the ESP32 test passes and the native test fails to build with the same `Multiple ways to build the same target were specified for ...` error.

bastianhjaeger, 2020-05-29:
Same here. 

If I start my platformio.ini with
```
[platformio]
default_envs = 
    esp32
    esp8266
    desktop
```

I get the error on "Processing common in esp32 environment"

and I i remove the "desktop" from the defaults, "common with desktop" is ignored

closed 2020-06-11 by valeros

valeros, 2020-06-16:
Thanks for the report! Please upgrade to the latest PlatformIO from the development branch via `pio upgrade --dev` and try again.

mje-nz, 2020-06-16:
Works for me, thanks :)

commit fdb83c24 (2020-06-11): Clean autogenerated files before running tests // Resolve #3523

Fixes possible conflicts between auxiliary test transport files when
project contains multiple environments with different platforms