platformio/platformio-core #3524: check: ignore library files

peterus, 2020-05-26:
- [X] **PlatformIO Core**.

### Additional info

I am using cppcheck for static code analysis. If I run `platformio check` also the used libraries are checked. Some of this libraries are not perfectly implemented and I don't want to have them in my overview.
It would be nice to have a flag to turn of cppchek on the library-files in `.pio/libdeps`

ivankravets, 2020-05-26:
https://docs.platformio.org/en/latest/projectconf/section_env_check.html#check-patterns

closed 2020-05-26 by ivankravets

lalten, 2020-07-10:
Can you please give an example how to suppress `cppcheck` checking of `.pio/libdeps`? I tried to configure `check_flags = --suppress=*.pio`, but it still prints a long list of defects in third party code (in this case, [ETL](https://www.etlcpp.com/), on platform [native](https://docs.platformio.org/en/latest/platforms/native.html))

peterus, 2020-07-10:
i am using this setting kn platformio.ini:

`check_flags = cppcheck: --suppress=*:*.pio\* --inline-suppr`

lalten, 2020-07-10:
Nice, the trick is to have both the wildcard before .pio and the escaped one after. Thanks!