platformio/platformio-core #3531: Multiple build middlewares with pattern crash

mcspr, 2020-05-28:
### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 4.3.5a1

### Description of problem

Following the documentation at https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#build-middlewares
When extra script defines **multiple** build middewares and one of them changes Node object via `env.Object(node, ...)`, build will crash because it does not expect a different type of object.

#### Steps to Reproduce

Just `pio run`

### Actual Results

Build crashes on the second iteration of the middleware handler:
```python
~/test> pio run
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
---
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.5.1 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20701.0 (2.7.1)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 61 compatible libraries
Scanning dependencies...
No dependencies
<class 'SCons.Node.FS.File'> /home/max/test/src/main.cpp
<class 'SCons.Node.NodeList'> ['/home/max/test/.pio/build/d1_mini/src/main.cpp.o']
AttributeError: 'NodeList' object has no attribute 'srcnode':
  File "/mnt/c/Users/prokh/Projects/platformio-core/platformio/builder/main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/max/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/max/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/max/.platformio/platforms/espressif8266/builder/main.py", line 216:
    target_elf = env.BuildProgram()
  File "/home/max/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/mnt/c/Users/prokh/Projects/platformio-core/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/home/max/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/mnt/c/Users/prokh/Projects/platformio-core/platformio/builder/tools/platformio.py", line 157:
    projenv.BuildSources(
  File "/home/max/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/mnt/c/Users/prokh/Projects/platformio-core/platformio/builder/tools/platformio.py", line 344:
    nodes = env.CollectBuildFiles(variant_dir, src_dir, src_filter)
  File "/home/max/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/mnt/c/Users/prokh/Projects/platformio-core/platformio/builder/tools/platformio.py", line 288:
    if pattern and not fnmatch.fnmatch(node.srcnode().get_path(), pattern):
================================================================================= [FAILED] Took 0.83 seconds =================================================================================
~/test>
```

### Expected Results

I'd guess `NodeList` should be expected somehow?

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
extra_scripts = pre:script.py
```

**Source file to reproduce issue:**
src/main.cpp
```cpp
#include <Arduino.h>
void setup() {}
void loop() {}
```

script.py
```python
Import("env")

print('---')

def middleware(node):
    override = env.Object(node, CPPFLAGS=env["CPPFLAGS"] + [("-DFOO","-DBAR")])
    print(type(node), node)
    print(type(override), override)
    return override

def dummy(node):
    print('this will never happen because of the crash')
    return node

env.AddBuildMiddleware(middleware, "*/src/main.cpp")
env.AddBuildMiddleware(dummy, "*/src/main.h")
```
### Additional info

mcspr, 2020-06-15:
Re df0e601

Is it Scons limitation or is the following intended? Small change to the `dummy` func to print the node:
```python
def dummy(node):
    print('dummy for', node)
    return node

env.AddBuildMiddleware(middleware, "*/src/main.cpp")
env.AddBuildMiddleware(dummy, "*/src/main.cpp")
```
When we get NodeList's [0] element, src node result will no longer match the `main.cpp` but `main.cpp.o`, so the `dummy()` function like above will never be called b/c we middleware modified it

ivankravets, 2020-06-23:
@mcspr  Thanks for the report! Please re-test with `pio upgrade --dev`.

Does it work now?

closed 2020-06-23 by ivankravets

mcspr, 2020-06-23:
Yes, thanks, that sorts it out.

commit df0e6016 (2020-06-15): Handle possible NodeList in source files when processing Middlewares // Resolve #3531

env.Object() returns a list of objects that breaks the processing of
subsequent middlewares since we only expected File nodes.
commit 82735dd5 (2020-06-23): Fixed an issue with improper processing of source files added via multiple Build Middlewares // Resolve #3531