platformio/platformio-core #3535: Spurious rebuilds - partial fix

MitchBradley, 2020-05-31:
- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Windows 10**:

**PlatformIO, version 4.3.4**

### Description of problem

I am using platformio to build the source code in https://github.com/bdring/Grbl_Esp32

Even with no changes to the source tree, it often recompiles most or all of the .cpp files and relinks the .elf file.

I can be reproduced by executing "pio run" several times in succession.  It should compile everything the first time, then subsequent times it should skip compilation and linking, since nothing has changed.  The problem is somewhat random.  There is a small chance that it will work correctly, but usually it performs unnecessary recompiles.

Using the "--debug=explain" flag in SCONSFLAGS, I determined that the root cause of the problem is that the order of "-I <include dir>" flags on the cpp command line changes from build to build.  It is always the same list of include directories, but the order is different, triggering SCons to recompile.  This is especially troublesome in conjunction with the "longcmd" trick that is used to work around the Windows command line limitation.

I fixed the spurious recompile problem by sorting the list of include files, adding the line marked with "+" in piomaxlen.py 

```
def long_incflags_hook(env, incflags):
    _incflags = env.subst(incflags).replace("\\", "/")
+    _incflags = " -I".join(sorted(_incflags.split(" -I")))
```

For the related spurious-relinking problem, I determined that the root cause is similar - the order of the library file (*.a) on the linker command line changes from build to build.  I have tried to find a similar fix by sorting the list of libraries, but I have not found the right place in the code to do that.

MitchBradley, 2020-05-31:
This is the output of two consecutive "pio run" invocations with "--debug=explain".  The patch above has been applied, so the sources are not recompiled unnecessarily.  As you can see, the library dependencies are discovered in a different order in the two runs, leading to a different linker command line, resulting in an unnecessary re-link.

[platformio-run.txt](https://github.com/platformio/platformio-core/files/4707121/platformio-run.txt)

MitchBradley, 2020-05-31:
I also tried all combinations  `pio run --jobs=1`  and `$Env:SCONSFLAGS = "--jobs=1"` in a attempt to stabilize the library dependency scanning.  No combination helped; in all cases, the order of libraries as shown in both the Dependency Graph and the linker command line changed from build to build.

MitchBradley, 2020-05-31:
Aha, I have solved it.  Adding the .sort line below, in piolib.py, stabilizes the order.

```
    def _print_deps_tree(root, level=0):
        margin = "|   " * (level)
+        root.depbuilders.sort()
```

MitchBradley, 2020-06-01:
The change above should read

```
    root.depbuilders.sort(key=lambda lb: lb.path)
```

I am trying to create a PR but it is very frustrating because the instructions for contributing do not work exactly under Windows, and the tests failed even without my changes.

valeros, 2020-06-16:
Hi @MitchBradley ! Sorry for the late reply. What Python version do you use? I'm able to occasionally reproduce the issue but only on Python 2.x. It looks like the issue is caused by the standard `dict` type in PY2 as it doesn't maintain insertion order by default. I saw your fix, but the changes you're suggesting may unintentionally break some user projects since the order of include paths affects build environment.

MitchBradley, 2020-06-16:
Until about 10 days ago I was using Python 2.7 but then I switched to Python 3.7.7.  I will retest with the new Python with the patches reverted and report on my results.  The dict insertion order effect would certainly explain it.

MitchBradley, 2020-06-17:
So far it is working well with Python 3.7.7.  I wonder if the right answer is to stop supporting old Python versions, or to switch to the OrderedDict class?

valeros, 2020-06-17:
We don't recommend using Python 2.x and plan to gradually switch users to Python 3, especially on Windows.  Feel free to reopen the issue if you still need any help.

closed 2020-06-17 by valeros