platformio/platformio-core #3538: Cant compile project UnicodeDecodeError: 'utf8' codec can't decode byte 0xe1 in position 69: invalid continuation byte:

nbJosh, 2020-06-01:
------------------------------------------------------------------

macOS 10.15.5

**Operating system**:

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 4.3.5a1

Core 4.3.4·Home 3.2.2

### Description of problem

#### Steps to Reproduce

1. Load project into PIO 
2. Attempt to build 

### Actual Results
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe1 in position 69: invalid continuation byte: File "/Users/joshua/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")

### Expected Results
project builds and runs on device. 
Project had been working years ago,  but sometime after moving to Vs code, was unable to build.  

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
;upload_port = /dev/cu.usbmodem1A1221
;upload_port = /dev/cu.usbmodem14511
;upload_port = /dev/cu.usbmodem14111
;upload_port = /dev/cu.usbmodem144420
upload_port = /dev/cu.usbmodem1444201

; [env:stable]
; platform = teensy
; board = teensy36
; framework = arduino
; upload_port = /dev/cu.usbmodem4040101

[platformio]
lib_dir=~/Documents/Arduino/Libraries

```

### Additional info

nbJosh, 2020-06-02:
I setup VSCode + PlatformIO + my project + my Arduino library folder. on a computer with a fresh OS install,   and I get the same error.

nbJosh, 2020-06-02:
This issue effects all projects,  including marlin firmware

ivankravets, 2020-06-03:
We don't recommend using Python 2 on Windows. Please switch to PlatformIO IDE for VSCode 2.0 beta. it has built-in Python 3. https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097

nbJosh, 2020-06-03:
Um,   ok,  thanks,  I'm not using windows,  thanks for closing this issue without providing any support.

ivankravets, 2020-06-03:
Sorry, it's my mistake, I thought it's Windows due to UnicodeDecodeError

1. Do you have non-ASCII chars in your full project path?
2. What is your macOS version?
3. Could you share somewhere a simple project to reproduce this issue?

nbJosh, 2020-06-03:
How can I view the full project path?

nbJosh, 2020-06-03:
Creating a new project,  targeting the ATMega, and hitting build, results in this issue.

ivankravets, 2020-06-03:
Could you provide a remote [AnyDesk](https://anydesk.com/en) readOnly session under your control? I tried to reproduce this issue on my macOS and can't :(

nbJosh, 2020-06-03:
I did a process of elimination,  
It turns out that it was one library in my Arduino library folder, with an author with a unicode character in they're name.

The Wifi_LInk library,   in the library.properties file.  Juraj Andrássy
https://github.com/jandrassy/arduino-library-wifilink

closed 2020-06-04 by nbJosh